espressif/esp-idf #12107: 使用https ota升级时，一直验证失败

Szeroy, 2023-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用的是simple ota的例子，只修改了这一段
```
    esp_http_client_config_t config = {
        .url = "https://24u0d66921.goho.co/simple_ota.bin",//CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
#ifdef CONFIG_EXAMPLE_USE_CERT_BUNDLE
        .crt_bundle_attach = esp_crt_bundle_attach,
#else
        .cert_pem = (char *)server_cert_pem_start,               //使用的是这个
#endif /* CONFIG_EXAMPLE_USE_CERT_BUNDLE */
        .event_handler = _http_event_handler,
        .keep_alive_enable = true,
#ifdef CONFIG_EXAMPLE_FIRMWARE_UPGRADE_BIND_IF
        .if_name = &ifr,
#endif
    };
```
一直报错验证过不去
```
I (5844) simple_ota_example: Starting OTA example task
I (5854) simple_ota_example: Attempting to download update from https://24u0d66921.goho.co
I (5864) main_task: Returned from app_main()
I (5874) wifi:<ba-add>idx:0 (ifx:0, 0e:3a:38:11:e7:36), tid:0, ssn:69, winSize:64
E (5984) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (5984) esp-tls-mbedtls: Failed to verify peer certificate!
E (5984) esp-tls: Failed to open new connection
E (5994) transport_base: Failed to open a new connection
E (5994) HTTP_CLIENT: Connection failed, sock < 0
E (6004) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6014) esp_https_ota: Failed to establish HTTP connection
E (6014) simple_ota_example: Firmware upgrade failed
```
应该是pem文件有问题，但是不知道应该怎么改，我用的是香橙派搭建了一个apache2的服务器，在内网里面，因为我没有公网ip，使用的是花生壳内网穿透工具，访问这个地址可以看到以前临时做的一个网页。
`https://24u0d66921.goho.co/wp-content/themes/dwthemes/`
这是花生壳的官网：
`https://hsk.oray.com/download/?utm_source=baidu&utm_medium=cpc&utm_campaign=hsk_download&referral_id=45629&wordid=pc010090075202105100006535&adp=cl1&bd_vid=7558192445521571442`
他们提供的域名是支持https的，现在就是不清楚应该怎么操作才能连接上我的服务器。

Szeroy, 2023-08-21:
[image]
这是menuconfig的配置，第一项在代码里面不起作用了，已经换成网址了

mahavirj, 2023-08-24:
@Szeroy 

Do you have `CONFIG_EXAMPLE_USE_CERT_BUNDLE` enabled in your sdkconfig? I would suggest that you turn on this config which will enable the certificate bundle feature and hence all commonly used root certificates shall be automatically available to the application.

Szeroy, 2023-08-25:
> @Szeroy
> 
> Do you have `CONFIG_EXAMPLE_USE_CERT_BUNDLE` enabled in your sdkconfig? I would suggest that you turn on this config which will enable the certificate bundle feature and hence all commonly used root certificates shall be automatically available to the application.

成功了，谢谢你

closed 2023-08-25 by Szeroy