espressif/esp-idf #12108: Got Image hash failed error during OTA

zzzh, 2023-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

core-esp32-c3

### Power Supply used.

USB

### What is the expected behavior?

Pass the OTA progress

### What is the actual behavior?

Image file downloaded ok with right size, but always get:

esp_image: Image hash failed - image is corrupt

### Steps to reproduce.

1. build from a copy of hello-world example
2. get a whole image by running "esptool.py --chip esp32c3 merge_bin -o hello_all.bin @flash_args" in build folder
3. put the image on a https server
4. make a copy of native_ota_example
5. set wifi, https server address, skip server certification verification in menuconfig
6. build, flash and monitor the native_ota_example image.
7. got Image hash failed error.

### Debug Logs.

```plain
I (8124) native_ota_example: esp_ota_begin succeeded
I (29884) native_ota_example: Connection closed
I (29884) native_ota_example: Total Write binary data length: 216960
I (29884) esp_image: segment 0: paddr=00110020 vaddr=3fcd6100 size=016bch (  5820) 
I (29894) esp_image: segment 1: paddr=001116e4 vaddr=403ce000 size=00930h (  2352) 
I (29904) esp_image: segment 2: paddr=0011201c vaddr=403d0000 size=02d28h ( 11560) 
E (29914) esp_image: Image hash failed - image is corrupt
E (29914) native_ota_example: Image validation failed, image is corrupted
E (29924) native_ota_example: Exiting task due to fatal error...
```

### More Information.

Did not try the latest version of esp-idf since I need version 4.4.2 to run my code, it would take long time to change them..

mahavirj, 2023-08-21:
@zzzh 

> get a whole image by running "esptool.py --chip esp32c3 merge_bin -o hello_all.bin @flash_args" in build folder
put the image on a https server

Not sure why you are merging all artifacts together but you are supposed to host only the application binary on the OTA update server, for this case it would be `native_ota.bin`. Please try it once and it should fix the problem. You may read more in this [technical post](https://blog.espressif.com/ota-updates-framework-ab5438e30c12).

zzzh, 2023-08-21:
@mahavirj 

You are right, sorry, my mistake.
Thanks for the help.

closed 2023-08-21 by zzzh