espressif/esp-idf #12110: boot:0x8 (SPI_FAST_FLASH_BOOT)

LCKway, 2023-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

SP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

Alvin1Zhang, 2023-08-21:
Thanks for reporting, would you please help provide more details as per [the issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

Alvin1Zhang, 2023-09-21:
Thanks for reporting, will close due to short of details and feedback, feel free to reopen with more updates. Thanks.

closed 2023-09-21 by Alvin1Zhang