espressif/esp-idf #12111: Build failure with examples/protocols/esp_http_client on Linux

AxelLin, 2023-08-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2317-g3247253671

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build should work.

### What is the actual behavior?

xel@alpha:~/esp/esp-idf/examples/protocols/esp_http_client$ idf.py fullclean build
Executing action: fullclean
Executing action: remove_managed_components
Executing action: all (aliases: build)
Running cmake in directory /home/axel/esp/esp-idf/examples/protocols/esp_http_client/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/axel/.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/axel/esp/esp-idf/examples/protocols/esp_http_client"...
CMake Error at CMakeLists.txt:7 (if):
  if given arguments:

    "STREQUAL" "linux"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/axel/esp/esp-idf/examples/protocols/esp_http_client/build/log/idf_py_stderr_output_2432400 and /home/axel/esp/esp-idf/examples/protocols/esp_http_client/build/log/idf_py_stdout_output_2432400

### Steps to reproduce.

$ idf.py fullclean build

### More Information.

v5.1 branch also has the same build error.

mahavirj, 2023-08-21:
@AxelLin 

Do you have any local changes in the example or SDK? Can you please share entire build log? FWIW, we have this example build covered for `linux` target in the CI ([reference](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/examples/protocols/esp_http_client/pytest_esp_http_client.py#L103)). I also tried to build it locally and did not face any issue with the latest master.

AxelLin, 2023-08-21:
@mahavirj 
It works when I try to provide you the entire build log now.
No idea why I got build failure in previous build, sorry for the noise.

igrr, 2023-08-21:
I think I got the same error when trying to build the example when no sdkconfig file was created. Doing `idf.py set-target <target>` before `idf.py build` fixed the issue. The example CMakeLists file doesn't seem to handle the case of IDF_TARGET variable not being set yet.

closed 2023-08-22 by AxelLin