espressif/esp-idf #12114: I want to use AAC L2 codec in A2DP sink.

hyojincha, 2023-08-22:
The board I use consists of ESP32 + VS1053.

And I know that VS1053 itself can decode AAC.

Currently, Blueroid A2DP library seems to be fixed as SBC codec.

Is there any way to change this to AAC and receive undecoded sound source data through a callback function?

Please give me a solution.

Thank you. sir.

boblane1, 2023-08-24:
Hi @hyojincha, 

Thanks for your report.  There is some other work to support AAC, such as A2DP stream discovering and codec negotiation. We intend to support other codec in the future, we lack human resource currently.

boblane1, 2024-03-01:
Hi @hyojincha,

We accept your feature request, but regret to inform you that we are still short of human resources at this time. We may develop this feature later this year or next year.

Finney-esp, 2026-05-09:
Hi, I'm very glad to tell you the following information：
- Currently, the master branch of ESP-IDF already supports A2DP sink using the AAC codec (external codec). The configuration to enable `BT_A2DP_CODEC_AAC_ENABLED` must be enabled, and the reported data is raw aac. The AAC decoding can be directly performed using `esp_audio_codec`.
- A2DP source currently does not support the AAC codec. Support for it will be provided in the near future. At the same time, the example in IDF will be improved in the near future to demonstrate how to use the AAC codec.

hyzgit, 2026-05-09:
> Hi, I'm very glad to tell you the following information：嗨，我很高兴告诉您以下信息：
> 
> * Currently, the master branch of ESP-IDF already supports A2DP sink using the AAC codec (external codec). The configuration to enable `BT_A2DP_CODEC_AAC_ENABLED` must be enabled, and the reported data is raw aac. The AAC decoding can be directly performed using `esp_audio_codec`.目前，ESP-IDF 的主分支已经支持使用 AAC 编解码器（外部编解码器）的 A2DP 接收端。启用 `BT_A2DP_CODEC_AAC_ENABLED` 的配置必须被启用，报告的数据是原始 AAC 数据。可以使用 `esp_audio_codec` 直接进行 AAC 解码。
> * A2DP source currently does not support the AAC codec. Support for it will be provided in the near future. At the same time, the example in IDF will be improved in the near future to demonstrate how to use the AAC codec.当前的 A2DP 源端不支持 AAC 编解码器。它将在不久的将来提供支持。同时，IDF 中的示例将在不久的将来得到改进，以展示如何使用 AAC 编解码器。

Can it be connected via Bluetooth on the mobile phone and play music encoded in AAC? Will the mobile phone automatically recognize the device (ESP32) as a device that supports AAC?