espressif/esp-idf #12115: [Question] If an OTA is interrupted, will the bootloader detect the invalid image? always?

chipweinberger, 2023-08-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have read the documentation here: 
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html

If the main OTA update did not complete (i.e. due to power loss):
1. I want to make sure my device bootloader *always* detects that the OTA was incomplete (i.e. invalid app)
2. and boots into a different 'safety' app instead.

I have a custom bootloader, so number '2' is no issue, I'm concerned about how reliable number '1' is.
- do I need to enable secure boot to make number '1' reliable?
- or maybe I should add some other partition to my device, i.e. with a "OTA IS IN PROGRESS" flag?

chipweinberger, 2023-08-22:
It looks like in non-secure mode we still check the appended SHA256

```
        // No secure boot, but SHA-256 can be appended for basic corruption detection
        if (sha_handle != NULL && !esp_cpu_dbgr_is_attached()) {
            err = verify_simple_hash(sha_handle, data);
            sha_handle = NULL; // calling verify_simple_hash finishes sha_handle
        }
```

But I do not understand this comment, which seems to contradict the above statement.

```
    // (For non-secure boot, we don't verify any SHA-256 hash appended to the bootloader because
    // esptool.py may have rewritten the header - rely on esptool.py having verified the bootloader at flashing time, instead.)
    verify_sha = (part->offset != ESP_BOOTLOADER_OFFSET) && do_verify;
```

mahavirj, 2023-08-23:
@chipweinberger 

- First of all, if the the OTA update process is interrupted in between then I believe the application won't call the API `esp_ota_set_boot_partition` (?). Even if the application calls this API to set next boot partition then the API first verifies the image integrity and then only updates the `otadata` partition.
- Yes, the application image integrity check happens during the bootup. Typically its the SHA256 checksum, provided the image has the `hash_appended` field set in the image header. Legacy images used to have only the CRC check (very old IDF releases). Please refer to the documentation regarding the format here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html
- In-fact the application integrity check consumes most of the boot-up time and hence we have a config option `BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON` to disable the check (if required)
- 2nd stage bootloader workflow is briefly documented here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/startup.html#second-stage-bootloader

Hope this helps!

chipweinberger, 2023-08-23:
@mahavirj thanks it does help!

closed 2023-08-23 by chipweinberger