espressif/esp-idf #12116: ETS_UNCACHED_ADDR redefined between `rtc_io_reg.h` and `soc.h`

amrsoll, 2023-08-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1 AND master commit: 3247253671

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Target: esp32s3

The permutation of these 2 include statements should be equivalent.

`ulp/ulp_main_A.c`

```c
#include "ulp_riscv_gpio.h"
#include "ulp_riscv_utils.h"

int main(void) { 
  ulp_riscv_gpio_output_level(GPIO_NUM_2, 0);
  return 0;
}
```

`ulp/ulp_main_B.c`

```c
#include "ulp_riscv_utils.h"
#include "ulp_riscv_gpio.h"

int main(void) {
  ulp_riscv_gpio_output_level(GPIO_NUM_2, 0);
  return 0;
}
```

### What is the actual behavior?

at build time, some redeclaration warnings are triggered.

On tag v5.1

`ulp/ulp_main_B.c` No warnings

`ulp/ulp_main_A.c` Causes compilation warning

```
[...]
[1/5] Building C object CMakeFiles/ulp_main.dir/mnt/develdata/home/axel/Spear/main/ulp/ulp_main.c.obj
In file included from /mnt/develdata/home/axel/esp/esp-idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_utils.h:15,
                 from /mnt/develdata/home/axel/Spear/main/ulp/ulp_main.c:6:
/mnt/develdata/home/axel/esp/esp-idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_register_ops.h:29: warning: "ETS_UNCACHED_ADDR" redefined
   29 | #define ETS_UNCACHED_ADDR(addr)     (RISCV_REG_CONV(addr))
      | 
In file included from /mnt/develdata/home/axel/esp/esp-idf/components/soc/esp32s3/include/soc/rtc_io_reg.h:18,
                 from /mnt/develdata/home/axel/esp/esp-idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:13,
                 from /mnt/develdata/home/axel/Spear/main/ulp/ulp_main.c:5:
/mnt/develdata/home/axel/esp/esp-idf/components/soc/esp32s3/include/soc/soc.h:44: note: this is the location of the previous definition
   44 | #define ETS_UNCACHED_ADDR(addr) (addr)
      | 
[...]
```

### Steps to reproduce.

1. `idf.py build` using the example ULP code.

### More Information.

The header file [`esp32s3/include/soc/soc.h:39`](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/components/soc/esp32s3/include/soc/soc.h#L39) checks against `ULP_RISCV_REGISTER_OPS`

```c
//Convenient way to replace the register ops when ulp riscv projects
//consume this file
#ifndef ULP_RISCV_REGISTER_OPS

//Registers Operation {{
#define ETS_UNCACHED_ADDR(addr) (addr)
#define ETS_CACHED_ADDR(addr) (addr)
```

Which is defined in [`components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_register_ops.h`](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_register_ops.h#L8)

```c
// [...]
#pragma once
#define ULP_RISCV_REGISTER_OPS

// [...]

/*
 * When COCPU accesses the RTC register, it needs to convert the access address.
 * When COCPU accesses the RTC memory, dont need to convert the access address.
 */
#define RISCV_REG_CONV(addr)        (((addr&0xffff)<<3 & 0xe000) | (addr & 0x1fff) | 0x8000)
#define ETS_UNCACHED_ADDR(addr)     (RISCV_REG_CONV(addr))
// [...]
```

Given the change of definition of `ETS_UNCACHED_ADDR(addr)` this warning should either be an error (preferably explaining that you need to include `ulp_riscv_utils.h` first) or there shouldn't be 2 definitions for `ETS_UNCACHED_ADDR`. Same goes for a lot of the other macros in both files.

sudeep-mohanty, 2023-08-23:
Thanks for bringing this to our notice @amrsoll. Yes indeed this should be handled in a better way such that the order of inclusion of the headers should not matter. We will work on adding a fix.

amrsoll, 2023-08-23:
Thank you for taking care of it :)

closed 2023-09-04 by espressif-bot