espressif/esp-idf #12118: Unable to connect to the locally built EMQX broker with the IPV4 address

SFNFIH, 2023-08-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dirty

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3 devkitc

### Power Supply used.

USB

### What is the expected behavior?

Successfully connect to the MQTT proxy server at the IPV4 address and subscribe to the topic

### What is the actual behavior?

The code is modified according to the official MQT-TCP demo, only the following part of the code is modified.
Can't connect to MQTT server.
As you can see from the following code snippet, I only set up an IPV4 address. In order to test the code, I also deliberately closed the account login of the MQTT server.

### Steps to reproduce.

static void mqtt_app_start(void)
{
    esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.hostname = "10.168.1.135",
    };
#if CONFIG_BROKER_URL_FROM_STDIN
    char line[128];

    if (strcmp(mqtt_cfg.broker.address.uri, "FROM_STDIN") == 0)
    {
        int count = 0;
        printf("Please enter url of mqtt broker\n");
        while (count < 128)
        {
            int c = fgetc(stdin);
            if (c == '\n')
            {
                line[count] = '\0';
                break;
            }
            else if (c > 0 && c < 127)
            {
                line[count] = c;
                ++count;
            }
            vTaskDelay(10 / portTICK_PERIOD_MS);
        }
        mqtt_cfg.broker.address.uri = line;
        printf("Broker url: %s\n", line);
    }
    else
    {
        ESP_LOGE(TAG, "Configuration mismatch: wrong broker url");
        abort();
    }
#endif /* CONFIG_BROKER_URL_FROM_STDIN */
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
    esp_mqtt_client_start(client);
}

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (467) cpu_start: Pro cpu start user code
I (467) cpu_start: cpu freq: 160000000 Hz
I (467) cpu_start: Application information:
I (472) cpu_start: Project name:     MQTT_MASSTER
I (477) cpu_start: App version:      1
I (482) cpu_start: Compile time:     Aug 23 2023 02:03:23
I (488) cpu_start: ELF file SHA256:  18afdb49ca9817a2...
I (494) cpu_start: ESP-IDF:          v5.1-dirty
I (499) cpu_start: Min chip rev:     v0.0
I (504) cpu_start: Max chip rev:     v3.99 
I (509) cpu_start: Chip rev:         v3.0
I (514) heap_init: Initializing. RAM available for dynamic allocation:
I (521) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (527) heap_init: At 3FFB7DA8 len 00028258 (160 KiB): DRAM
I (533) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (539) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (546) heap_init: At 40095514 len 0000AAEC (42 KiB): IRAM
I (553) spi_flash: detected chip: generic
I (556) spi_flash: flash io: dio
W (560) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (575) app_start: Starting scheduler on CPU0
I (579) app_start: Starting scheduler on CPU1
I (579) main_task: Started on CPU0
I (589) main_task: Calling app_main()
E (589) IP: 0X94,0Xe6,0X86,0X43,0Xc6,0X34
I (639) wifi:wifi driver task: 3ffbfe1c, prio:23, stack:6656, core=0
I (659) wifi:wifi firmware version: b2f1f86
I (659) wifi:wifi certification version: v7.0
I (659) wifi:config NVS flash: enabled
I (659) wifi:config nano formating: disabled
I (659) wifi:Init data frame dynamic rx buffer num: 32
I (659) wifi:Init management frame dynamic rx buffer num: 32
I (669) wifi:Init management short buffer num: 32
I (669) wifi:Init dynamic tx buffer num: 32
I (679) wifi:Init static rx buffer size: 1600
I (679) wifi:Init static rx buffer num: 10
I (689) wifi:Init dynamic rx buffer num: 32
I (689) wifi_init: rx ba win: 6
I (689) wifi_init: tcpip mbox: 32
I (699) wifi_init: udp mbox: 6
I (699) wifi_init: tcp mbox: 6
I (709) wifi_init: tcp tx win: 5744
I (709) wifi_init: tcp rx win: 5744
I (709) wifi_init: tcp mss: 1440
I (719) wifi_init: WiFi IRAM OP enabled
I (719) wifi_init: WiFi RX IRAM OP enabled
I (729) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (829) wifi:mode : sta (94:e6:86:43:c6:34)
I (829) wifi:enable tsf
I (829) WIFI: WIFI启动完成.
I (839) main_task: Returned from app_main()
I (849) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (849) wifi:state: init -> auth (b0)
I (849) wifi:state: auth -> assoc (0)
E (869) wifi:Association refused temporarily, comeback time 1048 mSec
I (1919) wifi:state: assoc -> assoc (0)
I (2919) wifi:state: assoc -> init (200)
I (2929) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2929) MQTT_EXAMPLE: 正在尝试连接WIFI,尝试次数1
I (5339) MQTT_EXAMPLE: 正在尝试连接WIFI,尝试次数2
I (5349) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (5349) wifi:state: init -> auth (b0)
I (5369) wifi:state: auth -> assoc (0)
I (5389) wifi:state: assoc -> run (10)
I (5539) wifi:connected with OrayBox, aid = 10, channel 6, BW20, bssid = a0:c5:f2:bd:ad:76
I (5539) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
I (5549) wifi:pm start, type: 1

I (5579) wifi:<ba-add>idx:0 (ifx:0, a0:c5:f2:bd:ad:76), tid:6, ssn:2, winSize:64
I (5609) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (6549) esp_netif_handlers: sta ip: 10.168.1.206, mask: 255.255.255.0, gw: 10.168.1.1
I (6549) MQTT_EXAMPLE: got ip:10.168.1.206
I (6549) WIFI: WIFI连接成功!
E (6559) mqtt_client: No scheme found
E (6559) mqtt_client: Failed to create transport list
```

SFNFIH, 2023-08-24:
Can anyone respond to me?

euripedesrocha, 2023-08-28:
@SFNFIH you need to set the [URI](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#address) correctly.

You should set it to:

`mqtt://IP_ADDRESS:PORT`

closed 2023-08-28 by euripedesrocha

SFNFIH, 2023-08-28:
@euripedesrocha Yes, I tried your method and it worked, thank you!