espressif/esp-idf #12121: Code stuck in function "xEventGroupWaitBits"

moreiraalexandre, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32 devkit

### Power Supply used.

USB

### What is the expected behavior?

The function:
```
EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);
```
should return after the bits are set. But for some reason it doesn't.

### What is the actual behavior?

*** The code is available on github: [](https://github.com/luanfrj/esp32-pix.git) ***
For some reason I'm having trouble setting some bits in the wifi driver configuration and the code throws errors or doesn't move forward.
I built 3 different scenarios for my code, scenario 1 (original code with no changes) and 2 other scenarios in an attempt to get everything right.
Spoiler: none of them worked.

Scenario 1[](https://youtu.be/1FQRgRpkaoo) : The event `s_wifi_event_group = xEventGroupCreate();` is created in the file "main.c". The device connects to the network and receives an IP address, but when executing the function:

```
EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);
```

the bits were not set, and the error below was generated:

```
"assertion failed: xEventGroupSetBits event_groups.c:559 (xEventGroup)

Backtrace: 0x400819b2:0x3ffc1f10 0x40088c99:0x3ffc1f30 0x4008fc95:0x3ffc1f50 0x4008c04d:0x3ffc2070 0x400db87f:0x3ffc2090 0x401707f5:0 x 3ffc20c0 0x4017034f:0x3ffc20f0 0x40170405:0x3ffc2130 0x4008b619:0x3ffc2150
0x400819b2: panic_abort at C:/Users/morei/esp/esp-idf/components/esp_system/panic.c:452

0x40088c99: esp_system_abort at C:/Users/morei/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4008fc95: __assert_func at C:/Users/morei/esp/esp-idf/components/newlib/assert.c:81
[... 138 lines cut ...]

0x40171650: main_task at C:/Users/morei/esp/esp-idf/components/freertos/app_startup.c:208

0x4008b619: vPortTaskWrapper at C:/Users/morei/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 350a1dad3a3809b3

Rebooting..."
```

### More Information.

The code is available on github: [](https://github.com/luanfrj/esp32-pix.git) 
https://github.com/luanfrj/esp32-pix.git

I added the videos to my YT channel, demonstrating the behavior of the code in the 3 defined scenarios.

Scenario 1 [](https://youtu.be/1FQRgRpkaoo): 
https://youtu.be/1FQRgRpkaoo

Scenario 2 [](https://youtu.be/wfg_RqT--8M):
https://youtu.be/wfg_RqT--8M

Scenario 3 [](https://youtu.be/XpO4eUzfRc0):
https://youtu.be/XpO4eUzfRc0

moreiraalexandre, 2023-08-23:
> s_wifi_event_group is not defined anywhere. This is an undefined variable
> 
> https://github.com/luanfrj/esp32-pix.git

s_wifi_event_group was defined in the file "wifi_station.h"

`#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

/* FreeRTOS event group to signal when we are connected*/
EventGroupHandle_t s_wifi_event_group;

void wifi_init_sta(char* wifi_ssid, char* wifi_password);`

nopnop2002, 2023-08-23:
Try removing the component /wifi_station and including this content in main.c.

moreiraalexandre, 2023-08-23:
> Tente remover o componente /wifi_station e incluir esse conteúdo no main.c.

I was able to get around this problem using the semaphore. Basically, the semaphore is activated when the wifi is successfully connected.

I transferred the entire wifi configuration part of the main file to the "wifi_station" component.

In this way the function

```
EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSO,
            pdFALSO,
            portaMAX_DELAY);
```

returned successfully.

In the main function I changed the condition of

`if ( bits & WIFI_CONNECTED_BIT)`

for

`if ( xSemaphoreTake(connectionWifiSemaphore, portMAX_DELAY)`

I need to do some more tests, for now everything is working fine.

ESP-Marius, 2023-08-29:
Seems like there was no IDF related problems here then, closing the issue.

Thanks to @nopnop2002 for replying!