espressif/esp-idf #12122: idf.py 是怎么做到按tab按键能够子命令补全的

m986883511, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

拜读了一下idf.py，但是没找到哪里实现了子命令补全，我想学习一下这个技术，希望不吝赐教

fhrbata, 2023-08-23:
Hello @m986883511 , 

my understanding is that the shell completion code is generated by `click` https://click.palletsprojects.com/en/8.1.x/shell-completion/ , which is used by `idf.py` , and included in e.g. `export.sh`  with `__enable_autocomplete()`. The completion is handled by shell. HTH

m986883511, 2023-08-24:
> 

very thank you

closed 2023-08-24 by m986883511