espressif/esp-idf #12123: Pre Encrypted BIN file not including version.txt app version

law-ko, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

With using `version.txt` the secure bin file created with `create_esp_enc_img(${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}.bin
    ${project_dir}/main/encrypted_ota/rsa_key/private.pem ${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}_secure.bin app)` should use the app version listed in `version.txt`

### What is the actual behavior?

The created `${CMAKE_PROJECT_NAME}_secure.bin` uses the default APP VERSION 1.0

### Steps to reproduce.

Use `pre_encrypted_ota` example with creating `version.txt` in the project directory.

hmalpani, 2023-08-24:
Hello @law-ko 
I tested the pre_encrypted_ota example and the `${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}_secure.bin` app has the same version as that of `${CMAKE_BINARY_DIR}/${CMAKE_PROJECT_NAME}.bin` app. So if you have added version.txt in the example folder, the application will build with the version mentioned in the version.txt file.

closed 2023-08-25 by law-ko

mahdidadashi65, 2024-01-01:
It seems that there is a bug in create_esp_enc_img that causes the _secure.bin output to not update. If you first delete the _secure.bin file manually and then build the program, the operation will be done correctly.

valkuc, 2024-05-02:
> It seems that there is a bug in create_esp_enc_img that causes the _secure.bin output to not update. If you first delete the _secure.bin file manually and then build the program, the operation will be done correctly.

Same for me