espressif/esp-idf #12124: Ble Gatt Server Example Memory Leak When Connect/Disconnect

HasanYektaDemirtas, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3 and v4.3.5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 DEVKIT V1, Custom board with ESP32-WROVER

### Power Supply used.

USB

### What is the expected behavior?

Having the same free ram size after connecting and disconnecting the phone to ESP32.

### What is the actual behavior?

I get lesser free ram size after connecting and disconnecting.

### Steps to reproduce.

1. Step: Start new project with "gatt_server" example
2. Step: Add while at the end of the main function and measure free internal ram size periodically
3. Step: Open "nRF Connect" app on android
4. Step: Find "ESP_GATTS_DEMO" device
5. Step: Connect to device
6. Step: Wait for 2-6 seconds
8. Step: Disconnect from device
9. Step: Repeat steps from 5.step a few times

### Debug Logs.

```plain
I (27) boot: ESP-IDF v5.0.3-dirty 2nd stage bootloader
I (27) boot: compile time 15:01:11       
I (27) boot: Multicore bootloader        
I (30) boot: chip revision: v1.0
I (34) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ee48h (126536) map
I (145) esp_image: segment 1: paddr=0002ee70 vaddr=3ffbdb60 size=011a8h (  4520) load
I (147) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7e9b0h (518576) map
I (338) esp_image: segment 3: paddr=000ae9d8 vaddr=3ffbed08 size=030c4h ( 12484) load
[... 172 lines cut ...]
I (34662) GATTS_DEMO: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 500
I (34832) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (34832) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (34902) GATTS_DEMO: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 24,latency = 0, timeout = 400
I (35102) app_main: Free Internal Ram Size: 237100

I (36102) app_main: Free Internal Ram Size: 237100

W (37002) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (37002) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (37102) app_main: Free Internal Ram Size: 239208

I (38102) app_main: Free Internal Ram Size: 239208

I (39102) app_main: Free Internal Ram Size: 239208

I (40102) app_main: Free Internal Ram Size: 239208
```

### More Information.

I was working on a project and i realised memory leak happens when i connect and disconnect with my phone.
To generate this problem, i created a ble gatt server example. The only change i made is printing free internal ram space.

Any idea?

zhp0406, 2023-08-23:
@HasanYektaDemirtas 
thanks for report the bug,Can you provide the code of your GATT server example?

HasanYektaDemirtas, 2023-08-23:
@zhp0406 

[gatts_demo.txt](https://github.com/espressif/esp-idf/files/12418138/gatts_demo.txt)

zhp0406, 2023-08-23:
@HasanYektaDemirtas 
It's worth noting that a slight decrease in memory after the initial connection and disconnection is a normal occurrence, as long as it's not continuously decreasing.

HasanYektaDemirtas, 2023-08-23:
@zhp0406 
Did you change anything in the code? Or is your ESP IDF version same with mine? Your result is okay because i can see free ram size stay at "228176" after disconnected. But in my case it decrease every time. And i see"update connection params status" print only one time after connect event triggered in your test result. But in my test, it show up 3 times. I don't know what it is but something is different between our tests.

zhp0406, 2023-08-23:
@HasanYektaDemirtas 
In the lasted log，I used v5.0.3 and your code （without any change）, and no Memory Leak found,

After multiple rounds of connecting and disconnecting, the remaining memory eventually stabilizes at 229,116. Upon reconnecting and disconnecting again, there's no significant change in the remaining memory, as expected. I have updated the latest log.

HasanYektaDemirtas, 2023-08-23:
@zhp0406 
I suspect about if problem caused by my development environment. I will try same process on another laptop and share result here.

HasanYektaDemirtas, 2023-08-23:
@zhp0406 Can you repeat test with not ESP32C3 but ESP32?

[image]

zhp0406, 2023-08-23:
@HasanYektaDemirtas I seems also ok:

I (27) boot: ESP-IDF v5.0.3-dirty 2nd stage bootloader
I (27) boot: compile time 20:17:25
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ee58h (126552) map
I (146) esp_image: segment 1: paddr=0002ee80 vaddr=3ffbdb60 size=01198h (  4504) load
I (148) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7e9b0h (518576) map
I (339) esp_image: segment 3: paddr=000ae9d8 vaddr=3ffbecf8 size=03104h ( 12548) load
I (344) esp_image: segment 4: paddr=000b1ae4 vaddr=40080000 size=17bd0h ( 97232) load
I (396) boot: Loaded app from partition at offset 0x10000
I (397) boot: Disabling RNG early entropy source...
I (408) cpu_start: Multicore app
I (408) cpu_start: Pro cpu up.
I (408) cpu_start: Starting app cpu, entry point is 0x400812c0
0x400812c0: call_start_cpu1 at /home/zhanghaipeng/esp/esp-idf/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (426) cpu_start: Pro cpu start user code
I (426) cpu_start: cpu freq: 160000000 Hz
I (426) cpu_start: Application information:
I (431) cpu_start: Project name:     gatt_server_demos
I (437) cpu_start: App version:      v5.0.3-dirty
I (442) cpu_start: Compile time:     Aug 23 2023 20:17:19
I (448) cpu_start: ELF file SHA256:  0f4132b4ca2060fd...
I (454) cpu_start: ESP-IDF:          v5.0.3-dirty
I (460) cpu_start: Min chip rev:     v0.0
I (464) cpu_start: Max chip rev:     v3.99 
I (469) cpu_start: Chip rev:         v3.1
I (474) heap_init: Initializing. RAM available for dynamic allocation:
I (481) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (487) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (493) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (500) heap_init: At 3FFC6660 len 000199A0 (102 KiB): DRAM
I (506) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (512) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (519) heap_init: At 40097BD0 len 00008430 (33 KiB): IRAM
I (526) spi_flash: detected chip: generic
I (530) spi_flash: flash io: dio
W (533) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (548) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (596) BTDM_INIT: BT controller compile version [ad2b7cd]
I (596) system_api: Base MAC address is not set
I (596) system_api: read default base MAC address from EFUSE
I (606) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (606) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (1086) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (1096) app_main: Free Internal Ram Size: 242812

I (1106) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (1106) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (1116) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (1116) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

[... 90 lines cut ...]
I (24096) app_main: Free Internal Ram Size: 238872

W (24096) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (24096) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (24766) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (24776) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (24846) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 16,latency = 0, timeout = 400
I (25096) app_main: Free Internal Ram Size: 238824

I (26096) app_main: Free Internal Ram Size: 238872

W (26736) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (26736) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (27096) app_main: Free Internal Ram Size: 240984

I (27406) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (27406) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (27486) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 16,latency = 0, timeout = 400
I (28096) app_main: Free Internal Ram Size: 238872

W (28766) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (28766) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (29096) app_main: Free Internal Ram Size: 240984

I (30096) app_main: Free Internal Ram Size: 240984

HasanYektaDemirtas, 2023-08-23:
@zhp0406 Your chip revision is 3.1 and mine is 1.0
Do you think that can cause of my problem?
And to be sure, can you repeat test with ESP 4.3.5?
I installed 4.3.5 first and installed 5.0.3 after that. Even if i changed configuration to use 5.0.3, libraries might be using 4.3.5
Is there a way to check it?

HasanYektaDemirtas, 2023-08-23:
Tomorrow, I am going to install vs code and esp idf 5.0.3 on a new formatted laptop to be sure about if the problem caused by environment

zhp0406, 2023-08-24:
@HasanYektaDemirtas 
It is ok（v4.3.5 esp32）：

I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 10:53:12
I (27) boot: chip revision: v3.1
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18654h ( 99924) map
I (132) esp_image: segment 1: paddr=0002867c vaddr=3ffbdb60 size=03464h ( 13412) load
I (138) esp_image: segment 2: paddr=0002bae8 vaddr=40080000 size=04530h ( 17712) load
I (145) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=5cb60h (379744) map
I (284) esp_image: segment 4: paddr=0008cb88 vaddr=40084530 size=1344ch ( 78924) load
I (328) boot: Loaded app from partition at offset 0x10000
I (328) boot: Disabling RNG early entropy source...
I (339) cpu_start: Pro cpu up.
I (340) cpu_start: Starting app cpu, entry point is 0x40081220
0x40081220: call_start_cpu1 at /home/zhanghaipeng/esp_v4/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (354) cpu_start: Pro cpu start user code
I (354) cpu_start: cpu freq: 160000000
I (354) cpu_start: Application information:
I (358) cpu_start: Project name:     gatt_server_demos
I (364) cpu_start: App version:      v4.3.5-dirty
I (370) cpu_start: Compile time:     Aug 24 2023 10:53:07
I (376) cpu_start: ELF file SHA256:  e2ba99b5d961c915...
I (382) cpu_start: ESP-IDF:          v4.3.5-dirty
I (387) cpu_start: Min chip rev:     v0.0
I (392) cpu_start: Max chip rev:     v3.99 
I (397) cpu_start: Chip rev:         v3.1
I (401) heap_init: Initializing. RAM available for dynamic allocation:
I (409) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (414) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (421) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (427) heap_init: At 3FFC4D48 len 0001B2B8 (108 KiB): DRAM
I (433) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (439) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (446) heap_init: At 4009797C len 00008684 (33 KiB): IRAM
I (453) spi_flash: detected chip: generic
I (457) spi_flash: flash io: dio
W (461) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (475) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (522) BTDM_INIT: BT controller compile version [2c56073]
I (522) system_api: Base MAC address is not set
I (522) system_api: read default base MAC address from EFUSE
I (532) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (932) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (942) app_main: Free Internal Ram Size: 245528

I (942) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (942) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (962) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (962) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (972) GATTS_DEMO: the gatts demo char length = 3

[... 111 lines cut ...]
I (30642) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 16,latency = 0, timeout = 400
I (30942) app_main: Free Internal Ram Size: 241712

I (31942) app_main: Free Internal Ram Size: 241712

I (32942) app_main: Free Internal Ram Size: 241712

W (33312) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (33312) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (33942) app_main: Free Internal Ram Size: 243832

I (34272) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (34272) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote df:f3:be:cf:80:ec:
I (34342) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 16,latency = 0, timeout = 400
I (34942) app_main: Free Internal Ram Size: 241712

W (35932) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (35932) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (35942) app_main: Free Internal Ram Size: 243652

I (36942) app_main: Free Internal Ram Size: 243832

I (37942) app_main: Free Internal Ram Size: 243832

Done

zhp0406, 2023-08-24:
@HasanYektaDemirtas 
Hi,
Is the issue on your side still present? Have you tested it using the new environment?

HasanYektaDemirtas, 2023-08-24:
@zhp0406 I am not able to test it with new environment today. I will try and give feedback to you next week;

zhp0406, 2023-08-31:
@HasanYektaDemirtas 
Hello, is there any feedback available?

HasanYektaDemirtas, 2023-08-31:
I can't install new environment to fresh formatted laptop. I get error when i try install esp idf with vs code extension and espressif ide. the error is related to the error in the link below. 
https://github.com/espressif/esp-idf/issues/7672

I think Espressif changing somethings lately without test enough and this causes many errors. I was suggesting using esp32 to everyone but i am not so sure anymore.

HasanYektaDemirtas, 2023-09-01:
@zhp0406 I have just opened issue due to installation error. I will test memory leak problem after that.
https://github.com/espressif/esp-idf/issues/12175#issue-1876980498

crispmfc, 2023-09-01:
> @zhp0406 I can't install new environment to fresh formatted laptop. I get error when i try install esp idf with vs code extension and espressif ide. the error is related to the error in the link below. #7672
> 
> I think Espressif changing somethings lately without test enough and this causes many errors. I was suggesting using esp32 to everyone but i am not so sure anymore.

I had problems like that and I solved them, so I have responded to the [issue](https://github.com/espressif/esp-idf/issues/12175#issue-1876980498) with the solutions.

zhp0406, 2023-09-11:
@crispmfc 
Have you resolved your issue?

HasanYektaDemirtas, 2023-09-11:
@zhp0406 I installed idf v5.0.3 with vs code extension on another new formatted laptop today and ram reduction stopped after 7-8th connect/disconnect cycle. Probably problem caused by my environment. I will format my laptop and make a clear install. We can close this issue.

zhp0406, 2023-09-11:
@HasanYektaDemirtas 
ok,thank you. If there are no further questions, you can close this issue.

closed 2023-09-11 by HasanYektaDemirtas