espressif/esp-idf #12127: Using ESP-PROG in a macOS + Docker Environment

imammy-hacomono, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am attempting to develop using ESP-IDF in a Docker environment. 
As I am using a Mac, I am aware that passing the ESP-PROG serial device (for example: /dev/tty.usbserial-14101) to the container and performing a docker run is not possible due to macOS limitations.

Therefore, even though there is an explanation regarding the specifications of the remote serial port here, it is not functioning as expected.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html#using-remote-serial-port

~~~
$ esp_rfc2217_server.py -v -p 4000 /dev/tty.usbserial-14101

INFO:root:RFC 2217 TCP/IP to Serial redirector - type Ctrl-C / BREAK to quit
INFO:root:Serving serial port: /dev/tty.usbserial-14101
INFO:root:TCP/IP port: 4000
~~~

~~~
$ docker run --rm -it --privileged -v $PWD:/project -v /dev:/dev -w /project espressif/idf:release-v5.1  idf.py --port rfc2217://host.docker.internal:4000?ign_set_control flash                                                                

zsh: no matches found: rfc2217://host.docker.internal:4000?ign_set_control
~~~

 The message "no matches found" is being displayed.

imammy-hacomono, 2023-08-24:
Hello @fhrbata .

Thanks to your assistance, I was able to make some progress.
https://github.com/espressif/esp-idf/issues/12060#issuecomment-1680574816

However, when I try to run idf.py monitor, I encounter the following error:

~~~
$ docker run --rm -it --privileged -v $PWD:/project -v /dev:/dev -w /project espressif/idf:release-v5.1 idf.py monitor --port 'rfc2217://host.docker.internal:4000?ign_set_control'

Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin
  /opt/esp/python_env/idf5.1_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Executing action: monitor
Running idf_monitor in directory /project
Executing "/opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p rfc2217://host.docker.internal:4000?ign_set_control -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /project/build/iot-device-controller-fw-v3.elf -m '/opt/esp/python_env/idf5.1_py3.8_env/bin/python' '/opt/esp/idf/tools/idf.py'"...
--- idf_monitor on rfc2217://host.docker.internal:4000?ign_set_control 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/esp_idf_monitor/base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/esp_idf_monitor/base/serial_reader.py", line 55, in run
    self.serial.open()
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/serial/rfc2217.py", line 487, in open
    self._reconfigure_port()
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/serial/rfc2217.py", line 508, in _reconfigure_port
    raise NotImplementedError('write_timeout is currently not supported')
NotImplementedError: write_timeout is currently not supported
~~~

fhrbata, 2023-08-24:
Hello @imammy-hacomono , 

this problem was probably introduced by esp-idf-monitor commit https://github.com/espressif/esp-idf-monitor/commit/10d03b497c475410da972713e70561cfacc04c33, where the write_timeout was added. It seems that pyserial, which is internally used by esp-idf-monitor, does not support write timeout when rfc2217 protocol is used.

Note that the latest docker image `espressif/idf` works, at least for me, because it contains older `esp-idf-monitor 1.1.1`, which does not have this commit. On the other hand `espressif/idf:release-v5.1` contains `esp-idf-monitor 1.1.3`, which contains this change.

I think as a workaround you can run the esp-idf-monitor directly from host, unless there is some reason it has to be started from within the docker.

I'm adding @peterdragun  so he can take a look.

fhrbata, 2023-08-24:
Just one more note. If I connect esp-idf-monitor through rfc2217, the chip seems to be in download mode, so I needed to press reset to get the app output. Maybe some initialization in rfc2217 is causing this, but I haven't looked any closer.