espressif/esp-idf #12128: 使用BLE_GATT进行数据读取，write某项attr的时候提示报错write char failed, error status = 5

zhoubonan, 2023-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

正常情况下，在gattc端触发ESP_GATTC_WRITE_CHAR_EVT回调后，收到ESP_GATT_OK的状态应答

### What is the actual behavior?

实际情况，在gattc端触发ESP_GATTC_WRITE_CHAR_EVT回调后
收到status == 5的报错状态，也就是 ESP_GATT_INSUF_AUTHENTICATION

### Steps to reproduce.

1、server和client端初始化
2、两边SMP配置交换认证方式，均配置使能OOB
3、使用相同OOB密钥，观察打印并正常完成 配对绑定等过程
4、Client端进行read和write操作测试
5、read读取数据正常
6、write操作提示报错
Client端调用情况如下：
    esp_ble_gattc_write_char(   gattc_num,
                                gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                handle,
                                strlen((const char *)write_char_data),
                                write_char_data,
                                ESP_GATT_WRITE_TYPE_RSP,
                                ESP_GATT_AUTH_REQ_NONE);   

Server端该handle对应的attr权限配置如下：

ESP_GATT_PERM_WRITE_ENCRYPTED | ESP_GATT_PERM_READ_ENCRYPTED,

zhoubonan, 2023-08-23:
Client端输出：
E (11671) GATTC_DEMO: write char failed, error status = 5

zhoubonan, 2023-08-23:
> @zhoubonan 感谢报告问题，可以给出完整的 log 吗？

好的。删除部分应用端打印，应不影响观察蓝牙的状态
其中可以看到Client的read操作正常，Server端触发ESP_GATTS_READ_EVT事件；而Client的write操作报错
帮忙看看，感谢 @zhp0406 

**Client端 log 如下**：

I (0) cpu_start: Starting scheduler on APP CPU.
W (411) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (411) BT_INIT: BT controller compile version [05195c9]
I (411) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (461) system_api: Base MAC address is not set
I (461) system_api: read default base MAC address from EFUSE
I (461) BT_INIT: Bluetooth MAC: 34:85:18:bd:2d:02

I (491) GATTC_DEMO: REG_EVT
▒

letter:/$ I (1111) GATTC_DEMO: Searched Adv Data Len 26, Scan Response Len 26
I (1111) GATTC_DEMO: Searched Device Name Len :14 , name:
I (1121) GATTC_DEMO: ZBN_GATTS_DEMO
I (1121) GATTC_DEMO: Device bda :
I (1121) GATTC_DEMO: 48 27 e2 17 78 62
I (1131) GATTC_DEMO:

I (1131) GATTC_DEMO: connect to the remote device.
I (1141) GATTC_DEMO: stop scan successfully
I (1191) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (1191) GATTC_DEMO: REMOTE BDA:
I (1191) GATTC_DEMO: 48 27 e2 17 78 62
I (1201) GATTC_DEMO: open success
I (1221) GATTC_DEMO: ESP_GAP_BLE_SEC_REQ_EVT
W (1221) BT_SMP: Non bonding: No keys will be exchanged
E (1261) BT_SMP: Association Model = SMP_MODEL_OOB

I (1261) GATTC_DEMO: ESP_GAP_BLE_OOB_REQ_EVT
oob value = 23 ea f f8 d4 ae f5 5f 22 1e 76 59 8d 13 54 35
I (1501) GATTC_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (1671) GATTC_DEMO: ESP_GAP_BLE_AUTH_CMPL_EVT
I (1671) GATTC_DEMO: remote BD_ADDR: 4827e2177862
I (1671) GATTC_DEMO: address type = 0
I (1671) GATTC_DEMO: pair status = success
I (1681) GATTC_DEMO: auth mode = ESP_LE_AUTH_REQ_MITM
I (2391) GATTC_DEMO: DIS_SRVC_CMPL_EVT
I (2391) GATTC_DEMO: discover service complete conn_id 0
I (2391) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (2391) GATTC_DEMO: start handle 40 end handle 65535 current handle value 40
I (2401) GATTC_DEMO: service found
I (2411) GATTC_DEMO: UUID16: ff
I (2411) GATTC_DEMO: Get service information from remote device
I (2421) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (2421) GATTC_DEMO: attr count 3
I (2431) GATTC_DEMO: notify reg, attr handle 46
I (2431) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (2441) GATTC_DEMO: write ccd, handle 47
I (2471) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 250, conn_id 0
I (2631) GATTC_DEMO: ESP_GATTC_NOTIFY_EVT, receive notify value:
I (2631) GATTC_DEMO: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e
I (2631) GATTC_DEMO: write descr success

letter:/$
letter:/$ write 42 "shfurgyuvrtb"
input handle 42; service_start_handle: 40
Return: 1, 0x00000001

letter:/$ E (11671) GATTC_DEMO: write char failed, error status = 5
letter:/$
letter:/$ read 42
input handle 42; service_start_handle: 40
Return: 1, 0x00000001

letter:/$ I (1688471) GATTC_DEMO: ESP_GATTC_READ_CHAR_EVT,handle 42 conn id 0
I (1688471) GATTC_DEMO: read value:
I (1688471) GATTC_DEMO:  "data":{"WorkMo
I (1688471) GATTC_DEMO: de":"0","Inverte
I (1688481) GATTC_DEMO: r":"1","router":
I (1688481) GATTC_DEMO: "1","mqtt":"1"},

**Server端Log如下**：

I (332) sleep: Enable automatic switching of GPIO sleep configuration
I (339) esp_core_dump_uart: Init core dump to UART
I (345) coexist: coexist rom version e7ae62f
I (350) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
&▒

letter:/$ I (418) BLE_MOD: bt module external cb reg success 1
I (418) BLE_MOD: bt module external cb reg success 2
I (428) BLE_MOD: bt module external cb reg success 3
W (448) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (448) BT_INIT: BT controller compile version [05195c9]
I (458) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (508) system_api: Base MAC address is not set
I (508) system_api: read default base MAC address from EFUSE
I (508) BT_INIT: Bluetooth MAC: 48:27:e2:17:78:62

I (528) BLE_MOD: BD ADDR: 48:27:e2:17:78:62

I (528) BLE_MOD: module smp setting

I (538) BLE_MOD: [GATTS]1 - ESP_GATTS_REG_EVT
I (538) BLE_MOD:     create attr table pass!
I (548) BLE_MOD: [GATTS]2 - ESP_GATTS_CREAT_ATTR_TAB_EVT
I (548) BLE_MOD:     create attribute table successfully, the number handle = 8

I (558) BLE_MOD: [GAP]1 - SET_LOCAL_PRIVACY_COMPLETE_EVT
I (558) BLE_MOD: [GATTS]3 - SERVICE_START_EVT
I (568) BLE_MOD:     status 0, service_handle 40
I (568) BLE_MOD: [GAP]2 - ADV_DATA_SET_COMPLETE_EVT
I (578) BLE_MOD: [GAP]3 - SCAN_RSP_DATA_SET_COMPLETE_EVT
I (588) BLE_MOD: [GAP]4 - ADV_START_COMPLETE_EVT
I (588) BLE_MOD:     advertising start successfully
I (598) BLE_MOD:     bt ble module start success!
I (758) BLE_MOD: [GATTS]4 - ESP_GATTS_CONNECT_EVT
I (758) BLE_MOD:     conn_id = 0
I (758) BLE_MOD: 34 85 18 bd 2d 02
I (808) BLE_MOD: [GAP]5 - OOB_REQ_EVT
I (808) BLE_MOD:    reply oob key,value:
I (808) BLE_MOD: 23 ea 0f f8 d4 ae f5 5f 22 1e 76 59 8d 13 54 35
I (1068) BLE_MOD: [GAP]6 - GAP_BLE_UPDATE_CONN_PARAMS
I (1068) BLE_MOD:    update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (1188) BLE_MOD: [GAP]7 - ESP_GAP_BLE_AUTH_CMPL_EVT
I (1188) BLE_MOD:     remote BD_ADDR: 348518bd2d02
I (1188) BLE_MOD:     address type = 0
I (1198) BLE_MOD:     pair status = success
I (1198) BLE_MOD:     auth mode = ESP_LE_AUTH_REQ_MITM
I (1208) BLE_MOD: Bonded devices number : 0
I (1208) BLE_MOD: Bonded devices list : 0

I (1988) BLE_MOD: [GATTS]5 - ESP_GATTS_MTU_EVT MTU 250
I (2148) BLE_MOD: [GATTS]6 - ESP_GATTS_WRITE_EVT
I (2148) BLE_MOD:    handle = 47, value len = 2, need_rsp = 0 value:
I (2148) BLE_MOD:
I (2158) BLE_MOD:    notify enable
I (2158) BLE_MOD: [GATTS]7 - ESP_GATTS_CONF_EVT
I (2168) BLE_MOD:    status = 0, attr_handle 46

letter:/$
letter:/$ I (1687968) BLE_MOD: [GATTS]8 - ESP_GATTS_READ_EVT
I (1687968) BLE_MOD:     conn_id 0,handle 42
acquire data: "data"

zhp0406, 2023-08-23:
@zhoubonan 
log 中 ESP_LE_AUTH_REQ_MITM 代表需要更高级别的加密，你的加密等级应该不够吧？加密等级是由 ESP_BLE_SM_IOCAP_MODE 和 ESP_BLE_SM_AUTHEN_REQ_MODE 决定的。可以使用 API esp_ble_gap_set_security_param 来设置。

为了进一步验证我的猜想，server 端的 log 也提供一下吧，并使用 esp_ble_gatts_show_local_database 展示一下你的 database （或者你提供一下 server 端 database 的代码），谢谢

zhoubonan, 2023-08-24:
@zhp0406 你好，关于更高级别的加密，从client和server的log可以看出加密级别均为  ESP_LE_AUTH_REQ_MITM 
server端的Log如下，以及server端database部分如下：

**Server端database如下**：
static const esp_gatts_attr_db_t bt_module_gatt_db[BT_MODULE_HRS_IDX_NB] =
{
    /* Service Declaration */
    [BT_MODULE_IDX_SVC] =
    {
        {ESP_GATT_AUTO_RSP},   
        {ESP_UUID_LEN_16, (uint8_t *)&uuid_tpye_val.gatts_uuid_pri_service,         //UUID Length,Value
         ESP_GATT_PERM_READ,                                 //Permission
         sizeof(uint16_t), sizeof(uuid_info.gatts_primary_service_uuid),  //Element Max,Current Length
         (uint8_t *)&uuid_info.gatts_primary_service_uuid}                //Element value
    },

    /*******************************
    *    General   Charateristic
    ********************************/  

    /* Router Characteristic */
    //Declaration
    [GN_CHAR_ROUTER_DECLA] =
    {
        {ESP_GATT_AUTO_RSP}, 
        {ESP_UUID_LEN_16, (uint8_t *)&uuid_tpye_val.gatts_uuid_char_declare, 
         ESP_GATT_PERM_READ,
         sizeof(uint8_t), sizeof(uint8_t), 
         (uint8_t *)&char_perm.rw }
    },
    // Value 
    [GN_CHAR_ROUTER_VAL] =
    {
        {ESP_GATT_RSP_BY_APP}, 
        {ESP_UUID_LEN_16, (uint8_t *)&uuid_info.gatts_char_router_info,      
         ESP_GATT_PERM_WRITE_ENCRYPTED | ESP_GATT_PERM_READ_ENCRYPTED,
         BT_MODULE_CHAR_VAL_LEN_MAX, sizeof(byRouterInfo), 
         (uint8_t *)&byRouterInfo[0]}
    },
}

**Server端Log如下**：

I (332) sleep: Enable automatic switching of GPIO sleep configuration
I (339) esp_core_dump_uart: Init core dump to UART
I (345) coexist: coexist rom version e7ae62f
I (350) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
&▒

letter:/$ I (418) BLE_MOD: bt module external cb reg success 1
I (418) BLE_MOD: bt module external cb reg success 2
I (428) BLE_MOD: bt module external cb reg success 3
W (448) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (448) BT_INIT: BT controller compile version [05195c9]
I (458) phy_init: phy_version 503,13653eb,Jun 1 2022,17:47:08
I (508) system_api: Base MAC address is not set
I (508) system_api: read default base MAC address from EFUSE
I (508) BT_INIT: Bluetooth MAC: 48:27:e2:17:78:62

I (528) BLE_MOD: BD ADDR: 48:27:e2:17:78:62

I (528) BLE_MOD: module smp setting

I (538) BLE_MOD: [GATTS]1 - ESP_GATTS_REG_EVT
I (538) BLE_MOD: create attr table pass!
I (548) BLE_MOD: [GATTS]2 - ESP_GATTS_CREAT_ATTR_TAB_EVT
I (548) BLE_MOD: create attribute table successfully, the number handle = 8

I (558) BLE_MOD: [GAP]1 - SET_LOCAL_PRIVACY_COMPLETE_EVT
I (558) BLE_MOD: [GATTS]3 - SERVICE_START_EVT
I (568) BLE_MOD: status 0, service_handle 40
I (568) BLE_MOD: [GAP]2 - ADV_DATA_SET_COMPLETE_EVT
I (578) BLE_MOD: [GAP]3 - SCAN_RSP_DATA_SET_COMPLETE_EVT
I (588) BLE_MOD: [GAP]4 - ADV_START_COMPLETE_EVT
I (588) BLE_MOD: advertising start successfully
I (598) BLE_MOD: bt ble module start success!
I (758) BLE_MOD: [GATTS]4 - ESP_GATTS_CONNECT_EVT
I (758) BLE_MOD: conn_id = 0
I (758) BLE_MOD: 34 85 18 bd 2d 02
I (808) BLE_MOD: [GAP]5 - OOB_REQ_EVT
I (808) BLE_MOD: reply oob key,value:
I (808) BLE_MOD: 23 ea 0f f8 d4 ae f5 5f 22 1e 76 59 8d 13 54 35
I (1068) BLE_MOD: [GAP]6 - GAP_BLE_UPDATE_CONN_PARAMS
I (1068) BLE_MOD: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (1188) BLE_MOD: [GAP]7 - ESP_GAP_BLE_AUTH_CMPL_EVT
I (1188) BLE_MOD: remote BD_ADDR: 348518bd2d02
I (1188) BLE_MOD: address type = 0
I (1198) BLE_MOD: pair status = success
I (1198) BLE_MOD: auth mode = ESP_LE_AUTH_REQ_MITM
I (1208) BLE_MOD: Bonded devices number : 0
I (1208) BLE_MOD: Bonded devices list : 0

I (1988) BLE_MOD: [GATTS]5 - ESP_GATTS_MTU_EVT MTU 250
I (2148) BLE_MOD: [GATTS]6 - ESP_GATTS_WRITE_EVT
I (2148) BLE_MOD: handle = 47, value len = 2, need_rsp = 0 value:
I (2148) BLE_MOD:
I (2158) BLE_MOD: notify enable
I (2158) BLE_MOD: [GATTS]7 - ESP_GATTS_CONF_EVT
I (2168) BLE_MOD: status = 0, attr_handle 46

letter:/$
letter:/$ I (1687968) BLE_MOD: [GATTS]8 - ESP_GATTS_READ_EVT
I (1687968) BLE_MOD: conn_id 0,handle 42
acquire data: "data"

zhoubonan, 2023-08-24:
@zhp0406 你好，下面是Client端和Server端 SMP的配置代码，应可以看出两端是一致的：

Client端SMP配置：
`
    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM;		//使用MITM保护和未启用连接的安全连接
    esp_ble_io_cap_t iocap = ESP_IO_CAP_KBDISP;
    uint8_t key_size = 16;      //the key size should be 7~16 bytes
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
     uint8_t oob_support = ESP_BLE_OOB_ENABLE;			//开启OOB
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

`

Server端SMP配置如下：

`
    //set static passkey
    uint32_t passkey = 123456;
    //set auth_req
    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM;     //bonding 
    esp_ble_io_cap_t iocap = ESP_IO_CAP_KBDISP;                         
    //set max key size 
    uint8_t key_size = 16;                                           
    //发起者密钥 
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    //响应者密钥
    uint8_t rsp_key  = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    //关闭仅接收指定的SMP身份验证要求
    uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE; //必须绑定才能开启
    //关闭支持OOB 带外数据
    uint8_t oob_support = ESP_BLE_OOB_ENABLE;  

    //设置静态密钥
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
    //设置认证要求 - 身份验证后与对方设备绑定
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    //设置IO能力 
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    //关闭仅接收指定的SMP身份验证要求
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    //关闭支持OOB 带外数据
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t)); 
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t)); 

`

zhp0406, 2023-08-24:
@zhoubonan 
看着好像没什么问题，你把 ESP_LE_AUTH_REQ_SC_MITM 改为 ESP_LE_AUTH_REQ_MITM 或者ESP_LE_AUTH_BOND 试一下问题还会出现吗？
如果问题还是出现，你可以把你的整个代码打包发给我，我使用你的代码复现问题。

zhoubonan, 2023-08-29:
@zhp0406 你好，进一步的测试，其余条件不变的情况下：
① 当配置OOB使能，且 auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND 的情况下：read 和 write操作均正常；
② 当配置OOB使能，且 auth_req = ESP_LE_AUTH_REQ_SC_MITM 或者 ESP_LE_AUTH_REQ_MITM 的情况下：
      read 正常，write操作报错（status == 5状态，也就是 ESP_GATT_INSUF_AUTHENTICATION）

后续我这边的开发应用场景，使用 auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND 这种带绑定的配置是可以的，也算是解决了我这边的后续开发问题。
但上述情况②中的现象，不知道是算正常的 还是 一个问题。

zhp0406, 2023-08-29:
@zhoubonan 
为了不影响你的开发，你可以先使用 ① 当配置OOB使能，且 auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND 

根据你的描述，这应该是一个 bug，我们会尝试复现该问题，如果确实有问题我们会进行修复。如果你再后续的使用过程中发现了该问题的其他信息，麻烦告诉一下我。

weiyuhannnn, 2024-07-01:
@zhoubonan Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates

closed 2024-07-01 by weiyuhannnn