espressif/esp-idf #12132: Add sockfd field to esp_https_server_user_cb_arg_t

dannybackx, 2023-08-23:
### Is your feature request related to a problem?

I have trouble putting the esp_https_server to use with authenticated clients.
Previous remarks I made resulted in adding functionality, however one of the fields I suggested was dropped. See (IDFGH-6652) #8288.

The user_cb is good to pass me the TLS info, but if I want to match that later on (in handlers of the HTTPS server), I don't seem to know which session I'm in.

The socket appears like a good handle for that. It was in my sample code but got dropped, I am asking again to include it.

### Describe the solution you'd like.

hp: {1564} diff -c include/esp_https_server.h.orig include/esp_https_server.h ; diff -c src/https_server.c.orig src/https_server.c
*** include/esp_https_server.h.orig     2023-08-23 04:36:08.000000000 +0200
--- include/esp_https_server.h  2023-08-23 17:34:15.950839687 +0200
***************
*** 36,41 ****
--- 36,42 ----
   */
  typedef struct esp_https_server_user_cb_arg {
      httpd_ssl_user_cb_state_t user_cb_state; /*!< State of user callback */
+     int sockfd;                               /*!< Socket of the https connection */
      esp_tls_t *tls;                    /*!< ESP-TLS connection handle */
  } esp_https_server_user_cb_arg_t;
  
*** src/https_server.c.orig     2023-08-23 04:36:08.000000000 +0200
--- src/https_server.c  2023-08-23 17:35:10.566596608 +0200
***************
*** 40,45 ****
--- 40,46 ----
          esp_https_server_user_cb_arg_t user_cb_data = {0};
          user_cb_data.user_cb_state = HTTPD_SSL_USER_CB_SESS_CLOSE;
          user_cb_data.tls = tls;
+       user_cb_data.sockfd = -1;
          (global_ctx->user_cb)((void *)&user_cb_data);
      }
  
***************
*** 158,163 ****
--- 159,165 ----
          esp_https_server_user_cb_arg_t user_cb_data = {0};
          user_cb_data.user_cb_state = HTTPD_SSL_USER_CB_SESS_CREATE;
          user_cb_data.tls = tls;
+       user_cb_data.sockfd = sockfd;
          (global_ctx->user_cb)((void *)&user_cb_data);
      }
  

### Describe alternatives you've considered.

See https://sourceforge.net/p/esp32s3-4827s043/code/HEAD/tree/trunk/https/main/WebServer.cpp

specifically the WebServer::isPeerSecure() to see where I need additional info.

Please take commit #81, in a later version I'm speculating that you'll help me by adding this field ;-)

laukik-hase, 2023-09-22:
Hello, @dannybackx!

Sorry for the late response. I think we can add the `sockfd` field to the callback arguments structure.

Could you explain a bit more on why the approach taken in the `https_server/simple` example (see [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/simple/main/main.c#L83-L89)) is not feasible for your implementation?

dannybackx, 2023-09-24:
Hmm, it is. My apologies, I overlooked this.

closed 2023-09-24 by dannybackx