espressif/esp-idf #12135: `tcpip_thread` runs after `esp_netif_deinit()` is called.

amrsoll, 2023-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with a BG77 modem and LoRaWAN module.

### Power Supply used.

USB

### What is the expected behavior?

I didn't find a way to unregister my network device driver. the alternative for me is to deinitialise the network stack, which works just as well in my use case.

After using `esp_netif_deinit()`, I expect the network stack should be deinitialised and inactive.

```c
esp_netif_deinit();
my_network_device_driver_deinit();
```

I open this issue because #4628 was marked as closed.

### What is the actual behavior?

`tcpip_thread` continues to run and tries to interact with uninitialised network device drivers

See [Debug Logs] for relevant diagnosis info.

### Steps to reproduce.

Please let me know in a comment if you don't already have enough context from the existing issues on GitHub.

### Debug Logs.

```plain
### `tcpip_thread` regularly pulls from uninitialised network device drivers

<details><summary><h2>`tcpip_thread` uses resources from uninitialised driver</h2></summary>
<p>

0x4201d34f: uart_write_bytes at /mnt/develdata/home/axel/esp/esp-idf/components/driver/uart/uart.c:1215 (discriminator 6)

0x4200dfe1: esp_modem_dte_send_data at /mnt/develdata/home/axel/Spear/components/sim7070/esp_modem.c:466

0x420a7503: esp_modem_dte_transmit at /mnt/develdata/home/axel/Spear/components/sim7070/esp_modem_netif.c:52

0x420a816f: esp_netif_transmit at /mnt/develdata/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1211

0x42037bbd: pppos_low_level_output at /mnt/develdata/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:195

0x4202c7fa: pppos_output_last at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:878

0x4202c945: pppos_write at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:241

0x420a7b99: ppp_write at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ppp.c:996

0x420357f1: fsm_sconfreq at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/fsm.c:757

0x42035a66: fsm_timeout at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/fsm.c:302

0x42025261: sys_check_timeouts at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c:401

0x4201e1cd: tcpip_timeouts_mbox_fetch at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:98
 (inlined by) tcpip_thread at /mnt/develdata/home/axel/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:142

E (33225) uart: uart_write_bytes(1217): uart driver error
```
</p>

### `tcpip_thread` (rarely) creates a system crash

It attempts to fetch a message from the global message box `tcpip_mbox`, and an assert fails

configASSERT( !( ( ( pvBuffer ) == NULL ) && ( ( pxQueue )->uxItemSize != ( UBaseType_t ) 0U ) ) );
```

<details><summary><b>Crash stacktrace</b></summary>
<p>
```
0x40375bfa: panic_abort at /opt/esp/idf/components/esp_system/panic.c:452
0x4037e569: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:84
0x40384a99: __assert_func at /opt/esp/idf/components/newlib/assert.c:81
0x4037ebdf: xQueueReceive at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/queue.c:1479 (discriminator 2)
0x4202e253: sys_arch_mbox_fetch at /opt/esp/idf/components/lwip/port/freertos/sys_arch.c:317
0x4201dfa3: tcpip_timeouts_mbox_fetch at /opt/esp/idf/components/lwip/lwip/src/api/tcpip.c:104
(inlined by) tcpip_thread at /opt/esp/idf/components/lwip/lwip/src/api/tcpip.c:142
```
</p>
```

### More Information.

I am working on a low power device where turning off the network stack is essential. The board doesn't stay awake that long after we deinit netif, but it is sufficient for error messages to show because `tcpip_thread` still tries to fetch resources, and we sometimes get a system crash which is very bad news for us.

I hope I didn't miss some uninitialisation steps when deactivating my device driver. I would be grateful to be pointed to the relevant documentation.

amrsoll, 2023-08-24:
Ah, it's only now that I see this point [in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#_CPPv416esp_netif_deinitv)

```
 Note: Deinitialization is not supported yet
```

Feel free to close

david-cermak, 2023-10-11:
@amrsoll How do you deinitialize the esp_netif instance that uses the UART driver? If you do it correctly you shouldn't get these issues:

> tcpip_thread regularly pulls from uninitialised network device drivers

The TCPIP task would still run and generate timeouts from time to time (which is not ideal for a low-power application), but you should be able to get rid of these crashes and errors. 

Could you please share some portions of your init and de-init code?

amrsoll, 2023-10-17:
Hi, sorry for the long delay. I was indeed able to find which part of our own code does attempt to use the network on a regular basis.

However, I would have hoped the network stack wouldn't attempt to use this interface once it had been de-init'ed? Could it simply fail the attempt instead of potentially cause a system crash when using de-initialised drivers?

```c
void modem_destroy()
{
	ESP_LOGW(TAG, "Destroying modem functionality");

	esp_event_handler_unregister(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event);
	esp_event_handler_unregister(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed);
	if (modem_netif_adapter) {
		esp_modem_netif_clear_default_handlers(modem_netif_adapter);
		esp_modem_netif_teardown(modem_netif_adapter);
		modem_netif_adapter = NULL;
	}
	if (modem_netif) {
		esp_netif_destroy(modem_netif);
		modem_netif = NULL;
	}
	if (dte) {
		dte->disable_power_supply_modem(dte);
		esp_modem_remove_event_handler(dte, modem_event_handler);
		dte->deinit(dte);
		dte = NULL;
	}
	if (dce) {
		dce->deinit(dce);
		dce = NULL;
	}
}
```

david-cermak, 2023-11-28:
The issue with deinit was reported few times already as https://github.com/espressif/esp-idf/issues/4628 or https://github.com/espressif/esp-idf/issues/1999.

TBH, this is not so complex to implement (and support deinit in lwip), but it always turned out that the deinit wasn't actually needed, as all the reported trouble was resolved differently (in the application code), so this feature doesn't have very high priority.

Closing, as it seems to me that this is also a similar case.

closed 2023-11-28 by david-cermak