espressif/esp-idf #12136: PCNT rotary encoder example, generating events in the queue at each step

0x0fe, 2023-08-24:
### Is your feature request related to a problem?

When checking the example encoer with PCNT https://github.com/espressif/esp-idf/blob/master/examples/peripherals/pcnt/rotary_encoder/main/rotary_encoder_example_main.c
I cannot find a way to generate event in the queue at each step, 

i checked the PCNT API, the documentation and i can find nothing that would allow to send an event in the queue at each step, so i wonder if this is because of PCNT limitation? If so what is the point of this example. Are we supposed to poll the count? It would be very wasteful and cancel the point of using PCNT, a simple GPIO interrupt based implementation then be a lot less wasteful (in CPU time).

Am i missing something obvious?

### Describe the solution you'd like.

an event at each step of the rotary encoder, so we could use it like input_key_service

### Describe alternatives you've considered.

well, if PCNT cannot acheive this a GPIO/IRQ based implementation would be the best alternative.

### Additional context.

UI control with rotary encoder

suda-morris, 2023-08-25:
Same feature request as https://github.com/espressif/esp-idf/issues/9604

FYI, our digital team is working with a hardware feature called "step notify" for the PCNT. Currently, you can try to use the GPIO interrupt together with the `pcnt_unit_get_count` function if you want to get notified by each step. i.e. register a gpio isr callback for a specific GPIO (either phase A or B), then in the callback, call the `pcnt_unit_get_count`.

> what is the point of this example?

The PCNT peripheral can also be used to measure the speed of a spinning motor :) Check [here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_bdc_speed_control/main/mcpwm_bdc_control_example_main.c#L88)

0x0fe, 2023-08-25:
@suda-morris Hello, thank you for your response.
I see, i ignored it was possible to register an interrupt on the GPIOs already used by the PCNT, i will do that then.

closed 2024-07-04 by espressif-bot