espressif/esp-idf #12137: USB Host Bulk Transfer IN overflow

kvhnuke, 2023-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2368-g52bca70b1a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKit C

### Power Supply used.

USB

### What is the expected behavior?

Expected to not overflow

### What is the actual behavior?

Callback returns
`USB_TRANSFER_STATUS_OVERFLOW`

### Steps to reproduce.

```
void bulk_transfer_read_cb(usb_transfer_t *transfer)
{
    printf("BULK: Transfer:Read type %d\n", transfer->num_bytes);
    printf("BULK: Transfer:Read status %d, actual number of bytes transferred %d, databuffer size %d, %d\n", transfer->status, transfer->actual_num_bytes, transfer->data_buffer_size, adsbdev->response_buf[8]);
}

int esp_libusb_bulk_transfer(class_driver_t *driver_obj, unsigned char endpoint, unsigned char *data, int length, int *transferred, unsigned int timeout)
{
    ESP_LOGI(TAG_ADSB, "esp_libusb_bulk_transfer info %d %d %d %d", endpoint, length, timeout, *transferred);
    assert(driver_obj->dev_hdl != NULL);
    size_t sizePacket = usb_round_up_to_mps(length, 64);
    usb_transfer_t *transfer = NULL;
    usb_host_transfer_alloc(sizePacket, 0, &transfer);
    ESP_LOGI(TAG_ADSB, "esp_libusb_bulk_transfer usb_host_transfer_alloc %d", sizePacket);
    transfer->num_bytes = sizePacket;
    transfer->device_handle = driver_obj->dev_hdl;
    transfer->bEndpointAddress = endpoint;
    transfer->callback = bulk_transfer_read_cb;
    transfer->context = (void *)&driver_obj;
    transfer->timeout_ms = timeout;
    esp_err_t r = usb_host_transfer_submit(transfer);
    if (r != ESP_OK)
    {
        ESP_LOGI(TAG_ADSB, "esp_libusb_bulk_transfer failed with %d", r);
        return -1;
    }
    usb_host_transfer_free(transfer);
    return 0;
}
```

chegewara, 2023-08-25:
Did you try to not use `printf` inside callback `bulk_transfer_read_cb`?

kvhnuke, 2023-08-25:
> Did you try to not use `printf` inside callback `bulk_transfer_read_cb`?
just checked, still the same transfer->status equals to 5 which is `USB_TRANSFER_STATUS_OVERFLOW`

Dazza0, 2023-08-25:
@kvhnuke This is usually a USB babble error. Meaning if I sent an IN request expecting only N bytes, the endpoint as transferred more than N bytes.

Could you dump paste the configuration descriptor of the USB device you are communicating with? In particular the endpoint descriptor of the endpoint that caused the overflow?

kvhnuke, 2023-08-25:
I actually went into a pretty deep dive around Babbel errors and cant seem to figure out why it is happening with ESP32S3, with libusb the same usb device works totally fine and no overflow.

@Dazza0 
```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1718
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc20
load:0x403cc700,len:0x2f38
SHA-256 comparison failed:
Calculated: 2010f2818c8c3c1d2bcfb5c648faae93c77af38d14f9b637ef726da50ef98ec1
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c9910
I (45) boot: ESP-IDF v5.2-dev-2368-g52bca70b1a-dirty 2nd stage bootloader
I (45) boot: compile time Aug 24 2023 17:34:09
I (46) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (54) boot.esp32s3: Boot SPI Speed : 80MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 00100000
I (107) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0e140h ( 57664) map
I (127) esp_image: segment 1: paddr=0001e168 vaddr=3fc92900 size=01eb0h (  7856) load
I (130) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=1ff2ch (130860) map
I (152) esp_image: segment 3: paddr=0003ff54 vaddr=3fc947b0 size=00abch (  2748) load
I (152) esp_image: segment 4: paddr=00040a18 vaddr=40374000 size=0e86ch ( 59500) load
I (173) boot: Loaded app from partition at offset 0x10000
I (173) boot: Disabling RNG early entropy source...
I (184) cpu_start: Multicore app
I (184) esp_psram: Found 2MB PSRAM device
I (185) esp_psram: Speed: 40MHz
I (185) cpu_start: Pro cpu up.
I (189) cpu_start: Starting app cpu, entry point is 0x40375290
0x40375290: call_start_cpu1 at /home/kvhnuke/esp/esp-idf/components/esp_system/port/cpu_start.c:170

I (0) cpu_start: App cpu up.
I (614) esp_psram: SPI SRAM memory test OK
I (624) cpu_start: Pro cpu start user code
I (624) cpu_start: cpu freq: 240000000 Hz
I (624) cpu_start: Application information:
I (627) cpu_start: Project name:     usb_host_lib_example
I (633) cpu_start: App version:      caa90fa-dirty
I (638) cpu_start: Compile time:     Aug 24 2023 17:34:01
I (644) cpu_start: ELF file SHA256:  03f302ad7...
I (650) cpu_start: ESP-IDF:          v5.2-dev-2368-g52bca70b1a-dirty
I (657) cpu_start: Min chip rev:     v0.1
I (662) cpu_start: Max chip rev:     v0.99 
I (666) cpu_start: Chip rev:         v0.1
I (671) heap_init: Initializing. RAM available for dynamic allocation:
I (678) heap_init: At 3FC95B28 len 00053BE8 (334 KiB): DRAM
I (685) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (691) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (697) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (704) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (712) spi_flash: detected chip: generic
I (716) spi_flash: flash io: dio
I (720) sleep: Configure to isolate all GPIO pins in sleep state
I (727) sleep: Enable automatic switching of GPIO sleep configuration
I (734) app_start: Starting scheduler on CPU0
I (739) app_start: Starting scheduler on CPU1
I (739) main_task: Started on CPU0
I (749) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (759) main_task: Calling app_main()
I (759) DAEMON: Installing USB Host Library
I (799) CLASS: Registering Client
I (1289) CLASS: Opening device at address 1
I (1299) CLASS: here2
I (1299) CLASS: Getting device information
I (1299) CLASS:         Full speed
I (1299) CLASS:         bConfigurationValue 1
I (1299) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0xbda
idProduct 0x2838
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (1329) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 34
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 4
bmAttributes 0x80
bMaxPower 500mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xff
        bInterfaceSubClass 0xff
        bInterfaceProtocol 0xff
        iInterface 5
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 1
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 0
        bInterfaceClass 0xff
        bInterfaceSubClass 0xff
        bInterfaceProtocol 0xff
        iInterface 5
I (1399) CLASS: Getting Manufacturer string descriptor
Realtek
I (1409) CLASS: Getting Product string descriptor
RTL2838UHIDIR
I (1419) CLASS: Getting Serial Number string descriptor
00000001
```

tore-espressif, 2023-09-01:
Hello @kvhnuke from you code snippet I can see that you free the transfer right after you submit it. Could you please remove this line?

``` c
usb_host_transfer_free(transfer);
```
Please note that the USB Host API is non-blocking. So the 'submit' call returns ESP_OK, but the transfer is not yet finished. The transfer is finished when you get your transfer callback. Only then you can free the transfer

kvhnuke, 2023-09-02:
@tore-espressif sorry I didnt cleanup the code properly for creating the issue, `usb_host_transfer_free(transfer);` gets called only after the usb exec is done as you can see in the below code. I have also tried it without that code and still the same situation
https://github.com/kvhnuke/esp32-rtl-sdr/blob/main/main/esp_libusb.c#L55-L76

kvhnuke, 2023-09-07:
@tore-espressif I tried same code on rpi pico with tinyusb stack and it works without any issues

roma-jam, 2023-09-21:
Hi @kvhnuke, 

May I ask which version of ESP-IDF do you use in your project?

kvhnuke, 2023-09-26:
tried it with "boot: ESP-IDF v5.2-dev-2934-g3b748a6cb7-dirty 2nd stage bootloader" still the same :( @roma-jam

Dazza0, 2023-11-27:
@kvhnuke Sorry for the delayed follow-up. I suspect this could be an issue with the RTL2838UHIDIR device being non-compliant when operating on a USB full-speed bus. I did a quick Google search for "RTL2838UHIDIR" and "Babble errors" and found [this post](https://forum.armbian.com/topic/18348-orange-pi-i96-usb-port-does-it-work/).

The post claims that the "RTL2838UHIDIR" when plugged into a HS Host ends up enumerating as an FS device, and when acting as an FS device, the logs show a `BUS BABBLE as a_host`.

Usually, babble errors are due to devices being non-compliant. Since most USB devices nowadays typically operate at HS or SS, it's not unusual to find non-compliance issues when these these devices are enumerated for LS or FS.

If you have a logic analyzer or oscilloscope, could you please run a capture of the BULK IN transfer. It should be very clear form the capture whether or not this is an issue with the device being non-compliant. According to the config descriptor above, Endpoint 1 IN has a maximum packet size (`wMaxPacketSize`) of `64`.

Thus, if this was a babble error,  the erroneous transaction of the BULK IN transfer should look something like the following:

```
Host -> Device          Device -> Host              Host -> Device

[Token Packet]          [Data Packet]               [HandShake Packet]
- PID: IN Token         - PID: DATA0/1              - N/A: Host should not respond with any handshake on a babble error
- Addr: 0x01            - Payload: > 64 bytes
- EndP: 0x01            - Causes babble error
```

Could you please check whether the scenario above is occurring?

Dazza0, 2023-12-05:
@kvhnuke I'm closing this issue as I think it's likely that device you are using is not compliant at full speed. If you have any new findings, please feel free to reopen this issue.

closed 2023-12-05 by Dazza0