espressif/esp-idf #12138: GATT_client怎么标识Server

king592, 2023-08-25:
### Is your feature request related to a problem?

我在使用ESP32的gatt_client例程，我想了解一下除了蓝牙名称还有什么办法标识被连接的设备，mac地址吗？
我想实现一个client，和很多个server通讯，他们平时是不连接的，只有在是想要获取某一个server的数据的时候才建立连接，那么我要怎么标识这些Server呢？

zhp0406, 2023-08-25:
@king592 
建议你使用 mac 地址标识设备。
另外，多连接你可以参考下：
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect

king592, 2023-08-25:
请问应该怎么获取到server端的mac地址呢？

 

 https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect

zhp0406, 2023-08-25:
@king592 
server 端获取自己的地址：
1-可以看 log 打印，例如 BLE_INIT: Bluetooth MAC 就指明了设备的地址：

```
I (368) BLE_INIT: BT controller compile version [85b425c]
I (368) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (408) system_api: Base MAC address is not set
I (408) system_api: read default base MAC address from EFUSE
I (408) BLE_INIT: Bluetooth MAC: 84:f7:03:67:48:8e
```

2-使用 api，可以调用  esp_bt_dev_get_address 函数来获取设备的 API

如果是 client 端想知道对端 server 的 mac 地址，需要通过扫描：

```
    uint32_t duration = 30;
    esp_ble_gap_start_scanning(duration);
```

扫描结果会在 ESP_GAP_BLE_SCAN_RESULT_EVT 事件报上来，其中包括 server 的设备地址，还包括其他的信息：

```
    struct ble_scan_result_evt_param {
        esp_gap_search_evt_t search_evt;            /*!< Search event type */
        esp_bd_addr_t bda;                          /*!< Bluetooth device address which has been searched */
        esp_bt_dev_type_t dev_type;                 /*!< Device type */
        esp_ble_addr_type_t ble_addr_type;          /*!< Ble device address type */
        esp_ble_evt_type_t ble_evt_type;            /*!< Ble scan result event type */
        int rssi;                                   /*!< Searched device's RSSI */
        uint8_t  ble_adv[ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX];     /*!< Received EIR */
        int flag;                                   /*!< Advertising data flag bit */
        int num_resps;                              /*!< Scan result number */
        uint8_t adv_data_len;                       /*!< Adv data length */
        uint8_t scan_rsp_len;                       /*!< Scan response length */
        uint32_t num_dis;                          /*!< The number of discard packets */
    } scan_rst;                                     /*!< Event parameter of ESP_GAP_BLE_SCAN_RESULT_EVT */
```

closed 2023-10-25 by Alvin1Zhang