espressif/esp-idf #12139: ESP32蓝牙无法读写128Bit的UUID

ForgetCSX, 2023-08-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-320-gb9e8e0ef4e

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

任意的esp32开发板

### Power Supply used.

USB

### What is the expected behavior?

链接到对端设备上发送数据，然后对端设备返回数据过来，我是通过手机蓝牙调试助手查看UUID，然后将ID写入到程序中，请看蓝牙调试助手显示：
[image]
结尾600的这个我选择这个UUID可以直接下发数据到对端设备

### What is the actual behavior?

esp报写入错误Write char failed, error status = 87
我不知道是我配置问题还是什么其他情况，
打补丁：
[image]
使用https://github.com/espressif/esp-idf/issues/12086#issuecomment-1692777212的补丁后也不行。

### Steps to reproduce.

1,get_idf
2,idf.py menuconfig,配置wifi和密码
3,idf.py idf.py build
4,idf.py -p /dev/ttyUSB0 flash monitor
5,浏览器打开esp32获取到的ip，添加一个设备，填写mac地址，选择类型3,这是我的配置，你可能需要更改成你的测试设备的UUID配置
6,重启设备蓝牙开始链接即可复现
下面是源文件：
[ble-gw.zip](https://github.com/espressif/esp-idf/files/12436823/ble-gw.zip)
@zhp0406 

### Debug Logs.

```plain
没有打补丁的日志：
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6624
load:0x40078000,len:15060
ho 0 tail 12 room 4
load:0x40080400,len:3816
[... 241 lines cut ...]
I (3391) ble_server: REMOTE BDA:
I (3391) ble_server: 57 4c 62 c0 30 15 
I (4991) ble_server: Status 0, MTU 200, conn_id 0
I (4991) ble_server: SEARCH RES: conn_id = 0 is primary service 1
I (4991) ble_server: start handle 16 end handle 23 current handle value 16
I (6751) wifi:new:<13,2>, old:<1,0>, ap:<255,255>, sta:<13,2>, prof:1
I (7531) wifi:state: init -> auth (b0)
I (7541) wifi:state: auth -> assoc (0)
I (7541) wifi:state: assoc -> run (10)
I (7651) wifi:connected with wifi-YF, aid = 7, channel 13, 40D, bssid = 20:76:93:4c:0c:86
I (7661) wifi:security: WPA2-PSK, phy: bgn, rssi: -39
I (7661) wifi:pm start, type: 1

I (7671) wifi:<ba-add>idx:0 (ifx:0, 20:76:93:4c:0c:86), tid:0, ssn:3, winSize:64
I (7701) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (8661) example: Starting webserver
I (8661) WEB SERVER: Starting server on port: '80'
I (8661) WEB SERVER: Registering URI handlers
I (8661) esp_netif_handlers: example_connect: sta ip: 192.168.123.66, mask: 255.255.255.0, gw: 192.168.123.1
I (8671) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.123.66
I (8771) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:32ae:a4ff:fe21:d560, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8771) example_connect: Connected to example_connect: sta
I (8781) example_connect: - IPv4 address: 192.168.123.66
I (8781) example_connect: - IPv6 address: fe80:0000:0000:0000:32ae:a4ff:fe21:d560, type: ESP_IP6_ADDR_IS_LINK_LOCAL
```

zhp0406, 2023-08-27:
@ForgetCSX 
Hi，使用下面的 patch 后问题还存在吗？
[0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch](https://github.com/espressif/esp-idf/files/12447138/0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch)

ForgetCSX, 2023-08-27:
@zhp0406 是的噢，打完补丁之后连server uuid都没获取到，您可以看下我发的日志，有补丁和没有补丁的

zhp0406, 2023-08-27:
@ForgetCSX 
首先确定一下打完补丁之后 为什么没有找到 server uuid：
基于 esp-idf/examples/bluetooth/bluedroid/ble/gatt_client 和 esp-idf/examples/bluetooth/bluedroid/ble/gatt_server，我写了一个测试的 patch，
[0002-fix-bt-bluedroid-Test-BLE-128-bit-uuid.patch](https://github.com/espressif/esp-idf/files/12447719/0002-fix-bt-bluedroid-Test-BLE-128-bit-uuid.patch)

你在第一个 patch 基础上测试下这个，看看能不能找到 128 bit uuid 的 service。

需要说明的是：
第一个 patch 中：修复了 client 报告 128 bit uuid 时，将长度自动转化成了 16 bit 的错误。

zhp0406, 2023-08-27:
@ForgetCSX 
我按照你的步骤复现问题没有成功：
@DESKTOP-1S387EO:~/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble-gw$ idf.py menuconfig
/home//esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: menuconfig
Running cmake in directory /home//esp/esp-idf/examples/bluetooth/bluedroid/ble/ble-gw/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home//.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home//esp/esp-idf/examples/bluetooth/bluedroid/ble/ble-gw"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/home//esp/esp-idf/examples/bluetooth/bluedroid/ble/ble-gw/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.34.1")
CMake Error at /home//esp/esp-idf/tools/cmake/project.cmake:349 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /home//esp/esp-idf/examples/common_components/led_strip
Call Stack (most recent call first):
  /home//esp/esp-idf/tools/cmake/project.cmake:433 (__project_init)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1

zhp0406, 2023-08-27:
@ForgetCSX 
根据你的代码，“Write char failed, error status” 是在 ESP_GATTC_WRITE_CHAR_EVT 事件报的，而 ESP_GATTC_WRITE_CHAR_EVT 事件是由 esp_ble_gattc_write_char API 触发的。
    ESP_GATT_ILLEGAL_PARAMETER      =   0x87,   /* 0x87 */    /* relate to BTA_GATT_ILLEGAL_PARAMETER in bta/bta_gatt_api.h */
 导致该错误的原因比较多。例如 handle 不对，你可以在 调用 esp_ble_gattc_write_char 之前加一些 log 来帮助判断。

ForgetCSX, 2023-08-27:
看您复现的日志，貌似idf的路径有问题，Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
/home//esp/esp-idf/examples/common_components/led_strip
Call Stack (most recent call first):
/home//esp/esp-idf/tools/cmake/project.cmake:433 (__project_init)
可以看到这里home这里有两个//，我不知道是不是这个原因，工程只依赖idf的代码，所以需要确保idf路径正确，您看是不是这个原因导致无法编译，下周晚些时候我会再去测试一下，谢谢了。

zhp0406, 2023-08-27:
@ForgetCSX 
好的，期待你的回复。

zhp0406, 2023-08-31:
@ForgetCSX 
你测试了吗?

ForgetCSX, 2023-09-14:
@zhp0406 您好，前段时间比较忙，这两条测试了一下，使用的是gattc_demo这个例子，将UUID等修改为我的设备的，然后打了您的测试补丁，都是到ESP_GATTC_CFG_MTU_EVT这里就停止了无法进入ESP_GATTC_SEARCH_RES_EVT等后面的事件，
[image]
可以看下日志，
[Uploading log.txt…]()

ForgetCSX, 2023-09-14:
@zhp0406 您好，前段时间比较忙，这两条测试了一下，使用的是gattc_demo这个例子，将UUID等修改为我的设备的，然后打了您的测试补丁，都是到ESP_GATTC_CFG_MTU_EVT这里就停止了无法进入ESP_GATTC_SEARCH_RES_EVT等后面的事件，
[image]
可以看下日志，

ForgetCSX, 2023-09-14:
@zhp0406 没有，我只改了客户端，用来链接我的设备，我来试下，那这样您这边不都可以测试了吗，您这边测试的怎么样。

zhp0406, 2023-09-14:
@ForgetCSX 
我这面测试是没问题的，默认的  server example 没有 128 bit uuid 来进行测试，所以我在 patch 里面也对 server example 做了更改，添加了 128 bit uuid 的 service，用于 128 bit uuid 的测试，因此，需要这两个example 都应用 patch 后一起测试。

需要说明的是，在 ESP_GATTC_SEARCH_RES_EVT 事件中，128 bit 的 base uuid 会被自动转换为 16 bit （即：p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_16），这是合理的。因为 128 bit base uuid 和对应的 16 bit uuid 是等价的。

如果应用了 [0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch] ，128 bit base uuid 不被会自动转化为对应的 16 bit uuid（即：p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_128）

但是，[0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch]  并没有提交到 IDF ，因为 128 bit base uuid 自动转换为对应的 16 bit uuid 是合理的。我们在我们的最新的 IDF 文档中对这点进行了说明。

zhp0406, 2023-09-14:
@ForgetCSX 
client 端 ESP_GATTC_SEARCH_CMPL_EVT 的打印有打印出来吗？

ForgetCSX, 2023-09-14:
@zhp0406 这个倒是有的
[image]

ForgetCSX, 2023-09-15:
@zhp0406 您好，能帮我写一个链接我这个设备的例子我测试一下吗，我不知道是不是我配置的不对导致无法通信的。
[image]

zhp0406, 2023-09-15:
@ForgetCSX 
好的，我可以给你写一部分来测试基本功能，这面测试没问题后给你。不过这可能得排到下周了，因为现在手头还有其他工作。
你可以先继续尝试一下，有问题直接报给我。

ForgetCSX, 2023-09-15:
好的好的，谢谢了，目前我使用下面这个设备的配置可以连上通信，这个设备是我自己写的，SERVICE_UUID是FFF0，特征值定一个可读可写可通知的FFF6
[image]
使用gatt_cilent例子简单修改下已经通信成功了，上面那个是第三方设备，调试无法通信，但是我使用手机蓝牙调试助手是可以发送命令的，应该是我那里没配置好，我再找找其他方法。

zhp0406, 2023-11-21:
@ForgetCSX 
你那面的问题解决了吗?