espressif/esp-idf #12140: Integrating rotary_encoder_service events with input_key_service events.

0x0fe, 2023-08-25:
### Is your feature request related to a problem?

So, i have a rotary_encoder_service, which generates its own events, and the input_key_service also generating its own events. 

### Describe the solution you'd like.

to find a way to integrate the encoder events into the input_key_service events, so that the same handler can be used for both events in a given application. 
I did not find a clear way to do this, nor a service for rotary encoder already integrated into the IDF, there is the PCNT example but it is not a service and not even usable without modifications, for the purpose of user interface key and encoder events are the same thing, the only difference being that encoder even need to pass the position too.

### Additional context.

i am using this within ADF but the question is not specific to ADF, in IDF the service/events meachanism is the same.

suda-morris, 2023-08-28:
I think this is a feature request for ESP-ADF. The examples in the esp-idf are not aiming for any mature solution or service-like applications. The examples in esp-idf are simply for illustrating the usage of the APIs. 

Or a feature request for [this component](https://components.espressif.com/components/espressif/knob)

0x0fe, 2023-09-10:
@suda-morris thank you for pointing this component, i was not aware of it so i implemented a rotary encoder service on one interrupt, but i will look into it for future.

closed 2023-09-10 by 0x0fe