espressif/esp-idf #12142: usb msc host example msc_host_vfs_register error

f1andrew, 2023-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I'm using ESP-IDF v5.1.1 release and ESP32S3 devkitC board.
I tried usb msc host example with two different usb drives and getting same error:

`I (31) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (31) boot: compile time Aug 26 2023 20:02:35
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0f104h ( 61700) map
I (115) esp_image: segment 1: paddr=0001f12c vaddr=3fc92200 size=00eech (  3820) load
I (116) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=316fch (202492) map
I (157) esp_image: segment 3: paddr=00051724 vaddr=3fc930ec size=019bch (  6588) load
I (159) esp_image: segment 4: paddr=000530e8 vaddr=40374000 size=0e1a0h ( 57760) load
I (181) boot: Loaded app from partition at offset 0x10000
I (182) boot: Disabling RNG early entropy source...
I (193) cpu_start: Multicore app
I (193) cpu_start: Pro cpu up.
I (193) cpu_start: Starting app cpu, entry point is 0x403752d4
0x403752d4: call_start_cpu1 at /home/andriy/esp_v5_1/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (211) cpu_start: Pro cpu start user code
I (212) cpu_start: cpu freq: 160000000 Hz
I (212) cpu_start: Application information:
I (215) cpu_start: Project name:     usb-msc
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Aug 26 2023 20:02:25
I (230) cpu_start: ELF file SHA256:  2de2bc1c4523baf1...
I (236) cpu_start: ESP-IDF:          v5.1.1
I (241) cpu_start: Min chip rev:     v0.0
I (246) cpu_start: Max chip rev:     v0.99 
I (250) cpu_start: Chip rev:         v0.1
I (255) heap_init: Initializing. RAM available for dynamic allocation:
I (262) heap_init: At 3FC95370 len 000543A0 (336 KiB): DRAM
I (269) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (275) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (281) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (289) spi_flash: detected chip: gd
I (292) spi_flash: flash io: dio
W (296) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) sleep: Configure to isolate all GPIO pins in sleep state
I (316) sleep: Enable automatic switching of GPIO sleep configuration
I (323) app_start: Starting scheduler on CPU0
I (328) app_start: Starting scheduler on CPU1
I (328) main_task: Started on CPU0
I (338) main_task: Calling app_main()
I (338) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (378) example: Waiting for USB stick to be connected
I (758) example: MSC device connected
I (758) example: connection...
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x930
idProduct 0x6544
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 200mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
Device info:
         Capacity: 7399 MB
         Sector size: 512
         Sector count: 15155199
         PID: 0x6544 
         VID: 0x 930 
         iProduct: USB Flash Memory 
         iManufacturer:          
         iSerialNumber: ECA86B243FB9C051C007411C 
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x4200971d
0x4200971d: app_main at /home/andriy/esp_v5_1/msc/build/../main/msc_example_main.c:201 (discriminator 1)

file: "../main/msc_example_main.c" line 201
func: app_main
expression: msc_host_vfs_register(msc_device, "/usb", &mount_config, &vfs_handle)

abort() was called at PC 0x40379c47 on core 0
0x40379c47: _esp_error_check_failed at /home/andriy/esp_v5_1/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x4037581e:0x3fc98c90 0x40379c51:0x3fc98cb0 0x4038000a:0x3fc98cd0 0x40379c47:0x3fc98d40 0x4200971d:0x3fc98d70 0x42030f23:0x3fc98ec0 0x4037c355:0x3fc98ef0
0x4037581e: panic_abort at /home/andriy/esp_v5_1/esp-idf/components/esp_system/panic.c:452

0x40379c51: esp_system_abort at /home/andriy/esp_v5_1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4038000a: abort at /home/andriy/esp_v5_1/esp-idf/components/newlib/abort.c:38

0x40379c47: _esp_error_check_failed at /home/andriy/esp_v5_1/esp-idf/components/esp_system/esp_err.c:50

0x4200971d: app_main at /home/andriy/esp_v5_1/msc/build/../main/msc_example_main.c:201 (discriminator 1)

0x42030f23: main_task at /home/andriy/esp_v5_1/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x4037c355: vPortTaskWrapper at /home/andriy/esp_v5_1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
`

0727fqh, 2023-09-04:
I also encountered the same problem, there is a solution?

leeebo, 2023-09-04:
@0727fqh @f1andrew there is a bug in [usb_host_msc version 1.1.0](https://components.espressif.com/components/espressif/usb_host_msc/versions/1.1.0), when enable PSRAM the error will occur.

Please fall back to [usb_host_msc version 1.0.4](https://components.espressif.com/components/espressif/usb_host_msc/versions/1.0.4) or disable PSRAM to make it work

leeebo, 2023-09-04:
Link https://github.com/espressif/idf-extra-components/issues/202

closed 2023-10-02 by tore-espressif

versamodule, 2024-09-14:
Using IDFv5.3.1 and the MSC example I still get this same error.