espressif/esp-idf #12143: ESP32 without PSRAM Wifi connection

alberk8, 2023-08-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have ESP32 Rev3 with IDF 4.4.5.   If I compile firmware with PSRAM enabled and deployed onto a hardware without PSRAM then wifi will not connect.  Is there a way to get that to work?

```
CONFIG_ESP32_REV_MIN_3=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_USE_MEMMAP=n
CONFIG_SPIRAM_USE_CAPS_ALLOC=n
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=n
CONFIG_SPIRAM_TYPE_ESPPSRAM64=n
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
```

alberk8, 2023-09-19:
Hi,
Can someone confirm this?  Thank you.

jimmo, 2023-11-09:
We're seeing this too. (Duplicate of https://github.com/espressif/esp-idf/issues/11971)

mhdong, 2024-03-22:
Hi @alberk8 
please refer https://github.com/nanoframework/nf-interpreter/pull/2837
Issue solved with https://github.com/espressif/esp-idf/commit/f37448cddb395939d2a168fdd86dbb0cd868fc59

closed 2024-04-24 by Sherry616