espressif/esp-idf #12144: Example examples/bluetooth/bluedroid/ble/gattc_multi_connect doesn't receive all data sent by server

LeeDr, 2023-08-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm running 3 instances of examples/bluetooth/bluedroid/ble/gatt_server with TEST_DEVICE_NAME changed to  "ESP_GATTS_DEMO_a",  "ESP_GATTS_DEMO_b", and  "ESP_GATTS_DEMO_c" and one instance of examples/bluetooth/bluedroid/ble/gattc_multi_connect on esp32-s3 boards.  It generally seems to be working but the servers appear to send 35 bytes of data'
```
I (4386) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (5546) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 200
I (5546) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 200
I (5626) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 43
I (5626) GATTS_DEMO: GATT_WRITE_EVT, value len 2, value :
I (5626) GATTS_DEMO: 01 00 
I (5636) GATTS_DEMO: notify enable
I (5636) GATTS_DEMO: ESP_GATTS_CONF_EVT, status 0 attr_handle 42
I (5706) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 2, handle 42
I (5706) GATTS_DEMO: GATT_WRITE_EVT, value len 35, value :
I (5706) GATTS_DEMO: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (5716) GATTS_DEMO: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
I (5726) GATTS_DEMO: 20 21 22 
```

while the client only seems to receive 15 bytes.  

```
I (584176397) GATTC_MULTIPLE_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (584177597) GATTC_MULTIPLE_DEMO: Status 0, MTU 200, conn_id 2
I (584177597) GATTC_MULTIPLE_DEMO: SEARCH RES: conn_id = 2 is primary service 1
I (584177607) GATTC_MULTIPLE_DEMO: start handle 40 end handle 43 current handle value 40
I (584177617) GATTC_MULTIPLE_DEMO: UUID16: ff
I (584177677) GATTC_MULTIPLE_DEMO: ESP_GATTC_NOTIFY_EVT, Receive notify value:
I (584177677) GATTC_MULTIPLE_DEMO: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e   <<<<<<<<<<<<<<
I (584177687) GATTC_MULTIPLE_DEMO: write descr success
I (584177757) GATTC_MULTIPLE_DEMO: Write char success
I (584177757) GATTC_MULTIPLE_DEMO: Scan start success
I (584177897) GATTC_MULTIPLE_DEMO:  
I (584177897) GATTC_MULTIPLE_DEMO: all devices are connected
I (584177897) GATTC_MULTIPLE_DEMO: Stop scan successfully
```

zhp0406, 2023-08-28:
@LeeDr 
In the `example` located at `esp-idf/examples/bluetooth/bluedroid/ble/gatt_server`, only 15 bytes of data have been notified. If you haven't made any modifications to the `example`, receiving 15 bytes of data is normal.

```
    case ESP_GATTS_WRITE_EVT: {
        ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, conn_id %d, trans_id %" PRIu32 ", handle %d", param->write.conn_id, param->write.trans_id, param->write.handle);
        if (!param->write.is_prep){
            ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, value len %d, value :", param->write.len);
            esp_log_buffer_hex(GATTS_TAG, param->write.value, param->write.len);
            if (gl_profile_tab[PROFILE_B_APP_ID].descr_handle == param->write.handle && param->write.len == 2){
                uint16_t descr_value= param->write.value[1]<<8 | param->write.value[0];
                if (descr_value == 0x0001){
                    if (b_property & ESP_GATT_CHAR_PROP_BIT_NOTIFY){
                        ESP_LOGI(GATTS_TAG, "notify enable");
                        uint8_t notify_data[15];
                        for (int i = 0; i < sizeof(notify_data); ++i)
                        {
                            notify_data[i] = i%0xff;
                        }
                        //the size of notify_data[] need less than MTU size
                        esp_ble_gatts_send_indicate(gatts_if, param->write.conn_id, gl_profile_tab[PROFILE_B_APP_ID].char_handle,
                                                sizeof(notify_data), notify_data, false);
                    }
                }else if (descr_value == 0x0002){
                    if (b_property & ESP_GATT_CHAR_PROP_BIT_INDICATE){
                        ESP_LOGI(GATTS_TAG, "indicate enable");
                        uint8_t indicate_data[15];
                        for (int i = 0; i < sizeof(indicate_data); ++i)
                        {
                            indicate_data[i] = i%0xff;
                        }
                        //the size of indicate_data[] need less than MTU size
                        esp_ble_gatts_send_indicate(gatts_if, param->write.conn_id, gl_profile_tab[PROFILE_B_APP_ID].char_handle,
                                                sizeof(indicate_data), indicate_data, true);
                    }
                }
                else if (descr_value == 0x0000){
                    ESP_LOGI(GATTS_TAG, "notify/indicate disable ");
                }else{
                    ESP_LOGE(GATTS_TAG, "unknown value");
                }

            }
        }
        example_write_event_env(gatts_if, &b_prepare_write_env, param);
        break;
    }
```

If the data notified by the server side is indeed 35 bytes, then it is necessary to consider whether the MTU meets the requirements:

**The size of indicate_data[] must be smaller than the MTU size.**

You can utilize the following API to adjust the MTU:

```
esp_err_t esp_ble_gatt_set_local_mtu(uint16_t mtu); // Set local MTU
esp_err_t esp_ble_gattc_send_mtu_req(esp_gatt_if_t gattc_if, uint16_t conn_id); // Configure the MTU size
```

If there are further issues, could you please provide the server-side code and the complete log for reference?

LeeDr, 2023-08-28:
Thanks @zhp0406 .  It's starting to make sense to me now.  But what I don't understand is what is the `param->write.value` being logged here.  It's not the data being used to notify (although it matches the same pattern of the byte index). 

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L357

And the MTU seems to be set to 500 here? 
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L730

I didn't list my code because I'm running the examples in master branch.

zhp0406, 2023-08-29:
@LeeDr 
The following code is used to display the data written by the client that the server has received:

```c
esp_log_buffer_hex(GATTS_TABLE_TAG, param->write.value, param->write.len);
```

zhp0406, 2023-08-29:
@LeeDr 
You also need to perform the following setup on the client side:

```
esp_err_t esp_ble_gatt_set_local_mtu(uint16_t mtu); // Set local MTU
conn_id); // Configure the MTU size
```

Next, initiate the MTU exchange request:

```
esp_err_t esp_ble_gattc_send_mtu_req(esp_gatt_if_t gattc_if, uint16_t mtu);
```

It's important to note that:

The server side needs to configure the local MTU, which will be subsequently used for negotiating the MTU with the client:

```
esp_err_t esp_ble_gatt_set_local_mtu(uint16_t mtu); // Set local MTU
conn_id); // Configure the MTU size
```

closed 2023-10-25 by Alvin1Zhang