espressif/esp-idf #12145: esp_http_client_read downloading speed

gl-chenxiaosheng, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used `esp_http_client_read ` to pull firmware from the server and then upgraded it. I found that if the server is in a local environment, the download speed is fast, but if it is a remote server, the download speed is only 7-8kb. What is the reason for this ?

hmalpani, 2023-08-30:
Hello @gl-chenxiaosheng 

I am hosting my firmware on GitHub. When I tried updating to that firmware, I got overall OTA speed ~25kbps. I am not sure but it might be because of poor network connectivity or some server issues. You can try improving the network performance using https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#improving-network-speed

kriegste, 2023-08-30:
Even with the network speed tweaks, OTA via download from the internet is slow for me, too. My laptop achieves much higher speeds downloading the exact same file.

However, OTA via local upload (HTTP server running on the ESP32) is fast. So I can rule out the flash writing as the bottleneck here.

I don't know why there is such a big difference. It could be caused by the SSL decryption.

kriegste, 2023-08-31:
SSL decryption speed should be depending on CPU frequency, right? I compared OTA times at 160 and 240 MHz. There is no difference. I get about 40 kB/s.

gl-chenxiaosheng, 2023-09-07:
> 

So what is the maximum transmission rate of OTA after your optimization？

gl-chenxiaosheng, 2023-09-07:
The current speed is about 30kb, this speed is still too low for us.

hmalpani, 2023-09-18:
> So what is the maximum transmission rate of OTA after your optimization？

I updated the esp_http_client example to calculate the throughput. I started a server and set the client to download a 1 GB file. After setting up the config options for optimization, I got download speed of ~950KBps. Note that this speed is the download speed by es_http_client and not the OTA speed.

gl-chenxiaosheng, 2023-09-18:
Thank you for your reply. The optimization has indeed improved, but the disadvantage is that it will increase internal RAM consumption. At present, after debugging, the internal ram consumption is acceptable, and the download speed is about 200kb.

closed 2023-09-18 by gl-chenxiaosheng

pnexus-wikii, 2026-11-10:
Sharing a field observation that may help others troubleshooting slow `esp_http_client_read()` / HTTPS OTA on **ESP32-S3**.

We download the full firmware image over **one HTTPS connection** using `esp_http_client`, then stream it into the OTA partition via `esp_ota_write()`. We do **not** use `esp_https_ota` partial download, HTTP `Range` requests, or reconnect every 4 KB; the server returns HTTP 200 with the complete `Content-Length`.

**One configuration change made a very large difference in our setup:**

```ini
# Before
CONFIG_ESP_WIFI_RX_BA_WIN=6

# After
CONFIG_ESP_WIFI_RX_BA_WIN=32
```

Observed OTA throughput increased **by more than an order of magnitude** (the application rate display went from a few “k” per second to around 100 “k” per second; the exact bit-vs-byte unit labeling is still being verified). This is a **single-environment field observation**, not a controlled benchmark or a guaranteed improvement for other devices/networks.

Other relevant settings were:

```ini
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_LWIP_TCP_WND_DEFAULT=23040
CONFIG_LWIP_TCP_RECVMBOX_SIZE=18
CONFIG_LWIP_TCP_MSS=1440
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
```

Application OTA I/O buffer: 4096 bytes; `esp_http_client_config_t.buffer_size`: 4096 bytes. TLS dynamic input buffering was disabled. The HTTP download loop reused the same client throughout the image transfer.

Espressif's ESP32-S3 Wi-Fi performance documentation says the RX BA window should be sized with reference to `min(2 * WIFI_STATIC_RX_BUFFER_NUM, WIFI_DYNAMIC_RX_BUFFER_NUM)`, which is **32** for the RX buffer settings above: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/wifi-driver/wifi-performance-and-power-save.html

**Takeaway:** If HTTPS OTA is unexpectedly stuck at a few k/s, checking `CONFIG_ESP_WIFI_RX_BA_WIN` and the related Wi-Fi RX buffer counts may be worthwhile *before* assuming that TLS, TCP window sizing, or the OTA flash write loop is the bottleneck. Larger BA windows may increase memory usage; please validate heap headroom, AP compatibility, and performance with repeatable A/B measurements. ESP-IDF version, precise throughput units, and repeat-test data are not yet available for this observation.