espressif/esp-idf #12147: 请问有没有基于IDF5.0以上，关于提供微信小程序连接和蓝牙控制芯片的例程？

Szeroy, 2023-08-28:


Szeroy, 2023-08-29:
> @Szeroy 请问你是指的经典蓝牙还是低功耗蓝牙呢？ 

低功耗蓝牙，我看到官方以前有个月球灯的项目，里面做了蓝牙配网和控制，但是那个idf版本太低了，很多函数都找不到

zhp0406, 2023-08-29:
@Szeroy 
月球灯的项目你是在哪里找到的?

horw, 2023-08-29:
> > @Szeroy 请问你是指的经典蓝牙还是低功耗蓝牙呢？
> 
> 低功耗蓝牙，我看到官方以前有个月球灯的项目，里面做了蓝牙配网和控制，但是那个idf版本太低了，很多函数都找不到

Hi，你找esp端代码还是微信小程序端代码

Szeroy, 2023-08-29:
> > > @Szeroy 请问你是指的经典蓝牙还是低功耗蓝牙呢？
> > 
> > 低功耗蓝牙，我看到官方以前有个月球灯的项目，里面做了蓝牙配网和控制，但是那个idf版本太低了，很多函数都找不到
> 
> 
> Hi，你找esp端代码还是微信小程序端代码

都需要，如果都有的话就好了

Jiangyafeng, 2023-09-01:
Hi @Szeroy 设备端直接使用 blufi demo：https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/blufi，进行烧录就可以了。
手机端可以直接在微信小程序里面搜索 ESPConfig 然后进行配网就可以实现（这个已经不维护了，建议使用blufi demo里面的指定 app）。
微信小程序 ESPconfig 源码可以查看链接：https://github.com/EspressifApps/ESP-Config-WeChat

Szeroy, 2023-09-06:
> Hi @Szeroy 设备端直接使用 blufi demo：https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/blufi，进行烧录就可以了。
> 手机端可以直接在微信小程序里面搜索 ESPConfig 然后进行配网就可以实现（这个已经不维护了，建议使用blufi demo里面的指定 app）。
> 微信小程序 ESPconfig 源码可以查看链接：https://github.com/EspressifApps/ESP-Config-WeChat

谢谢了，我研究一下😃

closed 2023-09-06 by Szeroy