espressif/esp-idf #12148: OTA upgrade fail

Jimis333, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

must have v5 but "git describe --tags" returns error. Command "idf.py" in VScode returns v1.0.3. Esspresif IDF Extension says v1.6.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

It is supposed to verify the certification file

### What is the actual behavior?

I have the same problem with issue https://github.com/espressif/esp-idf/issues/6833 I get 
"**esp-tls-mbedtls: Failed to verify peer certificate!**"

### Steps to reproduce.

I created a new example program from the IDF examples "advanced_https_ota" and setup the menuconfig for a WROOM board

### More Information.

Hi
I have the same problem with issue https://github.com/espressif/esp-idf/issues/6833
I have enabled EXAMPLE_SKIP_COMMON_NAME_CHECK, in menuconfig but still got the same problem

mahavirj, 2023-08-30:
@Jimis333 

Simplest option here is to enable the certificate bundle feature in the advanced_ota example, please see the code below:

https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/examples/system/ota/simple_ota_example/main/simple_ota_example.c#L95

This ensures that most commonly used root certs are bundled in the firmware and hence it does not need any additional certs.

Alternatively, you can share the endpoint and cert that you using and then we can evaluate what could be the reason of failure in the certificate.

Jimis333, 2023-08-31:
Hi mahavirj,
Thanks for the reply. I have already enabled the bundle as shown in picture. 
I had to include "esp_crt_bundle.h" for the crt bundle to work but it did.Thanks a lot! 

Below is the result that gives an error of version but that's another problem

I (5693) advanced_https_ota_example: Starting Advanced OTA example
I (5693) advanced_https_ota_example: OTA started
I (5703) main_task: Returned from app_main()
I (6443) esp-x509-crt-bundle: Certificate validated
I (8013) esp-x509-crt-bundle: Certificate validated
I (8733) esp_https_ota: Starting OTA...
I (8733) advanced_https_ota_example: Connected to server
I (8733) esp_https_ota: Writing to partition subtype 16 at offset 0x120000
I (8743) advanced_https_ota_example: Reading Image Description
I (8903) advanced_https_ota_example: Running firmware version: 1
W (8903) advanced_https_ota_example: Current running version is the same as a new. We will not continue the update.
E (8903) advanced_https_ota_example: image header verification failed
I (8913) advanced_https_ota_example: OTA abort
E (8913) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed

I have a few questions
1) why do I need to add this line since I already do it in menuconfig
2)  why the cert.pem file  from the example doesn't work ? I have also tried making a cert file using the command
echo "" 
[image]
| openssl s_client -showcerts -connect raw.githubusercontent.com:443 | sed -n "1,/Root/d; /BEGIN/,/END/p" | openssl x509 -outform PEM >ca_cert.pem

I attach my config and files
url : https://github.com/Jimis333/Test/blob/37b44668bb625586653054c6fc7b48fa8017df42/hello_world.bin?raw=true

Jimis333, 2023-08-31:
I guess I found the answer to my 1st question. Following line is missing from sdkconfig. They have updated simple_ota but not advanced_ota example
CONFIG_EXAMPLE_USE_CERT_BUNDLE=y
[image]

mahavirj, 2023-08-31:
@Jimis333 

1. Please see the cert bundle API usage documented [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_crt_bundle.html#configuration)
2. I think you are embedding the certificate for `raw.githubusercontent.com` endpoint but using firmware URL for `github.com`. Please change the config `CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL` to `https://raw.githubusercontent.com/Jimis333/Test/main/hello_world.bin` and it should fix the issue

Jimis333, 2023-08-31:
It worked thanks !

Jimis333, 2023-09-01:
I re-open the issue cause I still have a problem. If I sign my build I get a signature verification fail. 
So here are the details 
I have uploaded the same code in git but changed the version number. So both codes are signed using the same key. the version check passes but I get a signature fail. Why is that ? Below is the output 

I (0) cpu_start: App cpu up.
I (508) cpu_start: Pro cpu start user code
I (508) cpu_start: cpu freq: 160000000 Hz
I (508) cpu_start: Application information:
I (513) cpu_start: Project name:     advanced_https_ota
I (519) cpu_start: App version:      0.0.0.2
I (524) cpu_start: Compile time:     Sep  1 2023 13:33:37
I (530) cpu_start: ELF file SHA256:  f58c44250bb2e505...
I (536) cpu_start: ESP-IDF:          v5.1.1
I (541) cpu_start: Min chip rev:     v3.0
I (545) cpu_start: Max chip rev:     v3.99 
I (550) cpu_start: Chip rev:         v3.1
I (555) heap_init: Initializing. RAM available for dynamic allocation:
I (562) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (568) heap_init: At 3FFB7CE0 len 00028320 (160 KiB): DRAM
I (574) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (581) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (587) heap_init: At 40095488 len 0000AB78 (42 KiB): IRAM
I (595) spi_flash: detected chip: gd
I (598) spi_flash: flash io: dio
I (605) app_start: Starting scheduler on CPU0
I (607) app_start: Starting scheduler on CPU1
I (607) main_task: Started on CPU0
I (617) main_task: Calling app_main()
I (617) advanced_https_ota_example: OTA example app_main start
I (637) example_connect: Start example_connect.
I (647) wifi:wifi driver task: 3ffbf944, prio:23, stack:6656, core=0
I (657) wifi:wifi firmware version: ce9244d
I (657) wifi:wifi certification version: v7.0
I (657) wifi:config NVS flash: enabled
I (657) wifi:config nano formating: disabled
I (657) wifi:Init data frame dynamic rx buffer num: 32
I (667) wifi:Init management frame dynamic rx buffer num: 32
I (667) wifi:Init management short buffer num: 32
I (677) wifi:Init dynamic tx buffer num: 32
I (677) wifi:Init static rx buffer size: 1600
I (677) wifi:Init static rx buffer num: 10
I (687) wifi:Init dynamic rx buffer num: 32
I (687) wifi_init: rx ba win: 6
I (697) wifi_init: tcpip mbox: 32
I (697) wifi_init: udp mbox: 6
I (697) wifi_init: tcp mbox: 6
I (707) wifi_init: tcp tx win: 5744
I (707) wifi_init: tcp rx win: 5744
I (707) wifi_init: tcp mss: 1440
I (717) wifi_init: WiFi IRAM OP enabled
I (717) wifi_init: WiFi RX IRAM OP enabled
I (727) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (827) wifi:mode : sta (80:64:6f:13:a1:e4)
I (827) wifi:enable tsf
I (827) example_connect: Connecting to Emphasisdw...
I (837) example_connect: Waiting for IP(s)
I (3247) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4167) wifi:state: init -> auth (b0)
I (4177) wifi:state: auth -> assoc (0)
I (4187) wifi:state: assoc -> run (10)
I (4227) wifi:connected with Emphasisdw, aid = 4, channel 1, BW20, bssid = a4:88:73:43:b1:20
I (4227) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (4227) wifi:pm start, type: 1

I (4297) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4737) wifi:<ba-add>idx:0 (ifx:0, a4:88:73:43:b1:20), tid:6, ssn:2, winSize:64
I (5637) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:8264:6fff:fe13:a1e4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5737) esp_netif_handlers: example_netif_sta ip: 192.168.8.154, mask: 255.255.255.0, gw: 192.168.8.1
I (5737) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.8.154
I (5747) example_common: Connected to example_netif_sta
I (5747) example_common: - IPv4 address: 192.168.8.154,
I (5757) example_common: - IPv6 address: fe80:0000:0000:0000:8264:6fff:fe13:a1e4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
JIM connected...........
I (5767) wifi:Set ps type: 0, coexist: 0

I (5777) advanced_https_ota_example: Starting Advanced OTA example
I (5777) advanced_https_ota_example: OTA started
I (5787) main_task: Returned from app_main()
I (5797) wifi:<ba-add>idx:1 (ifx:0, a4:88:73:43:b1:20), tid:0, ssn:0, winSize:64
I (6837) esp_https_ota: Starting OTA...
I (6837) advanced_https_ota_example: Connected to server
I (6837) esp_https_ota: Writing to partition subtype 16 at offset 0x120000
I (6847) advanced_https_ota_example: Reading Image Description
I (6847) advanced_https_ota_example: Running firmware version: 0.0.0.2
I (6857) advanced_https_ota_example: Verifying chip id of new image: 0
I (35857) esp_image: segment 0: paddr=00120020 vaddr=3f400020 size=27230h (160304) map
I (35907) esp_image: segment 1: paddr=00147258 vaddr=3ffb0000 size=03630h ( 13872) 
I (35907) esp_image: segment 2: paddr=0014a890 vaddr=40080000 size=05788h ( 22408) 
I (35917) esp_image: segment 3: paddr=00150020 vaddr=400d0020 size=9b0ech (635116) map
I (36097) esp_image: segment 4: paddr=001eb114 vaddr=40085788 size=0fd00h ( 64768) 
I (36117) esp_image: segment 5: paddr=001fae1c vaddr=00000000 size=051b4h ( 20916) 
I (36127) esp_image: Verifying image signature...
I (36127) secure_boot_v2: Take trusted digest key(s) from running app
**E (36137) esp_image: Secure boot signature verification failed**
I (36137) esp_image: Calculating simple hash to check for corruption...
W (36357) esp_image: image valid, signature bad
E (36357) advanced_https_ota_example: Image validation failed, image is corrupted
E (36357) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed 0x1503
I (36357) advanced_https_ota_example: OTA finish

mahavirj, 2023-09-04:
@Jimis333 

Unfortunately this information is not sufficient to comment on the issue. Please provide detailed steps that you followed, complete console log and the efuse summary of the device. Thanks.

Jimis333, 2023-09-04:
@mahavirj 

I run the advanced_https_ota example from the ESP-IDF examples. I have included the sdkconfig file with my settings
I sign the build with my secure_boot_signing_key.pem
I take the advanced_https_ota.bin  and upload it to git (url in sdkconfig)
I change the version of the project to a lower one so that it can be upgraded from the .bin from git
(I change the version in the cmakeLists.txt like this :  set(PROJECT_VER "0.0.0.1") )
Now when I run my program it's supposed to call the url and download a lower version of itself
Upgrade stops in the verification process (full output in text file included) : 

I (38857) esp_image: segment 5: paddr=001fad28 vaddr=00000000 size=052a8h ( 21160) 
I (38867) esp_image: Verifying image signature...
I (38867) secure_boot_v2: Take trusted digest key(s) from running app
**E (38877) esp_image: Secure boot signature verification failed**
I (38877) esp_image: Calculating simple hash to check for corruption...
W (39097) esp_image: image valid, signature bad
E (39097) advanced_https_ota_example: Image validation failed, image is corrupted
E (39097) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed 0x1503
I (39097) advanced_https_ota_example: OTA finish

[output.txt](https://github.com/espressif/esp-idf/files/12513772/output.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12513773/sdkconfig.txt)

Jimis333, 2023-09-05:
I just found out from the last answer to my other opened issue what was the problem ! 
I signed by mistake using ECDSA key instead of RSA. (https://github.com/espressif/esp-idf/issues/12170) which is not supported by my chip EC3.

closed 2023-09-05 by Jimis333