espressif/esp-idf #12149: BLE_HID_DEVICE example error

monurbstnc, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello I am trying to build something using bluetooth/esp_hid_device example. But when I am try to build this I got undefined reference to `esp_bt_gap_set_cod' error. Thanks for your helps
I am using ESP32s3 devkitc-1.

zhp0406, 2023-09-11:
@monurbstnc 
thanks for report the issue,
I am using the master branch (commit a1c0d9c3a1d1bea876f57c9229065ecd7f048b33), and everything is working fine.
What version of ESP-IDF are you using?

zhp0406, 2023-09-11:
@monurbstnc 
Can you provide me with the complete log, please?

monurbstnc, 2023-09-11:
I solve the problem, thanks for the helps.

closed 2023-09-11 by monurbstnc