espressif/esp-idf #12151: Wi-Fi Easy Connect not working on ESP32-S3 (DPP)

Stencht, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

I'm trying to implement the Wi-Fi Easy Connect (DPP) for my ESP32-S3 project using ESP-IDF v5.0, using the [dpp_enrollee example](https://github.com/espressif/esp-idf/blob/release/v5.0/examples/wifi/wifi_easy_connect/dpp-enrollee/main/dpp_enrollee_main.c).

The project compiles without issues and the program displays the QR-Code. When I scan the code and try to add the device, I get following output in the serial monitor:

```
I (888) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (928) wifi:mode : sta (f4:12:fa:db:91:d8)
I (928) wifi:enable tsf
I (938) wifi dpp-enrollee: Started listening for DPP Authentication
I (16848) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_FAILURE), retry...
```

If I keep on trying this it eventually works (sometimes 3 times, sometimes +10 times), but this is obviously not workable.
The same issue occurs with ESP-IDF v5.1 and v5.2 (master-branch).

II also tried setting some extra config options as described in #10615:

```
CONFIG_WPA_11KV_SUPPORT=y
CONFIG_WPA_SCAN_CACHE=y
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_DPP_SUPPORT=y
```

But this results in the same problem.
I've tried with multiple ESP32-S3's, which have no additional HW connected.

- Can someone run this successfully?
- Any idea what is going on?

agsagmf, 2024-01-27:
same issue here. Any update on this ?

kapilkedawat, 2024-02-02:
Hi @agsagmf, we are still working on mitigating this issue. 

The problem arises when a station is connected to an AP and attempts to initiate DPP on another channel with an ESP device. The issue is that it doesn't wait long enough for the ESP to send a reply. Since the station moves away from the channel before the ESP can send a response, leading to transmission failures of DPP packets and eventually DPP failure.

jgujarathi, 2024-02-14:
Hi @Stencht , 
We have a potential fix that has proved to be helpful in mitigating the occurrence of this issue. Can you please try this fix as well and let us know if it works in your environment?

Steps to test fix : 
1. Download and unzip the following zip file : 
[potential_fix.zip](https://github.com/espressif/esp-idf/files/14276382/potential_fix.zip)
2. Apply the patch titled dpp_tx_fail_fix.patch after executing a git pull on the release/v5.1 branch (git apply {path/to/file})
3. Replace the file libpp.a within the esp-idf repository  stored at components/esp_wifi/lib/esp32s3/ with the libpp.a file within the zip file. 
4. Build and run the DPP enrollee example to test the solution. 

Please note that this is a provisional solution only made for esp32s3 on the release/v5.1 branch. 
Please reach out if you have any additional queries.

jgujarathi, 2024-02-28:
Hi @Stencht , 
Have you had the change to try this out in your environment?

michael-betz, 2024-07-22:
One more data-point: DPP is also broken in esp-idf v5.2.1

ryanrampage1, 2024-07-30:
Any updates on this?

kapilkedawat, 2024-07-31:
Hi @michael-betz @ryanrampage1  apologies.. could you please try the fix mentioned in https://github.com/espressif/esp-idf/issues/10615#issuecomment-2250107634 and let us know if that helps?

AxelLin, 2024-08-06:
> Hi @michael-betz @ryanrampage1 apologies.. could you please try the fix mentioned in [#10615 (comment)](https://github.com/espressif/esp-idf/issues/10615#issuecomment-2250107634) and let us know if that helps?

@kapilkedawat 
Please take a look at
https://github.com/espressif/esp-idf/issues/10615#issuecomment-2260666752
https://github.com/espressif/esp-idf/issues/10615#issuecomment-2270656550

kapilkedawat, 2025-08-22:
@AxelLin Multiple changes have been made since then, and users have reported that it’s working much better now. Most of the issues users encounter are due to the initiator not staying on the channel long enough to receive a response (it should remain for around 500 ms as expected). Since older Espressif chips don’t have ECDH hardware, if the initiator doesn’t stay on the channel long enough, we could not form the response and send it back in time. However, if the ROC implementation in the initiator is correct and it stays on the channel for sufficient time, there should be no issue.

jmberhault, 2025-09-08:
Hello, about this feature, I have a similar trouble, for the plateform v5.5.1

The [dpp_enrollee example](https://github.com/espressif/esp-idf/blob/v5.5.1/examples/wifi/wifi_easy_connect/dpp-enrollee/main/dpp_enrollee_main.c) doesn't work with log=info, doesn't work with log=debug ans it's working with wpa debug messages.

For my application : it doesnt work with log = info, and it's working with log = debug and wpa debug messages. When subscribe to WIFI_EVENT_ACTION_TX_STATUS, the status is WIFI_ACTION_TX_OP_CANCELLED when log = info.

Some timing troubleshoot ?

jmberhault, 2025-09-08:
The patch is working for me. Thank you !