espressif/esp-idf #12152: No buffers error from OpenThread port otPlatSettingsAdd function

ShawnDSP, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.1-1-gd3c99ed3b8

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-DA | custom board

### Power Supply used.

USB

### What is the expected behavior?

I expected not to see any error messages from the OpenThread code.

### What is the actual behavior?

I have an application based on the OpenThread border router example code in examples/openthread/ot_br. I commission 5 non-router devices (sleepy end devices) to the border router, which is acting as a router/leader. If I then reboot the border router, I get some error messages like below:

```
E (131026) OPENTHREAD: otPlatSettingsAdd(185): No buffers, err: 261
W(131036) OPENTHREAD:[W] Settings------: Error NoBufs adding ChildInfo
E (131046) OPENTHREAD: dropping radio frame: Parse
```

### Steps to reproduce.

1. Start up border router
2. Commission 5 sleepy end devices to join the same OpenThread network as the border router (these are custom devices build with nRF52 chip)
3. Reboot the border router

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6180
load:0x40078000,len:15584
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
I (31) boot: ESP-IDF v5.1.1-1-gd3c99ed3b8 2nd stage bootloader
I (31) boot: compile time Aug 28 2023 12:16:40
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v3.0
I (37) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00008000
I (67) boot:  1 otadata          OTA data         01 00 00011000 00002000
I (74) boot:  2 phy_init         RF data          01 01 00013000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00020000 007ee000
I (87) boot:  4 ota_1            OTA app          00 11 00810000 007ee000
I (94) boot: End of partition table
I (97) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=41b8ch (269196) map
I (201) esp_image: segment 1: paddr=00061bb4 vaddr=3ffbdb60 size=05d24h ( 23844) load
I (211) esp_image: segment 2: paddr=000678e0 vaddr=40080000 size=08738h ( 34616) load
I (226) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=140a70h (1313392) map
I (700) esp_image: segment 4: paddr=001b0a98 vaddr=40088738 size=14ad0h ( 84688) load
I (750) boot: Loaded app from partition at offset 0x20000
I (750) boot: Disabling RNG early entropy source...
E (883) APP: Start count is 31
E (893) APP: Power on reason is 1
> > W(38373) OPENTHREAD:[W] Mle-----------: Failed to process Child ID Request: Security
E (41023) OPENTHREAD: otPlatSettingsAdd(185): No buffers, err: 261
W(41023) OPENTHREAD:[W] Settings------: Error NoBufs adding ChildInfo
E (41033) OPENTHREAD: dropping radio frame: Parse
```

### More Information.

The error log shows that the "No buffers" error is being returned by otPlatSettingsAdd at line 185 of components/openthread/src/port/esp_openthread_settings.c. From some debugging, I see that get_next_empty_index function not returning ESP_OK because the key it is searching for it already found in the list. So I suspect the problem is that the otPlatSettingsAdd function is not handling the case of adding values to an existing setting, as described here (https://openthread.io/reference/group/plat-settings#otplatsettingsadd).

zwx1995esp, 2023-08-31:
Oh, that should be a bug, could you please have a try with adding a clear for the variable `found` to `false` at the beginning of the `for loop`?   Maybe after the [esp_openthread_settings.c:line40](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/openthread/src/port/esp_openthread_settings.c#L40)

ShawnDSP, 2023-08-31:
@zwx1995esp, I moved the initialization of "found" from line 38 to just after the start of the for loop, like you suggested. I no longer see the "no buffers" error.

zwx1995esp, 2023-09-01:
> @zwx1995esp, I moved the initialization of "found" from line 38 to just after the start of the for loop, like you suggested. I no longer see the "no buffers" error.

Good news! You can use this fix in your local code first. I will create the bugfix MR in our internal branch ASAP, and will sync to you when the MR merged and sync to github. Please just wait.

ShawnDSP, 2023-09-01:
Okay, thanks for your help with this problem.

chshu, 2023-09-11:
Fixed by https://github.com/espressif/esp-idf/commit/99394d5791c8ccd0bfeca0b266ebf8e0baa6bff7

closed 2023-09-11 by chshu