espressif/esp-idf #12153: Setting nimBLE periodic extended adv data after starting advertising

IsitaB, 2023-08-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

I am using nimble periodic extended advertising on sync to send out advertisements based off this example https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_periodic_adv. I do want to change the advertisement data being sent as the program gets new data.

I have started extended periodic advertising, and it works. I want to set new advertising data now with the same config.

### What is the actual behavior?

When I try to set advertising data after I have begun advertising, it doesn't work. 
return code=3. It panic aborts and reboots.

### Steps to reproduce.

1. This [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_periodic_adv) that starts advertising
2. Once we have begin advertising, set new adv data. (for the sake of debugging, you could just start periodic advertising and then call a function that tries to set new adv data, overriding the initially set advertising data)

rahult-github, 2023-08-29:
Hi @IsitaB , 

Can you please share the application layer changes done which we can use at our end to reproduce the issue ?
Also, can you please share the complete console log and the elf file ?

rahult-github, 2023-09-01:
Hi @IsitaB , 

 I made a sample example change in ble_periodic_adv to modify adv data after some time and can see the data being changed. You can have a look at the example and let me know if you have any queries. Please also note, that you will need to increase the value of NIMBLE_MAX_EXT_ADV_INSTANCES if you are adding more instances as default value for it is 1 in SDK. 
 
 
[example_changes.txt](https://github.com/espressif/esp-idf/files/12495948/example_changes.txt)

IsitaB, 2023-09-01:
Hello @rahult-github,

I see that you mentioned `number of instances`. Does this mean that every time I want to send out new information, I have to stop extended and periodic advertising and restart them with new data?

Also, I see that in your example, you're only using periodic advertising. In the [example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/ble_periodic_adv/main/main.c), it uses both periodic and extended advertising. I'm assuming I'm going to have to set the data for both? Or does periodic advertising support extended advertising (255 bytes)?

rahult-github, 2023-09-04:
Hi @IsitaB ,

> I see that you mentioned number of instances. Does this mean that every time I want to send out new information, I have to stop extended and periodic advertising and restart them with new data?

The example changes shared is one way to achieve the change in data. So, to understand what you were attempting, we had requested to share your application changes. 

> Also, I see that in your example, you're only using periodic advertising. In the [example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/ble_periodic_adv/main/main.c), [...]

Periodic advertisement basically will work only if extended adv is enabled. So  can you please elaborate on the query of using "only perdiodic advertisement"  . If the intention is to do multiple advertisement using extended adv , then i suggest you can look at the example [ble_multi_adv](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_multi_adv) . 

> Or does periodic advertising support extended advertising (255 bytes)?

252 bytes data for nimble.

IsitaB, 2023-09-06:
Okay. Thank you.

My changes:
I added these lines to the end of `app_main()` in the example

```
    sleep(20);
    ESP_LOGI(tag, "hit 20s");

    struct os_mbuf *data;
    static uint8_t new_periodic_adv_raw_data[] = {'C','O','O','K','I','E','S'};
    
    data = os_msys_get_pkthdr(sizeof(new_periodic_adv_raw_data), 0);
    assert(data);

    rc = os_mbuf_append(data, new_periodic_adv_raw_data, sizeof(new_periodic_adv_raw_data));
    assert(rc == 0);
    rc = ble_gap_periodic_adv_set_data(instance, data);
    assert (rc == 0);
    ESP_LOGI(tag, "rc %d", rc);
```

so my app_main() looks like this
```
void
app_main(void)
{
    int rc;
    /* Initialize NVS — it is used to store PHY calibration data */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ret = nimble_port_init();
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "Failed to init nimble %d ", ret);
        return;
    }

    /* Initialize the NimBLE host configuration. */
    ble_hs_cfg.reset_cb = periodic_adv_on_reset;
    ble_hs_cfg.sync_cb = periodic_adv_on_sync;
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;
    /* Set the default device name. */
    rc = ble_svc_gap_device_name_set("nimble_periodic_adv");
    assert(rc == 0);

    /* XXX Need to have template for store */
    ble_store_config_init();

    nimble_port_freertos_init(periodic_adv_host_task);
    sleep(20);
    ESP_LOGI(tag, "hit 20s");

    struct os_mbuf *data;
    static uint8_t new_periodic_adv_raw_data[] = {'C','O','O','K','I','E','S'};
    
    data = os_msys_get_pkthdr(sizeof(new_periodic_adv_raw_data), 0);
    assert(data);

    rc = os_mbuf_append(data, new_periodic_adv_raw_data, sizeof(new_periodic_adv_raw_data));
    assert(rc == 0);
    rc = ble_gap_periodic_adv_set_data(instance, data);
    assert (rc == 0);
    ESP_LOGI(tag, "rc %d", rc);
}
```
I do not want to do multi-adv. Every time my bt device gets new information; the device should advertise the new data, replacing the old advertisement data.

Let's backtrack

1. Confirm initial periodic extended advertisement data is being sent correctly
2. Set new advertisement data
3. Confirm new advertisement information is being sent correctly

How do I do '1 Confirm initial periodic extended advertisement data is being sent correctly`? I used a bt scanner app and it shows length of advertisement is 13. When you run the example, is this what you see on your end? I wrote an Android app to get the advertisement data, and it only logs the name set on line 80. Maybe my esp32 project setup is wrong. Please let me know. 

I think this would be a good starting point

rahult-github, 2023-09-08:
Hi @IsitaB 

The sequence of API being invoked in your examples is correct. 

But If i take the changes you mentioned above as is, am facing below compilation issue

```
../main/main.c: In function 'app_main':
../main/main.c:226:5: error: implicit declaration of function 'sleep' [-Werror=implicit-function-declaration]
  226 |     sleep(20);
      |     ^~~~~
../main/main.c:237:40: error: 'instance' undeclared (first use in this function)
  237 |     rc = ble_gap_periodic_adv_set_data(instance, data);
      |                                        ^~~~~~~~
../main/main.c:237:40: note: each undeclared identifier is reported only once for each function it appears in
```

Here, the intention is:
1. Wait for sometime after advertising is started and change then data
2. Use the same instance as previous for updated adv data. 

I fixed the application for above. Please find attached example_changes.txt 
[[example_changes.txt](https://github.com/espressif/esp-idf/files/12557451/example_changes.txt) ] i made which compile cleanly for above mentioned changes. 

With this change, i do not see any issue. The application runs cleanly and doesn't crash. On remote side i am able to see the adv data being changed from esp-periodic-adv to cookies and events are received correctly.

[v5.1_adv_logs.txt](https://github.com/espressif/esp-idf/files/12557460/v5.1_adv_logs.txt)
[v5.1_sync_logs.txt](https://github.com/espressif/esp-idf/files/12557461/v5.1_sync_logs.txt)

So, there doesn't seem to be problem with the way code is implemented, as the same example of yours works cleanly and shows the intended behaviour. 

Can you please ensure the codebase is cleanly synced ?  Also, is there any other change ( in example OR configuration change ) that has been done ? 
 
> How do I do '1 Confirm initial periodic extended advertisement data is being sent correctly`? 
  
One way is to run the ble_periodic_sync application in debug mode and you can see the adv data information Attached log for same 

> I used a bt scanner app and it shows length of advertisement is 13. When you run the example, is this what you see on your end?

 you can refer to attached log for complete details of received packet

IsitaB, 2023-09-12:
I do have the latest version of the codebase. At this point, I'm confirming the example works right. I set up my esp idf with vscode using the esp idf extension -- when configuring it, I selected its source to be github's master branch, so it gets the latest version of the master branch.

However, how do I check that I have the right bt permissions?

If I run, idf.py menuconfig, 

Component Config > Bluetooth > Host (Nimble - BLE only) -- that's enabled
Component Config > Bluetooth > BLE 5 feature -- everything under that is enabled as well. Max length of advertising data is 1650 

How do I confirm that I have this value configured/enabled `#if MYNEWT_VAL(BLE_PERIODIC_ADV_ENH)`. How do I confirm that the values in my `Kconfig.projbuild` are configured correctly?

I do not have a secondary esp32 device to run periodic_sync_demo. 

If you run ble_periodic_adv and try to detect it on the nrf connect app https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US&gl=US&pli=1

What do you see when you click on `RAW`? What's the length of the data sent?

Please let me know. Thank you.

IsitaB, 2023-09-12:
Here are the contents of my sdkconfig
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12590037/sdkconfig.txt)

sdkconfig.defaults file contents
```
# Override some defaults so BT stack is enabled
# in this example

#
# BT config
#
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_EXAMPLE_EXTENDED_ADV=y
CONFIG_BT_NIMBLE_EXT_ADV=y
CONFIG_BT_NIMBLE_MAX_PERIODIC_SYNCS=1
```

Kconfig.projbuild has these deps listed. I don't see half of them in my sdkconfig. The selected ones are the ones that I do have

- [x] EXAMPLE_EXTENDED_ADV
- [x] SOC_BLE_50_SUPPORTED
- [ ] EXAMPLE_RANDOM_ADDR <-- 
- [ ] EXAMPLE_PERIODIC_ADV_ENH <---
- [x] SOC_BLE_50_SUPPORTED
- [ ] SOC_BLE_PERIODIC_ADV_ENH_SUPPORTED <-- 
- [ ] BT_NIMBLE_PERIODIC_ADV_ENH <--

Not having example_random_addr is probably fine, but not having perms related to example_periodic_adv_enh is probably the problem.

rahult-github, 2023-09-14:
Hi @IsitaB ,

So just to clarify, periodic adv is a BT 5.0 feature.  Your original query reported esp32c3 as the chip, but description is mentioning esp32.  This feature will work for esp32c3, but not for esp32. This is mentioned even in the README file for the examples. 

> How do I confirm that I have this value configured/enabled 

Once a setting is configured via menuconfig, it can be viewed in sdkconfig to double confirm .  You can refer to project [configuration documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#) for more details. 

> What do you see when you click on RAW? What's the length of the data sent? On my end, it only picks up on the advertisement name, not the data. Although there are no errors after the data-sending part runs rc = 0.

This seems to be nrf connect application issue. It is showing the extended packet , but we see no option to create sync and hence no further information is seen.  

If we do periodic advertisement from esp32c3 side, an OTA for it shows the packet being sent in the intended format. Attaching snapshot for same. 

[image]

>  Kconfig.projbuild has these deps listed. I don't see half of them in my sdkconfig. The selected ones are the ones that I do have

So Kconfig.projbuild is used to setup configuration used only in examples.  The default menuconfig will show up all configurations ( both stack and examples ).  You can search for a configuration in menuconfig and press "?" to get its help.You can see the dependancy options too , to check why the option is being seen or not seen in normal menuconfig. 

>  BT_NIMBLE_PERIODIC_ADV_ENH 

This is for periodic adv enhancement which is part of the newer 5.3 spec. Please don't confuse this with periodic adv. For periodic adv, when BLE5.0 features are selected, the option BT_NIMBLE_ENABLE_PERIODIC_ADV will be enabled ( this is seen in the sdkconfig snapshot you have provided)

IsitaB, 2023-09-15:
Hi @rahult-github,

I misspoke with esp32, I am working with an esp32c3 board. 

I see it was a sync problem.

I ran the nimble bleprph example. Enabled all the example configuration requirements, and ext adv worked for me there. I'm not using the ble_periodic_adv example anymore, but I think the reason it never worked is the underlying config and the syncing part.

rahult-github, 2023-09-17:
Ok . Good to know. Closing issue. Feel free to reopen if needed.

closed 2023-09-17 by rahult-github