espressif/esp-idf #12154: Enable otNetworkTimeGet from OpenThread in ESP-IDF

Ereaey, 2023-08-29:
### Is your feature request related to a problem?

The problem I am facing is related to the use of otNetworkTimeGet from the OpenThread stack. ESP-IDF does not currently support this function, which is crucial for my application as it requires synchronization with the network time. Although it is possible to set up an SNTP server as an alternative, it is a more time-consuming and resource-intensive approach compared to using otNetworkTimeGet.

### Describe the solution you'd like.

I would like ESP-IDF to support the otNetworkTimeGet function from the OpenThread stack. This would significantly simplify the development process by avoiding the need to set up an SNTP server or implement another network time synchronization mechanism. Supporting this function will also make ESP-IDF more compatible with OpenThread and enhance its capabilities.

### Describe alternatives you've considered.

I have considered setting up an SNTP server as an alternative, but this approach is time-consuming and resource-intensive. It also increases the development time and maintenance efforts. Another alternative is to use a different chipset or platform that supports otNetworkTimeGet, but this would involve changing the hardware and software architecture of my application, which is not feasible at this stage of development.

### Additional context.

Supporting otNetworkTimeGet in ESP-IDF will not only benefit my application but also others that require network time synchronization. It is a fundamental feature for many applications and will make ESP-IDF more versatile and compatible with the OpenThread stack.

chshu, 2023-09-11:
@Ereaey The `otNetworkTimeGet()` is part of the time sync feature in openthread, it allows devices in the same network to synchronize to a single time, the time souce is from Leader.

The time sync feature is independent of SNTP, unless the Leader implements some mechanism and snyc to the network time periodically.

Note that this feature is experimental and not part of the Thread Specification.

Ereaey, 2023-09-11:
> 

Yes, you are completely right.
The SNTP server is not a viable solution for my example, as I want to precisely synchronize the cards in my openthreaded network with each other.
The simplified idea being to vary the lights at the same time

Ereaey, 2023-09-11:
> @Ereaey The `otNetworkTimeGet()` is part of the time sync feature in openthread, it allows devices in the same network to synchronize to a single time, the time souce is from Leader.
>
> The time sync [...]

Yes, you are completely right.
The SNTP server is not a viable solution for my example, as I want to precisely synchronize the cards in my openthreaded network with each other.
The simplified idea being to vary the lights at the same time

chshu, 2023-09-11:
@Ereaey Ok, we will enable the time synce feature on ESP platform, before the end of Sept.

chshu, 2023-10-24:
@Ereaey The Time Sync has been supported since this commit: https://github.com/espressif/esp-idf/commit/265288190027c3430ba9d92462720e9fa8239adb

You can simply enable it with `OPENTHREAD_TIME_SYNC` option. Feel free to have a try, let me know if any issues.

closed 2023-10-24 by chshu

SimeonAT, 2024-06-22:
Hi @chshu,

I am also trying to get Network Time Synchronization to work with my OpenThread devices. However, I'm getting build errors when I enable `OPENTHREAD_TIME_SYNC` on the ESP Thread Border Router SDK. I document the specific issues I found in Issue [14055](https://github.com/espressif/esp-idf/issues/14055).

If time permits, any steps that you can give me to help me debug this issue would greatly be appreciated.

Thank you,