espressif/esp-idf #12156: "FTM Procedure with peer failed" ( status-2) - ESP32-C3 -fails to perform ftm procedure with ftm responder when conneted with external non-esp station

Bggopal, 2023-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

Connect to an Non ftm AP, and try to perform FTM procedure,
should give the distance using RTT formula
**1. This feature works when connected to SoftAP ( which is not FTM enabled) in ESP32
2. And Not works when using non esp routers**

### What is the actual behavior?

In the presence of three esp32s , and one non-esp router,

1. ESP1 ( softAp - not an FTM enabled device)  connected to ESP2 (FTM initiator) responds to ESP3( FTM responder) with a distance
2. Router1 ( not an FTM enabled device)  connected to ESP2 (FTM initiator) responds to ESP3( FTM responder) with  FTM Procedure with peer failed 

### Steps to reproduce.

Used to 3 ESP32 and one non-esp router
1. Connect ESP1 ( softAp - not an FTM enabled device) with ESP2 (FTM initiator) and perform FTM procedure with ESP3

will result in giving distance 

2. Connect Router1 ( not an FTM enabled device)  connected to ESP2 (FTM initiator)  and perform FTM procedure with ESP3

will result with  " FTM Procedure with peer failed "

### Debug Logs.

```plain
ftm> scan
I (6975) ftm_station: sta start to scan
I (9475) ftm_station: [OnePlus7][rssi=-22]
I (9475) ftm_station: [ESP32C3_PNS1][rssi=-44][FTM Responder]
I (9585) ftm_station: sta scan done

ftm> ftm -I -s ESP32
I (27495) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
W (27495) wifi:Starting FTM session with 84:f7:03:40:75:65 in 0.023 Sec
W (27505) wifi:Mode: non-ASAP, Bursts: 8, FTM's per burst: 4, Burst Period: 200mSec, Burst Duration: 32000uSec
W (28965) wifi:FTM session ends with 30 valid readings out of 31, Avg raw RTT: 625.715 nSec, Avg RSSI: -44
I (28975) ftm_station: Estimated RTT - 514 nSec, Estimated Distance - 77.10 meters

ftm> sta ESP32C3_PNS1 12345678
I (77265) ftm_station: sta connecting to 'ESP32'
I (77975) ftm_station: Connected to ESP32C3_PNS1 (BSSID: 84:f7:03:40:75:65, Channel: 1)
ftm>
ftm>  ftm -I -s ESP32
I (92225) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
W (92225) wifi:Starting FTM session with 84:f7:03:40:75:65 in 0.008 Sec
W (92225) wifi:Mode: non-ASAP, Bursts: 8, FTM's per burst: 4, Burst Period: 200m Sec, Burst Duration: 32000uSec
W (93665) wifi:FTM session ends with 31 valid readings out of 31, Avg raw RTT: 3 39.905 nSec, Avg RSSI: -46
I (93665) ftm_station: Estimated RTT - 228 nSec, Estimated Distance - 34.20 meters

//when connecting to a non-ftm (not an esp32 device) 
ftm> sta OnePlus7 password
I (126115) ftm_station: sta connecting to 'OnePlus7'
I (126115) ftm_station: sta disconnected
I (127565) ftm_station: Connected to OnePlus7 (BSSID: xx:xx:xx:xx:xx:xx, Channel : 6)
ftm>  ftm -I -s ESP32C3_PNS1
I (131785) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
I (132285) ftm_station: FTM procedure with Peer(84:f7:03:40:75:65) failed! (Status - 2)
ftm>
```

### More Information.

I tried to connecting to different routers (including mobile hotspot all failed), where as all the esp32 gave me results,

Also in the issue id discussed #8976 which is for S3, but looking for the same in esp32-c3

@sarveshb14 can you look into this, as I have seen your tag for last fixes in S3

Bggopal, 2023-09-03:
Ok, in simple words i could say, any other router which is ftm enabled will show as a ftm responder during scan, but when trying to do ftm to fetch data , it thorws error as peer failed. If it is only ftm enabled from esp32 it fetches the data. This could give good insight on the issue, i faced. I verified with two are three routers, all were same.

sarveshb14, 2026-09-03:
Hi @Bggopal , Apologies for the delayed response. Over the past few years, we have fixed several FTM issues and made significant improvements to the implementation. Could you please retest using the latest version of any currently maintained ESP-IDF branch and let us know if the issue still persists?

closed 2026-09-03 by Bggopal