espressif/esp-idf #12157: Python error when exiting idf_monitor

KaeLL, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Describe: v5.1.1-1-gd3c99ed3b8

Pressing Ctrl + ] to close idf_monitor result in:
```
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/tools/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages/esp_idf_monitor/base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/tools/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages/esp_idf_monitor/base/serial_reader.py", line 100, in run
    self.close_serial()
  File "/tools/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages/esp_idf_monitor/base/serial_reader.py", line 105, in close_serial
    self.set_closing_wait(delay_sec=1)
  File "/tools/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages/esp_idf_monitor/base/serial_reader.py", line 128, in set_closing_wait
    fcntl.ioctl(self.serial.fd, termios.TIOCSSERIAL, buf)
PermissionError: [Errno 1] Operation not permitted

Done
```

peterdragun, 2023-08-30:
Hi, I am sorry for the issue. We are aware of it and we currently have a PR under review that implements the fix.
Possible workarounds are:
1. Downgrading esp-idf-monitor to a version lower than 1.1.3 - `pip install esp-idf-monitor==1.1.2`
2. Running monitor as a root - I would recommend doing this just in case  you really need a newer version that fixes issues on new kernels (e.g. waiting for 30 seconds before the serial port gets closed)

KaeLL, 2023-08-30:
Just make sure the fix will be available in `release/v5.1`.

peterdragun, 2023-08-30:
No worries, it will be definitely available in ESP-IDF 5.1. As the monitor is now a separate package we can release it without needing to wait for ESP-IDF. I will let you know once the new version with the fix is available.

dobairoland, 2023-09-07:
This can be resolved by running
```sh
pip install "esp-idf-monitor>=1.2.0"
```
in active ESP-IDF environments.

Or just re-run the install and export scripts again.

closed 2023-09-07 by dobairoland

4444monkey, 2023-12-25:
Hi there,
Since I had a similar issue on version 5.1.2, can the issue be resolved by running the lower version like pip install esp-idf-monitor==1.1.2? Has this issue been fully solved in version >=5.1?

peterdragun, 2023-12-25:
Hi @4444monkey this issue should be resolved completely in the latest versions of esp-idf-monitor (included in IDF 5.1). You can make sure that you have the latest version by running install and export scripts. If you still see the issue, please provide more details and open a new issue in [esp-idf-monitor repository](https://github.com/espressif/esp-idf-monitor) and we will take a look. Installing the mentioned lower version was just a workaround and should not be needed anymore.