espressif/esp-idf #12159: New I2C Master driver i2c_master_probe() returning ESP_OK for all addresses

ammaree, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2383-g82cceabc6e

### Operating System used.

macOS

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom hardware

### Power Supply used.

External 5V

### What is the expected behavior?

API to return ESP_OK only when a device is found on a bus.

### What is the actual behavior?

Returns ESP_OK for all addresses between 0x08 and 0x77

### Steps to reproduce.

static int halI2C_CheckAddress(u8_t eCh, u16_t Addr) {
	#if (buildI2C_MASTER_DRIVER == 0)
	u8_t i2cBuf[I2C_LINK_RECOMMENDED_SIZE(1)];
	i2c_cmd_handle_t i2cCmd = i2c_cmd_link_create_static(i2cBuf, sizeof(i2cBuf));
	i2c_master_start(i2cCmd);
	i2c_master_write_byte(i2cCmd, (Addr << 1) | I2C_MASTER_WRITE, ACK_CHECK_EN);
	i2c_master_stop(i2cCmd);
	int iRV = i2c_master_cmd_begin(eCh, i2cCmd, 10);
	i2c_cmd_link_delete_static(i2cCmd);
	return iRV;

	#elif (buildI2C_MASTER_DRIVER == 1)
	return i2c_master_probe(bus_handle[eCh], Addr, i2cTIMEOUT_DEFAULT);

	#endif
}

int	halI2C_CountDevices(u8_t eCh) {
	u8_t DevCount = 0;
	for (u16_t Addr = I2C_LOWEST_ADDR; Addr < I2C_TENBIT_ADDRESS; ++Addr) {
		if (halI2C_CheckAddress(eCh, Addr) == ESP_OK) ++DevCount;
	}
	return DevCount;
}

mythbuster5, 2023-08-30:
I noticed this issue already, will find time to fix it.

mythbuster5, 2023-08-30:
BTW, you can use i2c_tool in example for probe an I2C device

ammaree, 2023-09-01:
@mythbuster5 
@o-marshmallow 
@igrr 

Any chance we can get some focus on this bug, sounds like it is known and simple.

We would really like to try the new I2C Master driver but due to our I2C device auto discovery process using the probe functionality, our testing has been very restricted.

Even if you can send me a patch to get it going, we will be able to stress test it on a wide range of platform and sensors.

Thanks

André

mythbuster5, 2023-09-02:
@ammaree I have already merged changes in internal repo (it needs time to update on github). Let's see if it doesnt sync on github monday. I will append a patch here.

o-marshmallow, 2023-09-02:
@ammaree In case you are really in a hurry, apply the patch that @mythbuster5 merged on master:
[i2c_master.txt](https://github.com/espressif/esp-idf/files/12503168/i2c_master.txt)

ammaree, 2023-09-03:
@mythbuster5 @o-marshmallow 

I have tried with the patches but now it hangs up earlier, around lines 882 or 883 of i2c_new_master_bus()

Previously my call to i2c_new_master_bus() passed perfectly and it only crashed later at the calls to i2c_master_probe()

I am happy to test more patches since I am under pressure to get the I2C fixed.

Thanks

mythbuster5, 2023-09-04:
Can I see how did you call the functions? I want to see the function calling order

closed 2023-09-04 by espressif-bot

mythbuster5, 2023-09-04:
this is closed by commit command directly, feel free to reopen if you still have any problems

ammaree, 2023-09-04:
@mythbuster5 @o-marshmallow 

I am not currently next to the device to enable a higher level of logging (currently INFO) but when tested last night no error messages were shown, device just hung as the lines indicated, never returned in spite of waiting ~15minutes

Entry point for the I2C device discovery is at halI2C_DevicesDiscover ().
Calling order as follows:
```
const halI2C_Init_t	halI2C_Init[halI2C_NUM] = {
	[halI2C_0] = {
		.i2c_port = I2C_NUM_0,
		.sda_io_num = GPIO_NUM_26,
		.scl_io_num = GPIO_NUM_27,
		.clk_source = I2C_CLK_SRC_DEFAULT,
		.flags.enable_internal_pullup = 1,
	},
}

int	halI2C_DevicesDiscover(void) {
	for (u8_t eCh = 0; eCh < halI2C_NUM; ++eCh) {
		halI2C_BusConfig(eCh);							// general bus init
		halI2C_GeneralCall(eCh, I2C_GENCALL_RESET_WRITE);
		for (u16_t Addr = I2C_LOWEST_ADDR; Addr < I2C_TENBIT_ADDRESS; ++Addr) {
			int iRV = halI2C_CheckAddress(eCh, Addr);
			if (iRV == ESP_OK) {
				if (i2cDevCount < i2cMAX_DEVICES) {
					i2c_di_t * psI2C = &sI2C_DI[i2cDevCount++];
					psI2C->Port = eCh;
					psI2C->Addr = Addr;
				} else {
					break;
				}
			}
		}
	}
	i2cDevCount = (i2cDevCount < i2cMAX_DEVICES) ? i2cDevCount : i2cMAX_DEVICES;
	return i2cDevCount;
}

void halI2C_BusConfig(u8_t eCh) {
	const halI2C_Init_t * psI2C_Init = &halI2C_Init[eCh];
	int iRV = i2c_new_master_bus(psI2C_Init, &hI2Cbus[eCh]);
        printfx("I2C Port:%d  SCL=%d  SDA=%d  PU%s iRV=%d\r\n",
		psI2C_Init->i2c_port, psI2C_Init->scl_io_num, psI2C_Init->sda_io_num,
		psI2C_Init->flags.enable_internal_pullup ? "en" : "dis", iRV);
}

void halI2C_GeneralCall(u8_t eCh, u8_t cCmd) {
	i2c_di_t * psI2C = &sI2C_DI[eCh];
	i2c_device_config_t dev_cfg = {
		.dev_addr_length = I2C_ADDR_BIT_LEN_7,
		.device_address = I2C_GENCALL_ADDRESS,
		.scl_speed_hz = 100000,
	};
	i2c_master_dev_handle_t hndl;
	ESP_ERROR_CHECK(i2c_master_bus_add_device(hI2Cbus[psI2C->Port], &dev_cfg, &hndl));
	ESP_ERROR_CHECK(i2c_master_transmit(hndl, &cCmd, sizeof(u8_t), i2cTIMEOUT_DEFAULT));
	ESP_ERROR_CHECK(i2c_master_bus_rm_device(hndl));
}

int halI2C_CheckAddress(u8_t eCh, u16_t Addr) {
	return i2c_master_probe(hI2Cbus[eCh], Addr, i2cTIMEOUT_DEFAULT);
}
```

ammaree, 2023-09-04:
@o-marshmallow @mythbuster5 
Here is the complete log at highest level. As far as I can see not messages relating to I2C.
At the end the device is just hanging, not responding to console input, no output, no watchdog timeout.

`
0.017: #0 main vfs esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 4
0.020: #0 main wifi wifi driver task: 3ffd295c, prio:23, stack:3584, core=0
0.025: #0 wifi wifi wifi firmware version: 341633f
0.026: #0 wifi wifi wifi certification version: v7.0
0.027: #0 wifi wifi config NVS flash: enabled
0.028: #0 wifi wifi config nano formating: disabled
0.029: #0 wifi wifi Init data frame dynamic rx buffer num: 8
0.030: #0 wifi wifi Init management frame dynamic rx buffer num: 8
0.031: #0 wifi wifi Init management short buffer num: 32
0.032: #0 wifi wifi Init dynamic tx buffer num: 8
0.033: #0 wifi wifi Init static rx buffer size: 1600
0.034: #0 wifi wifi Init static rx buffer num: 4
0.034: #0 wifi wifi Init dynamic rx buffer num: 8
0.036: #0 wifi wifi hint=255, act_dur=<0,120>, pas_dur=360, nchans=14 history_num=0
0.037: #0 wifi wifi Set ps type: 0, coexist: 0

1.291: #0 wifi wifi clear blacklist
1.368: #0 wifi wifi ht40u freq=2422, chan=1
1.370: #0 wifi wifi filter: set rx policy=0
1.371: #0 wifi wifi mode : sta (c0:49:ef:c8:ad:84) + softAP (c0:49:ef:c8:ad:85)
1.373: #0 wifi wifi enable tsf
1.373: #0 wifi wifi filter: set rx policy=1
1.374: #0 wifi wifi connect status 0 -> 0
1.375: #0 wifi wifi filter: set rx policy=9
1.376: #0 wifi wifi ht40u freq=2422, chan=1
1.379: #0 wifi wifi Total power save buffer number: 4
1.380: #0 wifi wifi Init max length of beacon: 752/752
1.381: #1 wifi wifi Repeated 1x
1.382: #0 wifi wifi Start wifi connect
1.383: #1 main halRTC_GetTime IDF Time is 2023-09-04T09:39:42Z (SNTP flag=0)

1.383: #0 wifi wifi connect status 0 -> 0
1.385: #0 wifi wifi connect chan=0
1.386: #0 wifi wifi nvs=0, ssid=139Algernon, channel=255
1.386: #0 wifi wifi ssid=139Algernon match nvs 0, channel=255
1.386: #1 main SysFlag APrvrt BufOK LFS APP (x69)
1.387: #0 wifi wifi first chan=1
1.389: #1 main EvtFlag TIMER L1 (x801)
1.389: #0 wifi wifi connect status 0 -> 1
1.390: #1 main RunFlag 2/0x4 (x4)
1.391: #0 wifi wifi filter: set rx policy=3
1.392: #0 wifi wifi clear scan ap list
0x400fab90: cnx_start_handoff_cb at ??:?0f, priority=2, cb=0x400fab90, arg=0x00000000, ss_state=0x1, time=24702, index=0

1.395: #0 wifi wifi inter_channel_timeout: arg=0x00000000, ss_state=0x1
1.396: #0 wifi wifi perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
1.397: #0 wifi wifi change: chan<1,0>, dur<0,120>
1.398: #0 wifi wifi enter start op, arg=0x3ffbaaf6
1.399: #0 wifi wifi scan operation: state=0x3, chan<1,0>, arg=0x3ffbaaf6, status=0
1.400: #0 wifi wifi scan specific ssid=139Algernon
1.401: #0 wifi wifi start max timer
1.492: #0 main RunFlag 2/0x4 1/0x2 (x6)
1.522: #0 wifi wifi scan end: arg=0x00000000, status=0, ss_state=0x3
1.523: #0 wifi wifi back home chan=<1,1>, current chan=<1,1>
1.524: #0 wifi wifi start scan timer
1.555: #0 wifi wifi oper_channel: arg=0x00000000 time=186913
1.556: #0 wifi wifi perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
1.557: #0 wifi wifi change: chan<2,0>, dur<0,120>
1.558: #0 wifi wifi ht20 freq=2417, chan=2
1.559: #0 wifi wifi enter start op, arg=0x3ffbaaf6
1.560: #0 wifi wifi scan operation: state=0x3, chan<2,0>, arg=0x3ffbaaf6, status=0
1.562: #0 wifi wifi scan specific ssid=139Algernon
1.563: #0 wifi wifi start max timer
1.684: #0 wifi wifi scan end: arg=0x00000000, status=0, ss_state=0x3
1.685: #0 wifi wifi back home chan=<1,1>, current chan=<2,0>
[... 197 lines cut ...]
5.734: #0 wifi wifi state: auth -> assoc (0)
5.735: #0 wifi wifi restart connect 1s timer for assoc
5.748: #0 wifi wifi recv assoc: type=0x10
5.749: #0 wifi wifi filter: set rx policy=6
5.750: #0 wifi wifi state: assoc -> run (10)
5.751: #0 wifi wifi start 10s connect timer for 4 way handshake
5.754: #0 wifi wifi wpa_psk start
5.756: #0 wifi wifi wpa_psk handle succeed
5.765: #0 wifi wifi wpa_psk start
5.768: #0 wifi wifi wpa_psk handle succeed
5.770: #0 wifi wifi connected with 139Algernon, aid = 2, channel 13, BW20, bssid = d4:ca:6d:96:fe:08
5.771: #0 wifi wifi security: WPA2-PSK, phy: bgn, rssi: -49
5.772: #0 wifi wifi remove all except d4:ca:6d:96:fe:08 from rc list
5.774: #0 wifi wifi clear blacklist
5.774: #0 wifi wifi filter: set rx policy=7
5.775: #0 wifi wifi pm start, type: 0

5.776: #0 wifi wifi Send sta connected event
5.777: #0 wifi wifi connect status 1 -> 5
5.778: #0 wifi wifi obss scan is disabled
5.778: #0 wifi wifi start obss scan: obss scan is stopped
5.820: #0 main EvtFlag TIMER L2sta L1 (x803)
5.851: #0 wifi wifi AP's beacon interval = 102400 us, DTIM period = 1
6.286: #0 wifi wifi <ba-add>idx:0 (ifx:0, d4:ca:6d:96:fe:08), tid:0, ssn:3, winSize:64
`

mythbuster5, 2023-09-05:
Oh, I reproduce the issue, the root cause is that one interrupt is lacked. The fix is on the way

ammaree, 2023-09-05:
I am very glad you found it. I have spent 4 days trying to make this work, gave up around 01h00 this morning.
Welcomer to send the patch and I will give it a try.

mythbuster5, 2023-09-05:
Feel very sorry and apologize for this.Try with this please~ 
[i2c_master.txt](https://github.com/espressif/esp-idf/files/12521375/i2c_master.txt)

ammaree, 2023-09-05:
I have up to now applied the patch manually, editing line by line.
What would be the correct git command to apply the patch automatically?

ammaree, 2023-09-05:
Don't worry, very small, applied..

mythbuster5, 2023-09-05:
[i2c_master.txt](https://github.com/espressif/esp-idf/files/12522389/i2c_master.txt)
Yeah.. usually patch file needs two blank line at the end of file, but some sometime it might lack one line...BTW  can `i2c_master_probe` work now?

ammaree, 2023-09-05:
It now runs through, does not hang anymore, but ...
It does not find any devices on a system where these are 3, 2x PCF8574 (0x22 & 0x24) and 1x DS1307 (0x68)

mythbuster5, 2023-09-05:
let me check

mythbuster5, 2023-09-05:
Then it's very weird. It works well on my board. I can show you the very simple testing code and result. (I connect two devices on one bus)

```
i2c_master_bus_config_t i2c_bus_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = 0,
        .scl_io_num = 4,
        .sda_io_num = 5,
    };
    i2c_master_bus_handle_t bus_handle;

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_bus_config, &bus_handle));

    esp_err_t ret;
    uint8_t i;
    for (i =0x01; i < 0x90; i++) {
        ret =i2c_master_probe(bus_handle, i, -1);
        if (ret == 0) {
            printf("bingo! addr: %x\n", i);
        }
    }
```
[image]

mythbuster5, 2023-09-05:
What you can do maybe check on a logic analyzer and see if the sending address is exist, there should be an `ack` on the bus, otherwise, will be `nack`.

mythbuster5, 2023-09-06:
Which esp chip are you using. I tried esp32c3 and esp32 on my side. Both of them work pretty well..

ammaree, 2023-09-06:
ESP32 Rev 3

puppy112358, 2024-02-19:
Have this problem been solved ?I've met the same problem yet---I used i2c_master_probe to detect the address and it returns ESP_OK with all address.

R0nyx, 2024-02-23:
I also can't find any devices on my i2c bus. Also following "https://github.com/espressif/esp-idf/issues/12973" 
i've tried to use `i2c_master_transmit`  command yet it succeeds on any address. How can it succeed if the documentation states that ACK should parsed from the slave?

R0nyx, 2024-02-27:
@mythbuster5 Have you tried using the other pins for the i2c?

mythbuster5, 2024-02-28:
Yes @R0nyx What pin you are using. Did you pull up?

R0nyx, 2024-02-28:
> Yes @R0nyx What pin you are using. Did you pull up?

@mythbuster5 I'm using pin 2 and 8 on esp32c3 wroom 2.0 
I did pull up too.
Initially I tried probing at the address i knew existed and couldn't find anything. When I used transmit I got confirmation that everything was successful, but unfortunately no registers were changed. 

I'm fairly certain that I'm doing the right thing since with legacy i2c.h everything works flawlessly

mythbuster5, 2024-02-28:
@R0nyx I will probe a i2c device on my side to check. For transmit return everything true, this is I already know and I will find time to get it fixed.

mythbuster5, 2024-02-28:
@R0nyx Probe works on my side on master. Ok, what IDF version you are using and what pins you are using?

R0nyx, 2024-02-28:
@mythbuster5 do you mind sharing your code example?

R0nyx, 2024-02-28:
@mythbuster5  i need to use pins 2 and 8

R0nyx, 2024-02-28:
@mythbuster5  also im using 5.2

R0nyx, 2024-02-28:
Im using esp32c3 wroom 2.0 

Before everything i also configure and reset the pins:
```
/**
 * @brief Resets all GPIO pins used in the I2C configuration.
 */
void reset_all_pins()
{
    gpio_reset_pin(I2C_DATA_PIN);
    gpio_reset_pin(I2C_CLK_PIN);
    gpio_reset_pin(GPIO_EXP_EN);
    gpio_reset_pin(I2C_EN);
    gpio_reset_pin(GPIO_NUM_1);
}

/**
 * @brief Configures GPIO pins for the I2C communication as per the desired mode.
 */
void config_output()
{
    gpio_set_direction(I2C_DATA_PIN, GPIO_MODE_INPUT_OUTPUT);
    gpio_set_direction(I2C_CLK_PIN, GPIO_MODE_OUTPUT);
    gpio_set_direction(GPIO_EXP_EN, GPIO_MODE_OUTPUT);
    gpio_set_direction(I2C_EN, GPIO_MODE_OUTPUT);
    gpio_set_direction(GPIO_NUM_1, GPIO_MODE_INPUT_OUTPUT);
}

```

```
    gpio_set_level(GPIO_EXP_EN, 1);
    gpio_set_level(I2C_EN, 1);
    gpio_set_pull_mode(I2C_DATA_PIN, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(I2C_CLK_PIN, GPIO_PULLUP_ONLY);
    vTaskDelay(pdMS_TO_TICKS(2000));

```

```
i2c_master_bus_config_t i2c_mst_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT, // Source CLK can be changed for different power management
        .i2c_port = I2C_NUM_0,             // Some ESP32s might have multiple I2C ports, not C3.
        .scl_io_num = GPIO_NUM_2,          // Changed default SCLK pin
        .sda_io_num = GPIO_NUM_8,          // Changed default SDA pin
        .glitch_ignore_cnt = 7,
        .intr_priority = 0, // Glitch filter count
        .flags.enable_internal_pullup = true,
    };
 ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
    
     for (uint8_t i = 0; i <= 90; i++)
    {
        attempt_probe(bus_handle, i);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }

```
```

void attempt_probe(i2c_master_bus_handle_t bus, uint16_t dev_addr)
{
    esp_err_t r = i2c_master_probe(bus, dev_addr, -1);
    if (r == ESP_OK)
    {
        ESP_LOGI("attempt_probe", "Successful probe at 0x%x", dev_addr);
    }
    else
    {
        ESP_LOGE("attempt_probe", "Failed probe at 0x%x: %s", dev_addr, esp_err_to_name(r));
    }
}

```

R0nyx, 2024-02-28:
@mythbuster5 i may have overlooked my pin configuration and set the clock pin to be output only:
[image]

seems like probe is working now :)

puppy112358, 2024-02-28:
Im using ESP32-S3 with the master idf

R0nyx, 2024-02-28:
@puppy112358 try sharing your code

puppy112358, 2024-02-28:
`    i2c_master_bus_config_t i2c_config_handle1 = {
    .i2c_port = I2C_NUM_0,
    .sda_io_num = GPIO_NUM_5,
    .scl_io_num = GPIO_NUM_4,
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .intr_priority = 0,
    .glitch_ignore_cnt = 7,
    .trans_queue_depth = 256,
    .flags.enable_internal_pullup = true,
    };`

 `   ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_config_handle1,
                                       &iic_bus_handle[0]));`
`    for (uint8_t i = 0x60; i < 0x90; i ++) {//循环检测设备地址
        esp_err_t ret1 = i2c_master_probe(iic_bus_handle[0], i, 100);
        if (ret1 == ESP_OK) {
            ESP_LOGI(TAG,"address: %x",i);
            dev[device_number].addr = i;
            dev[device_number].channel = 1;
            device_number++;
        } else{
            ESP_LOGE(TAG,"address %x failed",0x68);
        }
`

R0nyx, 2024-02-28:
> [i2c_master.txt](https://github.com/espressif/esp-idf/files/12522389/i2c_master.txt)
> Yeah.. usually patch file needs two blank line at the end of file, but some sometime it might lack one line...BTW  can `i2c_master_probe` work now? 

@puppy112358 have you also applied @mythbuster5  patch?

puppy112358, 2024-02-28:
No, I am trying to apply it but I have some problems understanding some of it .
[image]
what does it mean in the file ? I feel so sorry to ask such a stupid question.

puppy112358, 2024-02-28:
> > [i2c_master.txt](https://github.com/espressif/esp-idf/files/12522389/i2c_master.txt)
> > Yeah.. usually patch file needs two blank line at the end of file, but some sometime it might lack one line...BTW  can `i2c_master_probe` work now?
> 
> @puppy112358 have you also applied @mythbuster5 patch?

Yeah ,I have applied it just now but it worked the same .

R0nyx, 2024-02-28:
@puppy112358 have you checked if your pins are pulled high?

puppy112358, 2024-02-28:
> @puppy112358 have you checked if your pins are pulled high?

yeah,
[image]

puppy112358, 2024-02-28:
```c
    gpio_set_pull_mode(GPIO_NUM_4,GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_5,GPIO_PULLUP_ONLY);
```
I also have these codes above .

puppy112358, 2024-02-29:
@R0nyx  Actually, I can run the "i2c tools" in the examples normally . I have checked my codes and it ,cannot find where the problem is.

R0nyx, 2024-02-29:
i've attached how my current i2c_master.c looks like
[i2c_lib_updated.txt](https://github.com/espressif/esp-idf/files/14445475/i2c_lib_updated.txt)
You can try giving it a go, see if it solves your issue

puppy112358, 2024-02-29:
@R0nyx  No, it did not change anything.

puppy112358, 2024-02-29:
@R0nyx Other part of my project can run normally. I think my hardware setup is connected correctly maybe.

R0nyx, 2024-02-29:
@puppy112358 
i've tried your provided code with few modifications:
[image]

```
  i2c_master_bus_handle_t iic_bus_handle[1];

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_config_handle1, &iic_bus_handle[0]));
    for (uint8_t i = 0x10; i < 0x90; i++)
    {
        esp_err_t ret1 = i2c_master_probe(iic_bus_handle[0], i, 100);
        if (ret1 == ESP_OK)
        {
            ESP_LOGI(TAG, "address: %x", i);

        }
        else
        {
            ESP_LOGE(TAG, "address %x failed", 0x68);
        }
    }
```
I'm new to development myself so i cannot provide you with a good asnwer

puppy112358, 2024-02-29:
> @puppy112358 i've tried your provided code with few modifications: ![image](https://private-user-images.githubusercontent.com/36990626/308872940-c5bdcc9f-084e-470c-a177-c6b620e50607.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkyMDAzMDIsIm5iZiI6MTcwOTIwMDAwMiwicGF0aCI6Ii8zNjk5MDYyNi8zMDg4NzI5NDAtYzViZGNjOWYtMDg0ZS00NzBjLWExNzctYzZiNjIwZTUwNjA3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDA5NDY0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgwOTY0NDQwNWFmODAyYmNkZWIxNDE5NTQ5YjdmNzc2YTRkMzRmYTc4ZjRjYTAwMDAzZmE0NWY4N2ExMzU1ZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KBE9qHf1KXsX7ZvLPpUQgTF94St2NOyajIkN1ZZvhsQ)
> 
> ```
>   i2c_master_bus_handle_t iic_bus_handle[1];
> 
>     ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_config_handle1, &iic_bus_handle[0]));
>     for (uint8_t i = 0x10; i < 0x90; i++)
>     {
>         esp_err_t ret1 = i2c_master_probe(iic_bus_handle[0], i, 100);
>         if (ret1 == ESP_OK)
>         {
>             ESP_LOGI(TAG, "address: %x", i);
> 
>         }
>         else
>         {
>             ESP_LOGE(TAG, "address %x failed", i);
>         }
>     }
> ```
> 
> I'm new to development myself so i cannot provide you with a good answer I would make sure that your sda pin is connected to sda pin on the slave, and same for the scl

Actually , if the connection is wrong, it should return NOT FOUND instead every address is ESP_OK. @R0nyx

puppy112358, 2024-02-29:
> @puppy112358 I know, but if you are using the same library as i do and follow [espressif implementation guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html), then you shouldn't be having any issues. make sure your:
> 
> ```
>   gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT_OUTPUT); 
>   gpio_set_direction(GPIO_NUM_5, GPIO_MODE_INPUT_OUTPUT);
> ```

I have add these codes , while it did not change anything ...

R0nyx, 2024-02-29:
@puppy112358 then code wise it seems like you are doing everything correctly. I would make sure your lines are pulled high with multi meter or logic analyzer. Also check if Esp32 SDA pin is connected to your slave SDA pin and the same for SCL. 

Oh and try:
```
    gpio_reset_pin(GPIO_NUM_4);
    gpio_reset_pin(GPIO_NUM_5);

```

puppy112358, 2024-02-29:
> > @puppy112358 then code wise it seems like you are doing everything correctly. I would make sure your lines are pulled high with multi meter or logic analyzer. Also check if Esp32 SDA pin is connected to your slave SDA pin and the same for SCL.
> > Oh and try this before setting pin configurations:
> > ```
> >     gpio_reset_pin(GPIO_NUM_4);
> >     gpio_reset_pin(GPIO_NUM_5);
> > ```
> 
> Oh ,the pins are not pulled high as I use the multi meter! but I have config the pin as it should be .

Should I add other codes to pull up the pins?

R0nyx, 2024-02-29:
@puppy112358 

> @puppy112358 Try using the gpio_reset_pin() command before setting pins to high:
> 
> ```
>     gpio_reset_pin(GPIO_NUM_4);
>     gpio_reset_pin(GPIO_NUM_5);
>     gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT_OUTPUT); 
>     gpio_set_direction(GPIO_NUM_5, GPIO_MODE_INPUT_OUTPUT);
>     gpio_set_pull_mode(GPIO_NUM_4,GPIO_PULLUP_ONLY);
>     gpio_set_pull_mode(GPIO_NUM_5,GPIO_PULLUP_ONLY);
> ```

puppy112358, 2024-02-29:
[image]
My earlier codes has included these ....

R0nyx, 2024-02-29:
@puppy112358 does it work then?

puppy112358, 2024-02-29:
No ,,,,how wired!!!
And as I said ,I can use the i2ctools in the i2c examples with the same ESP32-S3 and the same device .

R0nyx, 2024-02-29:
it might be a dumb idea but try and see if the pin is still not high:
```
    gpio_reset_pin(GPIO_NUM_4);
    gpio_reset_pin(GPIO_NUM_5);
    gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT_OUTPUT); 
    gpio_set_direction(GPIO_NUM_5, GPIO_MODE_INPUT_OUTPUT);
    gpio_set_pull_mode(GPIO_NUM_4,GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_5,GPIO_PULLUP_ONLY);
gpio_set_level(GPIO_NUM4, 1);
gpio_set_level(GPIO_NUM5, 1);
```

R0nyx, 2024-02-29:
@puppy112358 Try using the gpio_reset_pin() command  before setting pins to high:

```
    gpio_reset_pin(GPIO_NUM_4);
    gpio_reset_pin(GPIO_NUM_5);
    gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT_OUTPUT); 
    gpio_set_direction(GPIO_NUM_5, GPIO_MODE_INPUT_OUTPUT);
    gpio_set_pull_mode(GPIO_NUM_4,GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_NUM_5,GPIO_PULLUP_ONLY);
```

R0nyx, 2024-02-29:
But are the SDA and SCL pins High when you meassure them now?

ammaree, 2024-02-29:
I am not an I2C expert but have the new I2C master working on at least 7 different hardware platforms and with a total of 18 different I2C devices attached to the various platforms.

In all our configurations the initialisation is as follows:
```
const halI2C_Init_t halI2C_Init = {
	.i2c_port = I2C_NUM_0,
	.sda_io_num = GPIO_NUM_21,
	.scl_io_num = GPIO_NUM_22,
	.clk_source = I2C_CLK_SRC_DEFAULT,
	.glitch_ignore_cnt = 7,
	.intr_priority = 0,								// driver select default 1/2/3 etc
	.trans_queue_depth = 10,						// required to enable callbacks
	.flags.enable_internal_pullup = 1,                                // GPIO pull-ups enable
};
	ESP_ERROR_CHECK(i2c_new_master_bus(&halI2C_Init, &hI2Cbus));
```

We are not doing any other GPIO pin initialisation related to the I2C SD/SCL pins, everything is done within the i2c_new_master_bus() API.

To add a specific device, the following is all that is required.
```
	i2c_device_config_t dev_cfg = {
		.dev_addr_length = I2C_ADDR_BIT_LEN_7,
		.device_address = 0x38,                                    // device address
		.scl_speed_hz = 100000,                                   // bus speed for device
	};
	ESP_ERROR_CHECK(i2c_master_bus_add_device(hI2Cbus, &dev_cfg, &dev_hndl));
```

[5 more replies]