espressif/esp-idf #12160: examples/wifi/softap_sta not working as intended.

Saum025, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After downloading and configuring this example, I build the code and uploaded it to my ESP32 dev board (ESP32 Wroom 23). ESP 32 dev board was able to connect to my Wifi Router and it was also able to create an AP as intended. But Wifi NAT router functionality was not working. I was not able to access internet through ESP32's AP. Example code can be viewed [here](https://github.com/espressif/esp-idf/blob/master/examples/wifi/softap_sta/main/softap_sta.c). 

### Debug Terminal

```
I (30) boot: ESP-IDF v5.2-dev-2383-g82cceabc6e 2nd stage bootloader
I (30) boot: compile time Aug 30 2023 11:05:44
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v1.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=205f8h (132600) map
I (153) esp_image: segment 1: paddr=00030620 vaddr=3ffb0000 size=0371ch ( 14108) load
I (158) esp_image: segment 2: paddr=00033d44 vaddr=40080000 size=0c2d4h ( 49876) load
I (179) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8109ch (528540) map
I (370) esp_image: segment 4: paddr=000c10c4 vaddr=4008c2d4 size=09838h ( 38968) load
I (398) boot: Loaded app from partition at offset 0x10000
I (398) boot: Disabling RNG early entropy source...
I (409) cpu_start: Multicore app
I (410) cpu_start: Pro cpu up.
I (410) cpu_start: Starting app cpu, entry point is 0x400811e0
0x400811e0: call_start_cpu1 at C:/Users/saumy/esp/esp-idf/components/esp_system/port/cpu_start.c:170

I (0) cpu_start: App cpu up.
I (428) cpu_start: Pro cpu start user code
I (428) cpu_start: cpu freq: 160000000 Hz
I (428) cpu_start: Application information:
I (432) cpu_start: Project name:     softap_sta
I (438) cpu_start: App version:      1
I (442) cpu_start: Compile time:     Aug 30 2023 11:14:26
I (448) cpu_start: ELF file SHA256:  0f1b260f8...
I (454) cpu_start: ESP-IDF:          v5.2-dev-2383-g82cceabc6e
I (460) cpu_start: Min chip rev:     v0.0
I (465) cpu_start: Max chip rev:     v3.99
I (470) cpu_start: Chip rev:         v1.0
I (475) heap_init: Initializing. RAM available for dynamic allocation:
I (482) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (488) heap_init: At 3FFB77D0 len 00028830 (162 KiB): DRAM
I (494) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (500) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (507) heap_init: At 40095B0C len 0000A4F4 (41 KiB): IRAM
I (514) spi_flash: detected chip: generic
I (517) spi_flash: flash io: dio
W (521) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (535) app_start: Starting scheduler on CPU0
I (540) app_start: Starting scheduler on CPU1
I (540) main_task: Started on CPU0
I (550) main_task: Calling app_main()
I (610) wifi:wifi driver task: 3ffbf610, prio:23, stack:6656, core=0
I (630) wifi:wifi firmware version: a581065
I (630) wifi:wifi certification version: v7.0
I (630) wifi:config NVS flash: enabled
I (630) wifi:config nano formating: disabled
I (640) wifi:Init data frame dynamic rx buffer num: 32
I (640) wifi:Init management frame dynamic rx buffer num: 32
I (650) wifi:Init management short buffer num: 32
I (650) wifi:Init dynamic tx buffer num: 32
I (660) wifi:Init static rx buffer size: 1600
I (660) wifi:Init static rx buffer num: 10
I (660) wifi:Init dynamic rx buffer num: 32
I (670) wifi_init: rx ba win: 6
I (670) wifi_init: tcpip mbox: 32
I (670) wifi_init: udp mbox: 6
I (680) wifi_init: tcp mbox: 6
I (680) wifi_init: tcp tx win: 5744
I (690) wifi_init: tcp rx win: 5744
I (690) wifi_init: tcp mss: 1440
I (690) wifi_init: WiFi IRAM OP enabled
I (700) wifi_init: WiFi RX IRAM OP enabled
I (700) WiFi SoftAP: ESP_WIFI_MODE_AP
I (720) WiFi SoftAP: wifi_init_softap finished. SSID:BhaiHo password:123456789 channel:1
I (720) WiFi Sta: ESP_WIFI_MODE_STA
I (720) WiFi Sta: wifi_init_sta finished.
I (730) phy_init: phy_version 4771,450c73b,Aug 16 2023,11:03:10
I (810) wifi:mode : sta (3c:61:05:1f:53:4c) + softAP (3c:61:05:1f:53:4d)
I (810) wifi:enable tsf
I (810) wifi:Total power save buffer number: 16
I (810) wifi:Init max length of beacon: 752/752
I (820) wifi:Init max length of beacon: 752/752
I (820) WiFi Sta: Station started
I (830) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (3660) wifi:ap channel adjust o:1,1 n:5,2
I (3660) wifi:new:<5,2>, old:<1,1>, ap:<5,2>, sta:<5,2>, prof:1
I (3660) wifi:state: init -> auth (b0)
I (3660) wifi:state: auth -> assoc (0)
I (3670) wifi:state: assoc -> run (10)
I (3680) wifi:connected with Cephlon 2G, aid = 2, channel 5, 40D, bssid = bc:62:d2:c4:c2:02
I (3680) wifi:security: WPA2-PSK, phy: bgn, rssi: -51
I (3690) wifi:pm start, type: 1

I (3700) wifi:<ba-add>idx:0 (ifx:0, bc:62:d2:c4:c2:02), tid:0, ssn:0, winSize:64
I (3760) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (9190) WiFi Sta: Got IP:192.168.1.35
I (9190) esp_netif_handlers: sta ip: 192.168.1.35, mask: 255.255.255.0, gw: 192.168.1.1
I (9190) WiFi Sta: connected to ap SSID:Cephlon 2G password:Sayala10$@
W (9200) esp_netif_lwip: napt disabled on esp_netif:0x3ffc4754
I (9200) main_task: Returned from app_main()
I (98490) wifi:<ba-add>idx:1 (ifx:0, bc:62:d2:c4:c2:02), tid:6, ssn:0, winSize:64
```

The third line from bottom, says that NAPT is disabled on esp_netif but it return ESP_OK.
> W (9200) esp_netif_lwip: napt disabled on esp_netif:0x3ffc4754

### Screenshots
[image]

Saum025, 2023-09-01:
Can anyone please help me solve this issue. 
@methodmissing 
@svenfuchs 
@paulreimer

mgssnr, 2023-09-08:
Looking at esp_netif_napt_enable(), there's disagreement between the argument being passed to ip_napt_enable_netif() and the log message.  It's not clear to me if the argument to ip_napt_enable_netif() is wrong (passing netif->lwip_netif) or if the argument to ESP_LOGW() is wrong (passing esp_netif).

The comment "Disable napt on all other interface (sic)" and "if (netif != esp_netif)" which seems to indicate the intention is to disable napt on all other interfaces except esp_netif.  If that's the case, then the argument to ESP_LOGW() is incorrect, and the argument to ESP_LOGW() should be netif->lwip_netif instead of esp_netif, because clearly esp_netif is NOT the netif being modified.

mgssnr, 2023-09-08:
Incidentally, I have the softap_sta code running, and both a linux machine and an iPad successfully reach the internet through the ESP32.  At first, DNS wasn't working because it was pointing to the ESP32 which has no DNS running.  Once I set both host's DNS to 8.8.8.8, I was able to browse the internet, ping hosts by name, etc, via the softAP.

Truly amazing for a $5 board and a bunch of free software.

letorresdev, 2024-01-21:
@Saum025  did you find a solution for your issue, I'm facing the same problem.

Saum025, 2024-01-24:
@letorresdev I have not found any solutions. Let me know if anyone finds any solution. @mgssnr can you post your code.

mgssnr, 2024-01-24:
I would love to post the code, but I don't think I have it any longer.  I'm looking around.

mgssnr, 2024-01-24:
I found the code, and git diff shows that the only changes I made were in the log messages in esp_netif_napt_enable (ESP_LOGW("napt disable on esp_netif:%p", netif) instead of ESP_LOGW("napt disable on esp_netif:%p", esp_netif)), and in components/lwip/port/lwopopts.h where I set IP_FORWARD = 1 and IP_NAPT = 1.

hansw123, 2024-04-08:
hi @Saum025 @mgssnr @letorresdev 
sorry for later reply! 
the default softap dns addr is 192.168.4.1,but softap don't have dns server
so you can add the func

void softap_set_dns_addr(esp_netif_t *esp_netif_ap){

    esp_netif_dns_info_t dns;
    dns.ip.u_addr.ip4.addr = ipaddr_addr("8.8.8.8");
    dns.ip.type = IPADDR_TYPE_V4;
    uint8_t dhcps_dns_value = 2;
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_dhcps_stop(esp_netif_ap));
    ESP_ERROR_CHECK(esp_netif_dhcps_option(esp_netif_ap, ESP_NETIF_OP_SET, ESP_NETIF_DOMAIN_NAME_SERVER, &dhcps_dns_value, sizeof(dhcps_dns_value)));
    ESP_ERROR_CHECK(esp_netif_set_dns_info(esp_netif_ap, ESP_NETIF_DNS_MAIN, &dns));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_dhcps_start(esp_netif_ap));

}

in the code  call the func after after esp_netif_t *esp_netif_ap = wifi_init_softap();

Indastri, 2024-11-12:
Since there was no further response from the user I am closing this issue. If you will encounter any further problems please reopen this or create new issue related to the problem.

Cheers!

closed 2024-11-12 by Indastri