espressif/esp-idf #12161: PHY ESP32-C3 calibration problem on first startup, when powered by AC-DC

MacWyznawca, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After uploading to the manufacturer's brand-new module with **ESP32-C3** in the **230 V**-powered bulb, Wi-Fi signal strength (**ESP-NOW**) is minimal. The range is about 1 cm (10 mm).  The receiver works properly, only the signal transmitted by the module is (at the first launch) minimal.

If the first launch (pairing process) is successful, the Wi-Fi (**ESP-NOW**) range is completely normal (tens of meters).

It helps a little to lower the Wi-Fi power to 8 (2 dBm) at the first startup. After that, it usually works normally.

I have checked the various PHY calibration settings and the various methods of storing calibration data available in menuconfig. 

The problem does not occur when powered from the same module with 3.3 V.

I point out that after the first startup (if the module exchanges a few data packets with the base using **ESP-NOW**), it continues to make the whole thing work properly, despite powering off and on, so I know that the antenna circuit is done correctly.

Translated with www.DeepL.com/Translator (free version)

MaxwellAlan, 2023-09-07:
Hi @MacWyznawca 

Did you try disable store phy calibration data in NVS ?

CONFIG_ESP_PHY_CALIBRATIOIN_AND_DATA_STORAGE=n

From you describtion, seems like phy calibration issue when 220V power supply.

Could you send a test device for us to reproduce this and debug ?

Or some other ways such as you could provide a env for we to debug?

Thanks.

MacWyznawca, 2023-09-07:
Thanks for the feedback @MaxwellAlan 
Unfortunately, I checked disabling calibration data storing in flash. It had no effect. Zeroing the bulb (with the on/off sequence) clears the calibration data for test.

Where would we send a sample of the bulb? We are based in Poland (we border the Czech Republic), but the bulb manufacturer is based in China.
We can ship from Poland, but I can also ask the manufacturer to ship.

What data can I send you from our environment?
I'm using ESP-IDE 5.1.1 (the latest stable version), but the bug drags on until version 5.0 and on beta 5.2 everything looks the same.

MacWyznawca, 2023-09-07:
@MaxwellAlan  

I'll still remind you that we use ESP-NOW exclusively.

This is what the code snippet responsible for initiating the radio looks like.

During pairing, the radio channel is changed, and the power is lowered to 8 (2 dBm) for the filament. Of course, with other powers, I also did test - lowering gives the best results. After pairing, the power is restored to a higher level and the standard Wi-Fi channel is set.

Pairing we tested on channel 6 and now on channel 1. Normal operation on channel 11.

```c

esp_err_t LuAirChangeChannel(uint8_t ch){
    uint8_t primary, sec;
    esp_err_t ret = esp_wifi_get_channel(&primary, &sec);
    ESP_LOGE(LUAIR_TAG, "LuAirChangeChannel cur: %d, new: %d. %s", primary, ch, esp_err_to_name(ret)); 
    if(ret != ESP_OK) return ret;
    if(primary == ch) return ESP_OK;
    ret = esp_wifi_set_channel(ch, WIFI_SECOND_CHAN_NONE);
    if(ret == ESP_OK) {
        _curentChannel = ch;
    }
    vTaskDelay(pdMS_TO_TICKS(200));
    esp_wifi_get_channel(&primary, &sec);
    ESP_LOGE(LUAIR_TAG, "LuAirChangeChannel cur: %d, set: %d. %s", primary, ch, esp_err_to_name(ret));
    return ret;
}

void LuAirChangeMaxPower(int8_t power){
    _curentPower = power > 15 ? power : MAX_TX_POWER;
    ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(_curentPower));
    int8_t powerGet;
    esp_wifi_get_max_tx_power(&powerGet);
    ESP_LOGE(LUAIR_TAG, "Driver LuAirChangeMaxPower, channel: %d, POWER: %d",_curentChannel, powerGet);
}

//#define ESP_WIFI_CONNECTIONLESS_INTERVAL_DEFAULT_MODE (0)

void LuAirWindowSet(uint16_t window, uint16_t interval){
    interval = ((interval < 100) ? 100 : interval);
    interval = ((interval > 500) ? 500 : interval);

    window = ((window < 56) ? 56 : window);

#else
    ESP_ERROR_CHECK(esp_wifi_connectionless_module_set_wake_interval(ESP_WIFI_CONNECTIONLESS_INTERVAL_DEFAULT_MODE)); // ESP_WIFI_CONNECTIONLESS_INTERVAL_DEFAULT_MODE
    ESP_ERROR_CHECK(esp_now_set_wake_window(65535));
    ESP_LOGE(LUAIR_TAG, "CONFIG_ESP_WIFI_STA_DISCONNECTED_PM disable");
    return;
#endif
    ESP_ERROR_CHECK(esp_wifi_connectionless_module_set_wake_interval(interval)); // ESP_WIFI_CONNECTIONLESS_INTERVAL_DEFAULT_MODE
    if(window > interval){
        window = interval;
    }
    ESP_ERROR_CHECK(esp_now_set_wake_window(window));
    ESP_LOGE(LUAIR_TAG, "CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE window: %d, interval: %d", window, interval);
}

static void LuAirNow_init(void){
    ESP_ERROR_CHECK(esp_now_init());
    ESP_ERROR_CHECK(esp_wifi_config_espnow_rate(ESP_IF_WIFI_STA, WIFI_PHY_RATE_LORA_500K));

    esp_now_peer_info_t *peer = malloc(sizeof(esp_now_peer_info_t));
    if (peer == NULL) {
        ESP_LOGE(LUAIR_TAG, "Malloc peer information fail");
        LuAirDeinit();
        return;
    }
    memset(peer, 0, sizeof(esp_now_peer_info_t));
    peer->channel = 0;
    peer->ifidx = ESP_IF_WIFI_STA;
    peer->encrypt = false;
    memcpy(peer->peer_addr, broadcastMac, ESP_NOW_ETH_ALEN);
    ESP_ERROR_CHECK(esp_now_add_peer(peer));
    free(peer);

    ESP_ERROR_CHECK(esp_now_register_send_cb(LuAirSend_cb));
    ESP_ERROR_CHECK(esp_now_register_recv_cb(LuAirRecieve_cb));
}

static void LuAirRadio_init(void){
  //  ESP_ERROR_CHECK(esp_netif_init());// moved to main AppStart
  //  ESP_ERROR_CHECK(esp_event_loop_create_default()); //moved to main AppStart
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
 //   wifi_country_t country_01 = {.cc="01", .schan=1, .nchan=11, .policy=WIFI_COUNTRY_POLICY_MANUAL};
    wifi_country_t country = {
        .cc = "CN",
        .schan = 1,
        .nchan = 13, 
  //      .max_tx_power = _curentPower / 4, 
        .policy = WIFI_COUNTRY_POLICY_MANUAL,
    };
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());

    esp_wifi_set_ps(WIFI_PS_NONE); // WIFI_PS_MAX_MODEM

    ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR | WIFI_PROTOCOL_11B)); // WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_LR
    ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(_curentPower));
    ESP_ERROR_CHECK(esp_wifi_set_channel(_curentChannel, WIFI_SECOND_CHAN_NONE)); // channel 11
    LuAirNow_init();
    int8_t powerGet;
    esp_err_t ret = esp_wifi_get_max_tx_power(&powerGet);
    ESP_LOGE(LUAIR_TAG, "LuAirRadio_init, channel: %d, POWER: %d/%d, RET: %ld",_curentChannel, _curentPower, powerGet, ret);
}
```

Xiehanxin, 2023-09-07:
hi @MacWyznawca, I noticed that you can provide sample device . Could you please send an email to _**sales@espressif.com**_ and specify that it's related to an github issue _https://github.com/espressif/esp-idf/issues/12161_ , and Our RD is interested in receiving samples from you, and we will provide you with a shipping contact once we receive your email.

MacWyznawca, 2023-09-07:
Thank You @Xiehanxin I sent an email from my company account (@luon.me)

MacWyznawca, 2023-10-13:
Hi @MaxwellAlan, @zhangyanjiaoesp, @Xiehanxin    Is there any new news? Because we stand with mass production.

MaxwellAlan, 2023-10-13:
Hi @MacWyznawca 

Shanghai office had just received the bulb.
Since it's a  complete production, could you give the detail steps for we to reproduce your issue? 
PS: we may need to  break it down to conduct more tests about power supply, which may also involve re-flashing new debug firmware. 
So could you give some more detail about this?
Thanks.

MaxwellAlan, 2023-10-16:
@MacWyznawca 
Thanks for the info, we will take high priority to support this issue.
Could you arrange some engineers to connect directly with us?
The communication b/w us in the github is not so efficiently. 
Thanks.

MaxwellAlan, 2023-10-23:
Hi @MacWyznawca 

The root cause of this issue is that there is one inductor missing on the circuit board, after adding the inductor, the bulb works well.

Thanks.

closed 2023-10-23 by MaxwellAlan

MacWyznawca, 2023-10-23:
And I spent many week looking for a problem in my code 😉However, sometimes it is worth trusting yourself and looking for a mistake somewhere else. Thank you very much for your help! We wouldn't handle it ourselves.