espressif/esp-idf #12162: idf_monitor.py process locks up when umlauts are entered

crackwitz, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We just noticed, within VSCode as well as separate powershell, that pressing "weird" keys on the keyboard may lock up the `esp-idf\tools\idf_monitor.py` process. Even exiting the terminal does not actually end the process. The COM port remains occupied. Killing the Python process via Task Manager releases the COM port.

We can reliably reproduce this on Win10 and Win11, with esp-idf v5.1. We tested keys Ä/Ö/Ü or ä/ö/ü. You can reproduce this by setting your keyboard to a German layout. I didn't test whether copy-pasting those symbols works the same.

peterdragun, 2023-09-01:
Hi, can you tell me more about your environment? Are you using some console interactive app or just a regular one? Are you using native USB of the chip or the uart bridge when communicating with the chip? What is the version of esp-idf-monitor that you are using? - The version can be displayed by running `pip show esp-idf-monitor`

crackwitz, 2023-09-01:
Nothing special. Reproducible with arbitrary windows powershell but also inside of VSCode with the espressif extension.

We have ESP32 and FT2232. We did not check if serial connection with Putty causes issues.

None of us noticed this issue except for the IDF Monitor.

I'm not in front of the work computer right now. We are using ESP-IDF v5.1.

closed 2023-09-15 by espressif-bot

crackwitz, 2023-09-15:
Great, thanks! Please excuse my not submitting that to the correct repository.