espressif/esp-idf #12163: USB MSC example crashes when specific drive is connected

Maldus512, 2023-08-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-beta1-978-gd3c99ed3b8

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

I am running the USB MSC example from https://github.com/espressif/esp-idf/tree/release/v5.1/examples/peripherals/usb/host/msc 
When connecting a USB drive it should display its information and run a few operations on the file system.

### What is the actual behavior?

With a specific (cheap drive gifted as advertisement gadget) drive model the `usb_msc` component crashes with what appears to be a NULL pointer exception.

### Steps to reproduce.

Run the msc_usb example and connect a usb drive with the aforementioned characterstics.

### Debug Logs.

```plain
I (395) example: Waiting for USB stick to be connected
I (845) example: MSC device connected
I (845) example: connection...
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x90c
idProduct 0x1000
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200a409  PS      : 0x00060b30  A0      : 0x8200973b  A1      : 0x3fc98d00  
0x4200a409: msc_host_get_device_info at /home/maldus/Source/esp-idf-v5.1/examples/peripherals/usb/host/msc/managed_components/espressif__usb_host_msc/src/msc_host.c:549 (discriminator 4)

A2      : 0x3fc98db8  A3      : 0x3fc98d68  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000002  A7      : 0x00000000  A8      : 0x8200a3fc  A9      : 0x3fc98ce0  
A10     : 0x3fc98db4  A11     : 0x00000000  A12     : 0xffffffff  A13     : 0xfffffff7  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  
0x400556d5: strlen in ROM

0x400556e5: strlen in ROM

Backtrace: 0x4200a406:0x3fc98d00 0x42009738:0x3fc98d50 0x42031063:0x3fc98ec0 0x4037c37d:0x3fc98ef0
0x4200a406: msc_host_get_device_info at /home/maldus/Source/esp-idf-v5.1/examples/peripherals/usb/host/msc/managed_components/espressif__usb_host_msc/src/msc_host.c:547 (discriminator 4)

0x42009738: app_main at /home/maldus/Source/esp-idf-v5.1/examples/peripherals/usb/host/msc/main/msc_example_main.c:206 (discriminator 2)

0x42031063: main_task at /home/maldus/Source/esp-idf-v5.1/components/freertos/app_startup.c:208 (discriminator 13)

0x4037c37d: vPortTaskWrapper at /home/maldus/Source/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 5ad8d65b7b44cff5

Rebooting...
```

tore-espressif, 2023-09-04:
Hello @Maldus512 , 
Thank you for reporting this! it seems that the cheap USB drive you are using does not contain all the descriptors that the MSC driver expects.
We will fix in next version of MSC driver

Maldus512, 2023-09-04:
Hello @tore-espressif,
Thank you for the clarification; will that be the version 1.1.0?

tore-espressif, 2023-09-04:
We aim for 1.1.1

(Version 1.1.0 was yanked, because it did not work with some edgecases when PSRAM was enabled https://components.espressif.com/components/espressif/usb_host_msc/versions/1.1.0 )

closed 2023-10-02 by tore-espressif