espressif/esp-idf #12167: [Help]how to download/upload file from SPIFFS

fatsheep2020, 2023-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

when i want upload many files to SPIFFS and download it to check, how to do that?
is any tool can do that?
Thanks!

adokitkat, 2023-09-04:
Hello. You can check out our [spiffsgen](https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffsgen) example, which demonstrates how you can generate a SPIFFS partition on ESP32 from a folder on your computer. You will use `spiffs_create_partition_image(<partition> <base_dir> [FLASH_IN_PROJECT] [DEPENDS dep dep dep...])` in CMakeLists.txt file. More documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spiffs.html#spiffsgen-py). 

To download them wirelessly you would need ESP32 to run e.g. a [HTTP server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html) ([HTTP file serving example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving)). If you need it to be secure look at [HTTPS documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html) page and adjust your code accordingly.

adokitkat, 2023-09-11:
I am closing due to no answer. Feel free to reopen this if you want to continue discussing this issue.

closed 2023-09-11 by adokitkat