espressif/esp-idf #12170: ECDSA signing issue

Jimis333, 2023-08-31:
Hi
I try to sign my build with a signing key file but it fails. For my board version [WROOM ](https://www.sparkfun.com/products/17381) there is only the EDCSA option so I need to create a file with the following command
openssl ecparam -name secp256k1 -genkey -noout -out secure_boot_signing_key_jim.pem

This returns an error
FAILED: signature_verification_key.bin....
A fatal error occurred: Supports NIST192p and NIST256p keys only

If I use the other command with RSA key it fails also 
openssl ecparam -name prime256v1 -genkey -noout -out secure_boot_signing_key_jim.pem
[image]

Thanks in advance

AdityaHPatwardhan, 2023-09-01:
@Jimis333 Can you also share the output of `esptool.py chip_id` command.
I would like to know the ECO version for your chip.
If it is ECO3 then you may use Secure Boot Version 2 instead of Secure Boot version 1.
You would be able to execute the above command from an IDF environment since esptool script should be available there.

Jimis333, 2023-09-01:
After a lot of searching I did the following steps to make it work

Step 1: check chip version
[image]

Step 2: set the minimum chip revision in menuconfig (this limits your options in security features on next step)
[image]

Step 3: set the security features (RSA)
[image]

Step 4: generate the key (RSA)
openssl ecparam -name prime256v1 -genkey -noout -out secure_boot_signing_key_jim.pem

Thank you for your replies

closed 2023-09-02 by Jimis333

AdityaHPatwardhan, 2023-09-04:
@Jimis333 
Its good that the issue got resolved.
I was wondering about the chip revision that you are using. Looking at the [link](https://www.sparkfun.com/products/17381) I was under the impression that you have the old modules with `ECO < 3`.
I see that you have used RSA signing for secure boot V2, yes that is the expected way.
For secure Boot V2 ECDSA signing is not an option for esp32.

Also when you mention this 

> Step 4: generate the key (RSA)
openssl ecparam -name prime256v1 -genkey -noout -out secure_boot_signing_key_jim.pem

Just wanted to note that the command generates an ECDSA key and not an RSA key.
Thanks for closing the issue

Jimis333, 2023-09-05:
@AdityaHPatwardhan 

You just saved me a lot of trouble with your mention ! I'm not very familiar with certifications and the command I used was indeed for ECDSA. This was the answer to my other opened issue ! https://github.com/espressif/esp-idf/issues/12148
A proper command would be e.g
openssl genrsa -out private-key.pem 3072

Thanks a lot !

AdityaHPatwardhan, 2023-09-05:
You can also use the standard commands provided by `espsecure.py`

``` 
espsecure.py generate_signing_key --version 2 --scheme rsa3072 secure_boot_signing_key.pem
```
It takes care of all the versions and respective algorithm requirements so we dont even have to worry about using a wrong algorithm which is not supported.

Jimis333, 2023-09-05:
I don't generally use py commands I try to let VS code do those automatically. For that particular command I get error

espsecure.py generate_signing_key --version 2 --scheme rsa3072 secure_boot_signing_key.pe 
  File "<stdin>", line 1
    espsecure.py generate_signing_key --version 2 --scheme rsa3072 secure_boot_signing_key.pe
                 ^^^^^^^^^^^^^^^^^^^^
SyntaxError: invalid syntax

AdityaHPatwardhan, 2023-09-05:
The command works correctly on my side.
Do you have espsecure.py script available in your environment?

Can you try `espsecure.py --help`?

Jimis333, 2023-09-05:
espsecure.py --help
espsecure.py : The term 'espsecure.py' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify 
that the path is correct and try again.
At line:1 char:1
+ espsecure.py --help
+ ~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (espsecure.py:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Jimis333, 2023-09-05:
Yes you are right, but I think I have it uninstalled at some point. 
If I install the extension in VSCode I don't know what is the reason of installing it in windows also. It's a bit confusing. I guess it should work on VScode terminal environment but maybe not everything is supported there.

Jimis333, 2023-09-05:
It still didn't work. I removed the .py from the command and it worked
espsecure generate_signing_key --version 2 --scheme rsa3072 secure_boot_signing_key.pem
instead of 
espsecure.py generate_signing_key --version 2 --scheme rsa3072 secure_boot_signing_key.pem