espressif/esp-idf #12171: How use poxis func like mutex and cond?

Shaco-Ma, 2023-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

use idf v5.1-dirty
chip:esp32c6
  when I use pthread.h func,like pthread_mutex_lock and pthread_cond_broadcast and so on, also include #include <pthread.h>,use idf4.4.4 is complie ok,but when use idf5.1, it's had wrong in ld fail and multiple definition:
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_unlock':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:244: warning: pthread_cond_broadcast is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_destroy':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:130: warning: pthread_cond_destroy is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_cond_var.c.obj): in function `s_check_and_init_if_static':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_cond_var.c:45: warning: pthread_cond_init is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_cond_var.c.obj): in function `pthread_cond_wait':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_cond_var.c:94: warning: pthread_cond_timedwait is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_rdlock':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:181: warning: pthread_cond_wait is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(libbuf.c.obj): in function `libBuf_free':
/home/shaco/Work/Code/hs28p_rk3328_v02/main/libbuf/libbuf.c:87: warning: pthread_condattr_destroy is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(libbuf.c.obj): in function `libBuf_init':
/home/shaco/Work/Code/hs28p_rk3328_v02/main/libbuf/libbuf.c:57: warning: pthread_condattr_init is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread.c.obj): in function `esp_pthread_set_cfg':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:135: warning: pthread_getspecific is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread.c.obj): in function `esp_pthread_init':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:72: warning: pthread_key_create is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:77: warning: pthread_key_delete is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_init':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:75: warning: pthread_mutex_destroy is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread.c.obj): in function `pthread_mutex_init_if_static':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:628: warning: pthread_mutex_init is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_destroy':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:122: warning: pthread_mutex_lock is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/MyData/esp/esp-idf-5.1/components/pthread/pthread_rwlock.c:131: warning: pthread_mutex_unlock is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libstdc++.a(locale.o): in function `.L0 ':
/builds/idf/crosstool-NG/.build/riscv32-esp-elf/build/build-cc-gcc-final/riscv32-esp-elf/rv32imac_zicsr_zifencei/ilp32/no-rtti/libstdc++-v3/include/riscv32-esp-elf/bits/gthr-default.h:700: warning: pthread_once is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libc.a(lib_a-fclose.o): in function `_fclose_r':
/builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/fclose.c:73: warning: pthread_setcancelstate is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread.c.obj): in function `esp_pthread_set_cfg':
/home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:143: warning: pthread_setspecific is not implemented and will always fail
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libnosys.a(pthread.o): in function `pthread_attr_destroy':
/builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/libgloss/libnosys/pthread.c:33: multiple definition of `pthread_attr_destroy'; esp-idf/pthread/libpthread.a(pthread.c.obj):/home/MyData/esp/esp-idf-5.1/components/pthread/pthread.c:767: first defined here
/home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/shaco/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libnosys.a(pthread.o): in function `pthread_attr_getdetachstate':
[... cut ...] -esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libnosys.a(pthread.o): in function `pthread_setspecific':
/builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/libgloss/libnosys/pthread.c:630: multiple definition of `pthread_setspecific'; esp-idf/pthread/libpthread.a(pthread_local_storage.c.obj):/home/MyData/esp/esp-idf-5.1/components/pthread/pthread_local_storage.c:196: first defined here
How To fix This?
who can help me?

ESP-Marius, 2023-09-01:
https://github.com/espressif/esp-idf/tree/master/examples/system/pthread seems to compile fine for me on C5 on v5.1, so I'm unable to reproduce your issue.

I suggest you try with that example first and see if that works. You can also try to clean your build, `idf.py fullclean`. Maybe you switched between 4.4 and 5.1 without properly setting everything up again and something got stuck?

Shaco-Ma, 2023-09-01:
> https://github.com/espressif/esp-idf/tree/master/examples/system/pthread seems to compile fine for me on C5 on v5.1, so I'm unable to reproduce your issue.
>
> I suggest you try with that example first [...]

I had idf.py fullclean and rm build dir, just build is same wrong, it's had some in .git or sdkconfig?

ESP-Marius, 2023-09-01:
Do you have the same issue if you build the example I linked to?

Shaco-Ma, 2023-09-01:
> 

No,I use the example and example/system/pthread is also ok,just my project had this issue
only the code,I had add components with websocket in esp-protocols
And I had rm sdkconfig, sdkconfig.old and build,but do idf.py menuconfig,but it had the same issue...

ESP-Marius, 2023-09-01:
if you give me a minimal example that reproduces it then I can take a look.

You can also try to add `REQUIRES pthread` to your cmakelist requires and see if that helps.

Shaco-Ma, 2023-09-01:
> if you give me a minimal example that reproduces it then I can take a look.
> 
> You can also try to add `REQUIRES pthread` to your cmakelist requires and see if that helps.

@ESP-Marius 
Sorry,I get the question why,because I use the function :
pthread_condattr_init(&attr);
pthread_condattr_setclock(&attr, CLOCK_MONOTONIC);
pthread_cond_init(&cond, &attr);
when I remove the cond attribute set function,It's OK
alos when pthread_condattr_init and pthread_condattr_setclock will be support?

closed 2023-09-01 by Shaco-Ma

ESP-Marius, 2023-09-01:
Supporting more posix functions are on our roadmap, but I cannot give you a timeline at the moment unfortunately.