espressif/esp-idf #12172: Unable to install in vsCode - Intel Mac

eedea, 2023-08-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v 5.1.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I want to install espIDF as vsCode extension.

### What is the actual behavior?

It fails in building lxml with error: #error architecture not supported

### Steps to reproduce.

1. Install Espressif IDF extension on intel mac (10.15.7)
2. Run ESP-IDF: configure ESP-IDF extension
3. Follow installation steps
4. It fails to install python packages

### Build or installation Logs.

```plain
running build_ext
      building 'lxml.etree' extension
      creating build/temp.macosx-10.14.6-x86_64-cpython-38
      creating build/temp.macosx-10.14.6-x86_64-cpython-38/src
      creating build/temp.macosx-10.14.6-x86_64-cpython-38/src/lxml
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -DCYTHON_CLINE_IN_TRACEBACK=0 -I/opt/local/include/libxml2 -I/opt/local/include -Isrc -Isrc/lxml/includes -I/Users/amirsaghaleini/.espressif/python_env/idf5.1_py3.8_env/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c src/lxml/etree.c -o build/temp.macosx-10.14.6-x86_64-cpython-38/src/lxml/etree.o -w -flat_namespace
      In file included from src/lxml/etree.c:96:
      In file included from /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8/Python.h:11:
      In file included from /Library/Developer/CommandLineTools/usr/lib/clang/12.0.0/include/limits.h:21:
      In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/limits.h:63:
      /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/cdefs.h:807:2: error: Unsupported architecture
      #error Unsupported architecture
```

### More Information.

it looks like it is related to "-arch arm64" key in `clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64...`

dobairoland, 2023-09-01:
> running build_ext

Where this is coming from? I don't think it is from ESP-IDF.

dobairoland, 2023-12-20:
I think we can close this for inactivity.

closed 2024-01-04 by dobairoland