espressif/esp-idf #12176: GPIO feature: Add gpio_get_direction() and gpio_get_pull_mode()

IhorNehrutsa, 2023-09-01:
It is useful to have functions that return the current setting of a pin:

**1)** getter 
```
esp_err_t gpio_get_direction(gpio_num_t gpio_num, gpio_mode_t *mode);
```
like the setter:
```
esp_err_t gpio_set_direction(gpio_num_t gpio_num, gpio_mode_t mode);
```
https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/components/driver/gpio/include/driver/gpio.h#L169C11-L169C29

**2)** getter 
```
esp_err_t gpio_get_pull_mode(gpio_num_t gpio_num, gpio_pull_mode_t *pull);
```
like the setter:
```
esp_err_t gpio_set_pull_mode(gpio_num_t gpio_num, gpio_pull_mode_t pull);
```
https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/components/driver/gpio/include/driver/gpio.h#L184C11-L184C29

IhorNehrutsa, 2023-09-05:
@songruo 
Hi.
Will you work on this issue?
Thanks.

songruo, 2023-09-05:
@IhorNehrutsa  If I introduce a dump API that could dump out all information on the requested IO, including input_en = 0/1, output_en = 0/1, pullup_en = 0/1, pulldown_en = 0/1, func_sel, signal_connected, etc. together, instead of separate APIs for every characteristic. Will it work for you?

IhorNehrutsa, 2023-09-05:
@songruo Yes, perfect.
Thanks.

I need this feature in v5.0.2 ESP-IDF and later.
Thanks.

KaeLL, 2023-09-05:
@songruo Maybe make `gpio_config` call this function in order to show gpio cfg? That would be less redundant.

IhorNehrutsa, 2023-10-12:
@songruo Is any progress ?

songruo, 2023-10-16:
@IhorNehrutsa It has been merged to master, please wait for its sync to Github. However, we are not considering backporting the new feature to previous IDF versions. You could help yourself to pick the commit!

closed 2023-10-25 by espressif-bot