espressif/esp-idf #12180: internal compiler error: Segmentation fault in esp_lcd_panel_rgb.c:747:1:

engrzamanimran, 2023-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

I expected to compile lvgl and lovyanGFX components to compile program to display text on lcd. (WT32-SC01 PLUS). 

### What is the actual behavior?

When I build the following error occur.

[image]

 internal compiler error: Segmentation fault

### Steps to reproduce.

1. Click on build button 
2. The error will occur during compilation
3. It stop there with exit code 1
...

### Debug Logs.

```plain
C:/Users/Zaman/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap':
C:/Users/Zaman/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c:747:1: internal compiler error: Segmentation fault
  747 | }
      | ^
```

### More Information.

It is not lvgl fault it is in esp_lcd component

suda-morris, 2023-09-04:
it seems like it's not a "bug" in the esp-idf, but the GCC meets some unexpected conditions and fails. Many reasons can lead to such an error. for example, reduce the parallel build pipe numbers, check if you are running out of disk space.

HamzaHajeir, 2023-11-03:
Any updates to this issue? I've the exact same issue:
```
during RTL pass: ira
C:/Users/Hamza/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap':
C:/Users/Hamza/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c:747:1: internal compiler error: Segmentation fault
  747 | }
      | ^
libbacktrace could not find executable to open
```

engrzamanimran, 2023-11-04:
> Any updates to this issue? I've the exact same issue:
>
> ```
> during RTL pass: ira
> C:/Users/Hamza/esp/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap': [...]

I have restarted my PC and build the program again it works fine for me at that time. Make sure delete the build folder before build.

ztefanjo, 2023-12-18:
We are also getting sporadic compiler crashes (segmentation fault) when compiling this file after updating from v4.4.6 to v5.1.2:

`FAILED: esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_rgb.c.obj ccache D:\DevOps\agt11\_work\mower\Ext\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.1-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -ID:/DevOps/agt11/_work/mower/SW/WifiBle/51.42_WiFi-BLE-Dongle-Loader/Out/config -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_lcd/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_lcd/interface -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/newlib/platform_include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/freertos/FreeRTOS-Kernel/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/freertos/esp_additions/include/freertos -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/freertos/esp_additions/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/freertos/esp_additions/arch/xtensa/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_hw_support/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_hw_support/include/soc -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_hw_support/include/soc/esp32s3 -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_hw_support/port/esp32s3/. -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_hw_support/port/esp32s3/private_include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/heap/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/log/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/soc/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/soc/esp32s3 -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/soc/esp32s3/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/hal/esp32s3/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/hal/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/hal/platform_port/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_rom/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_rom/include/esp32s3 -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_rom/esp32s3 -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_common/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_system/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_system/port/soc -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_system/port/include/private -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/xtensa/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/xtensa/esp32s3/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/include/apps -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/include/apps/sntp -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/lwip/src/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/port/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/port/freertos/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/port/esp32xx/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/lwip/port/esp32xx/include/arch -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/deprecated -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/analog_comparator/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/dac/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/gpio/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/gptimer/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/i2c/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/i2s/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/ledc/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/mcpwm/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/parlio/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/pcnt/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/rmt/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/sdio_slave/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/sdmmc/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/sigma_delta/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/spi/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/temperature_sensor/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/touch_sensor/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/twai/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/uart/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/usb_serial_jtag/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/driver/touch_sensor/esp32s3/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_pm/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_ringbuf/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_mm/include -ID:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_psram/include - [... cut ...] :/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c during RTL pass: ira D:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap': D:/DevOps/agt11/_work/mower/Ext/esp-idf/components/esp_lcd/src/esp_lcd_panel_rgb.c:747:1: internal compiler error: Segmentation fault 747 | } | ^ libbacktrace could not find executable to open Please submit a full bug report, with preprocessed source (by using -freport-bug).`

dedmu5, 2024-01-29:
Any update on this issue?

SleepInfinity, 2024-10-23:
Same issue here.

daleffe, 2024-12-30:
Hi,

I'm using 5.3.0 version and solved it by including ```set(EXCLUDE_COMPONENTS esp_lcd)``` at root *CMakeLists.txt* file.

It's workaround since i don't use LCD.

itavero, 2025-01-10:
> I'm using 5.3.0 version and solved it by including `set(EXCLUDE_COMPONENTS esp_lcd)` at root _CMakeLists.txt_ file.
> 
> It's workaround since i don't use LCD.

We are using v5.0.1 and we also see this segmentation fault pop-up when the esp_lcd sources are compiled. "Unfortunately" we are using an LCD, so simply disabling it won't be a solution for us.

I find it interesting that for us it also fails when compiling `rgb_panel_draw_bitmap`. It feels that the function it self is the cause of this error. Any plans to investigate this at Espressif, @suda-morris?

We are mainly developing on Windows 10/11 machines and use CMake +ninja to do the build.
Number of jobs will vary from machine to machine (due to specs).
Let me know if you need additional information for your investigation.

suda-morris, 2025-01-13:
Hi @itavero which gcc version are your using in your project? are you using WSL? Does a clean build help? (idf.py clean; idf.py set-target esp32s3; idf.py build;) Does the issue still appear even when you compile the esp-idf built-in LCD examples?

sreuter, 2025-03-04:
This is still happening on all of our machines with v5.4.0.

itavero, 2025-03-04:
> Hi @itavero which gcc version are your using in your project? are you using WSL? Does a clean build help? (idf.py clean; idf.py set-target esp32s3; idf.py build;) Does the issue still appear even when you compile the esp-idf built-in LCD examples? 

Sorry for the late reply. Somehow your reply slipped through. I'm currently out of office, so I'll provide the information that I have on hand.

We are using the Windows build that comes with EDP-IDF v5.0.1. I'd have to check what the exact version is, but we install using the Python script in the SDK.

We have integrated ESP-IDF in our own CMake project as we also do cross-compilation for other architectures from the same code base. Therefore we don't use `idf.py` for building. However we did try to do a clean build from CMake, with the same result.

I'm not sure if we tried to build the example at the time as well, but I'll try to reproduce this again when I'm back at work later end of this month.

mame82, 2025-03-11:
> This is still happening on all of our Windows machines with v5.4.0.

Same after updating from 5.3.1 to 5.4.0 (win 11)

```
during RTL pass: ira
C:/Users/xxxx/esp/v5.4/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap':
C:/Users/xxxx/esp/v5.4/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c:686:1: internal compiler error: Segmentation fault
  686 | }
      | ^
```

tiantian1645, 2025-03-14:
I use wintogo and sometimes it will generate CMakeFiles outside build dir, and sometimes it will generate CMakeCache.txt int project dir path.

hmolavi, 2025-03-27:
> Same after updating from 5.3.1 to 5.4.0 (win 11)
>
> ```
> during RTL pass: ira
> C:/Users/xxxx/esp/v5.4/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap': [...]

I am also on Windows 11 and am getting this exact error, would greatly appreciate a solution to this.

xmcx-wang, 2025-03-29:
This is still happening on my pc(5.4.0), but after cleaning and compiling again, no errors，I'm not sure if it will happen again

CircuitGuy, 2025-04-15:
This is really odd. v5.4.0.

`/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c:686:1: internal compiler error: Segmentation fault`
 
Windows 11; only ever happens on my laptop. Ryzen 5 3500U with 24 GB of RAM.
 
Retrying fixes it 90% of the time. Clearly looks like a GCC bug or anti-virus contribution to a hang.

dr-glenn, 2025-05-27:
Same problem. ESP-IDF v5.4.0, Windows 11 Pro., 32 GB RAM. The error is Seg fault at line 686.
Clearing build directory does not work. Retrying has never helped. And I am trying to compile a project that uses LCD, so I can't exclude it.

Lapshin, 2026-01-08:
From the discussion, it seems that this issue occurs only on Windows machines.

As a temporary workaround, I suggest using the Linux compiler within a Windows WSL environment.
To identify the root cause, I’ll need an **example application that I can run locally** to reproduce the problem

gitlrj, 2026-08-21:
D:/esp32/v6.0.2/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap':
D:/esp32/v6.0.2/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c:745:1: internal compiler error: Segmentation fault
  745 | }
      | ^

I'm also encountering this issue in version 6.0.2.
If I recall correctly, this problem has been around since V5.x.x (maybe a year or two ago) and has persisted ever since.
After a clean build, this error shows up most of the time during the first compilation. However, if I just run the build again right after the failure, it usually compiles successfully without any issues.

V6.0.2版本中也有遇到这个问题,
我印象中是去年或者前年的时候,
V5.x.x版本中就有这个问题,一直存在,
清理构建后, 重新编译大概率都会遇到这个报错,
报错后再次编译就能正常通过了,

Windows10 
64GB 
VSCode-ESP IDF

Lapshin, 2026-09-29:
Good news: it's fixed in https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=cd15f0e559442c9b387d3cb7c6ce07d7109b825e

New toolchain releases will contain the fix

Bryght-Labs-Richard, 2026-09-29:
Thanks @Lapshin !