espressif/esp-idf #12182: Core  0 panic'ed when BLE signal is low

matislovas, 2023-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Olimex ESP32-POE-ISO Wroom 32E

### Power Supply used.

USB

### What is the expected behavior?

No core panic

### What is the actual behavior?

I am using ESP32 in `gattc` mode, and it sometimes panics, when BLE signal is on the edge of disconnect

### Steps to reproduce.

1. Setup BLE Gattc.
2. Put server device in near-out-of-range location.

### Debug Logs.

```plain
ASSERT_PARAM(0 0), in lld_evt.c at line 1487
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008fadc: f01d020c 00004136 f01d0000
0x4008fadc: btdm_sleep_check_duration at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:949

Core  0 register dump:
PC      : 0x4008fae3  PS      : 0x00060f34  A0      : 0x80083ad5  A1      : 0x3ffc18b8  
0x4008fae3: r_assert at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:1889

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f412287  
A6      : 0x000005cf  A7      : 0xfffffffb  A8      : 0x8000814b  A9      : 0x3ffc1828  
A10     : 0x00000000  A11     : 0x3ffc184c  A12     : 0x3ffc17f7  A13     : 0x00000037  
A14     : 0x00000000  A15     : 0x3ffc17fd  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400839dd  LEND    : 0x400839e5  LCOUNT  : 0x00000000  
0x400839dd: r_assert_param at ??:?

0x400839e5: r_assert_param at ??:?

Backtrace: 0x4008fae0:0x3ffc18b8 |<-CORRUPTED
0x4008fae0: r_assert at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:1888
```
```

zhp0406, 2023-09-04:
@matislovas 
thanks for report the issues，
The issue have been fixed in the latest version. could you update the IDF version and see if the problem still exists?
if also exists，could you please provide me with the complete log or share your code with me to help me reproduce the issue?

matislovas, 2023-09-04:
> @matislovas thanks for report the issues， The issue have been fixed in the latest version. could you update the IDF version and see if the problem still exists? if also exists，could you please provide me with the complete log or share your code with me to help me reproduce the issue?

My current code base is a bit incompatible with `5.1` release (I'm in process to make it compatible). Maybe you can point me to commits with a fix for this problem? I will patch my current `esp-idf` version with it, and will give it a go.

zhp0406, 2023-09-04:
git show 909ec6673e2b7935bd13916931b281e98842eccf
commit 909ec6673e2b7935bd13916931b281e98842eccf
Author: chenjianhua <chenjianhua@espressif.com>
Date:   Sun Jun 11 11:42:36 2023 +0800

    Update bt-lib for esp32
    
    1 Fixed disconnection delay on peripheral when slave latency is not zero
    2 Fixed LLCP info init for central and peripheral
    3 Fixed sent cmd status evt twice when process disconnect cmd
    
    
 warning：Applying this commit directly may result in conflicts

zhp0406, 2023-09-11:
@matislovas 
Have you resolved your issue?

Alvin1Zhang, 2023-10-25:
Thanks for sharing the updates, and glad to hear that the issue is resolved. Feel free to reopen.

closed 2023-10-25 by Alvin1Zhang