espressif/esp-idf #12183: ESP32-C3 esp_deep_sleep_start changes enable of internal pullup and pulldown internal resistor

spilz87, 2023-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.5

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

I'm building a remote control with an ESP32-C3.
As it works on battery, low consumption is verry important.

I want to wake up from deep sleep when a switch is relased on GPIO 0. I use 100k pull up resistors for the switch to consume less as possible.

when I enter in sleep mode, I want to disable internal pull-up/pull-down resistors as I use the external ones and I expect the ESP32-C3 wakes up from LOW to HIGH on GPIO 0.

### What is the actual behavior?

when ESP32-C3 is deep sleep mode and I relase the switch, I mesure only 0.4v on GPIO 0 instead of 3.3V (cf external 100k pull-up resistor) so ESP32-C3 doesn't wake up.

### Steps to reproduce.

Step 1 : connect a 100k pull-up resistor on GPIO 0 
Step 2 : connect a switch between GPIO 0 and GND
Step 3 : close the switch
Step 4 : enter in deep sleep mode with GPIO wake up on HIGH for GPIO 0
Step 5 : open the switch
Step 6 : ESP32-C3 doesn't wake up
Step 7 : measure the voltage on GPIO 0 : it will be 0.4V instead of 3.3V

[image]

### More Information.

On 4.4.4 the issue was only in the first enter in deepsleep mode
On 4.4.5 the issue is alway present.

I discover that **esp_deep_sleep_start** enables internal pull-down resistor (~10k) which is not compatible with the external 100k pull-up resistor.

We should be able to choose if we want to use or not the internal resistor.

this issue is linked to #11024 which constat the issue but doesn't really solve it.
this issue is linked to [https://github.com/espressif/arduino-esp32/issues/8447](url)

thanks for your help

spilz87, 2023-09-03:
In **sleep_modes.c** : 

esp_deep_sleep_start(void) calls esp_sleep_start : 
[https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L641](url)
esp_sleep_start calls gpio_deep_sleep_wakeup_prepare : 
[https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L444](url)
gpio_deep_sleep_wakeup_prepare change internal pull-up and internal pull-down settings 
[https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L1169](url)

funkBuild, 2023-09-04:
+1 
I've run into the exact same issue. I was previously using `gpio_sleep_set_pull_mode` to set the pull mode during deep sleep but that seems to get overridden by `esp_deep_sleep_start` with the new logic that was introduced. Ideally there should be a manual way of specifying the pull config instead of assuming the value

spilz87, 2023-09-10:
> In **sleep_modes.c** :
> 
> [esp_deep_sleep_start(void) calls esp_sleep_start](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L641) [esp_sleep_start calls gpio_deep_sleep_wakeup_prepare](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L444) [gpio_deep_sleep_wakeup_prepare change internal pull-up and internal pull-down settings](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/sleep_modes.c#L1169)

By removing these forced internal resistor, and compiling again, my board works perfectly. 

And the consumption in deep sleep mode from 400uA to 80uA, which is much better !!! 

I don’t get why they keep this way to do 
I don’t know how to propose this modification.

spilz87, 2023-09-12:
Which version do you use ? 
i had a workaround on V4.4.4 which didn’t work anymore on V4.4.5 :(

actually I changed the file as explained and everything works as expected, with no workaround needed.

QElis, 2023-09-12:
Dear "spilz87",

I am using arduino-esp32 2.1.1 which is based on ESP-IDF-4.4.5 — my workaround should be fine for those, who do not want to recompile arduino-esp32, e.g., because they want to distribute their projects to workd out of the box. I perfectly agree that removing "esp_deep_sleep_wakeup_prepare()” from “esp_start_sleep()” in “sleep_modes.c” altogether is the proper solution, but I wonder, how long it takes to find its way to main-stream ardiuno-esp32.

BTW: what was your workarount in 4.4.4? I did not notice relevant changes in “seep_modes.c” recently. And the IDF documentation explicitly states that it will set pull-ups and pull-downs when entering sleep. Circumventing these to take effect by “gpio_hold_en(.)”+”gpio_deep_sleep_hold_en()” appears to me a compliant solution for the time being.

Kind Regards,
Thomas.

spilz87, 2023-09-12:
> esp_deep_sleep_enable_gpio_wakeup(1UL << GPIO_NUM_4, ESP_GPIO_WAKEUP_GPIO_HIGH);
> gpio_hold_en(GPIO_NUM_4);  
> gpio_deep_sleep_hold_en();  
> esp_deep_sleep_start();

seems to work, good job !

I still think the way they do is not logical and it's not normal we have to find a workaround for it because they want to help people who don't know how to do.

hope this workaround will still works in next updates 

I was working on it for more than 3 months to find a way to solve it.

10086loutianhao, 2023-09-13:
@spilz87 I am considering whether it is possible to add an option to support customers in configuring external pull-up or pull-down resistors without using the hold function.

10086loutianhao, 2023-09-13:
[image]
@spilz87 @QElis Hello, I believe that commenting out the code within the box can also achieve self-implemented external pull-up and pull-down without interference. Could you please help me verify this? Testing conditions on my end are not very convenient. Thank you. If it's confirmed to work after being verified, I will add this option.

spilz87, 2023-09-13:
> commenting out the code within the box can also achieve self-implemented external pull-up and pull-down without interference

@10086loutianhao, I commented this code and compile again and I can confirm it solves the problem with the use of external pull-up pull-down resistors. It was like magic after 3 months trying to find what was wrong :D

spilz87, 2023-09-13:
> @spilz87 I am considering whether it is possible to add an option to support customers in configuring external pull-up or pull-down resistors without using the hold function.

That sounds a very good idea. 👍

spilz87, 2023-09-13:
For me, as electronic engineer, the use of internal resistor has to be always done consciously. If you want to use them, you have to specify it, other while they are not set. Like when you define a GPIO in input mode you specially precise if you want to active internal resistor, they are not by default.

10086loutianhao, 2023-09-18:
@spilz87 I have submitted a merge request (MR) with the intention of adding this option to the master branch. It will be available for customers to disable internal resistors approximately after the release of version 5.2. However, the default setting will still enable internal pull-up and pull-down resistors. This decision is made to avoid a significant breaking change, as changing the default setting to disable internal resistors would have a substantial impact. Your understanding is appreciated.

spilz87, 2023-09-18:
Hi @10086loutianhao 
Thank you :)

I understand the decision about default settings. 

What about version 4.4 ? Is there a way have this merge too ? As lot of people as me use arduino environnement which works with V4.4

thank for your help again :)

10086loutianhao, 2023-09-19:
@spilz87  hello, I will backport this change to v4.4

spilz87, 2023-09-19:
> @spilz87 hello, I will backport this change to v4.4

Thanks a lot :)

10086loutianhao, 2023-11-15:
https://gitlab.espressif.cn:6688/espressif/esp-idf/-/merge_requests/26043 has fixed it

Alvin1Zhang, 2023-11-15:
Thanks for reporting, feel free to reopen.

closed 2023-11-15 by Alvin1Zhang

MiguelMLR9, 2025-09-21:
> I have observed the same behaviour, however with GPIO4 (should make no difference) and with 39K as opposed to 100K (leaving me with about 0.7V instead of 3.3V, still far to low for a successful wakeup). The documentation of "[esp_deep_sleep_enable_gpio_wakeup(...)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#_CPPv433esp_deep_sleep_enable_gpio_wakeup8uint64_t33esp_deepsleep_gpio_wake_up_mode_t)" actually states that "...you don’t need to care to pull-up or pull-down before using this function, because this will be set internally in esp_deep_sleep_start based on the wakeup mode...". So this is not an issue but intended behaviour.
> 
> As a workaround, I now invoke "gpio_hold_en(...)" and "gpio_deep_sleep_hold_en()" before entering deep sleep. This prevents the "gpio_pulldown_en()" coded in "esp_deep_sleep_start()"/"esp_deep_sleep_wakeup_prepare()" from taking any effect. Although the documentation of "[gpio_hold_en(...)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/gpio.html#_CPPv412gpio_hold_en10gpio_num_t)" does not explicitly mention that it also latches pull-ups/pull-downs, it apparently does.
> 
> The below code works fine and shows 3.2V at the GPIO when the button is opened during sleep.
> 
> ```
> esp_deep_sleep_enable_gpio_wakeup(1UL << GPIO_NUM_4, ESP_GPIO_WAKEUP_GPIO_HIGH);
> gpio_hold_en(GPIO_NUM_4);  
> gpio_deep_sleep_hold_en();  
> esp_deep_sleep_start();
> ```

It's working for me, thanks