espressif/esp-idf #12184: esp_coex is missing function implementations for ESP32

Pespiri, 2023-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Project to build

### What is the actual behavior?

Project fails to build due to missing function implementation

### Steps to reproduce.

1. Step: build project implementing at least two of BT, BLE or WiFi
1. Step: compile and wait for fail

### Build or installation Logs.

```plain
Processing pico32 (platform: espressif32@6.4.0; framework: espidf; board: pico32)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/pico32.html
PLATFORM: Espressif 32 (6.4.0) > ESP32 Pico Kit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- libraries
Building in release mode
Compiling .pio/build/pico32/src/utilities/esp_coex_hack.o
Linking .pio/build/pico32/firmware.elf
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/pico32/esp-idf/bt/libbt.a(bta_dm_main.o):(.literal.BTA_DmCoexEventTrigger+0x0): undefined reference to `esp_coex_status_bit_set'
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/pico32/esp-idf/bt/libbt.a(bta_dm_main.o):(.literal.BTA_DmCoexEventTrigger+0x4): undefined reference to `esp_coex_status_bit_clear'
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/pico32/esp-idf/bt/libbt.a(bta_dm_main.o): in function `BTA_DmCoexEventTrigger':
/home/lenovo/.platformio/packages/framework-espidf/components/bt/host/bluedroid/bta/dm/bta_dm_main.c:458: undefined reference to `esp_coex_status_bit_set'
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lenovo/.platformio/packages/framework-espidf/components/bt/host/bluedroid/bta/dm/bta_dm_main.c:459: undefined reference to `esp_coex_status_bit_clear'
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lenovo/.platformio/packages/framework-espidf/components/bt/host/bluedroid/bta/dm/bta_dm_main.c:462: undefined reference to `esp_coex_status_bit_clear'
/home/lenovo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lenovo/.platformio/packages/framework-espidf/components/bt/host/bluedroid/bta/dm/bta_dm_main.c:463: undefined reference to `esp_coex_status_bit_clear'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico32/firmware.elf] Error 1
```

### More Information.

Searching the ESP-IDF master branch (up to commit https://github.com/espressif/esp-idf/commit/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a) for references for the functions reported missing does not turn up any code implementing these functionalities.
Functions are only declared, but not implemented.

wmy-espressif, 2023-09-04:
Hi @Pespiri , the undefined functions in the build log are in fact implemented in a git submodule of esp-idf: `esp-idf/components/esp_coex/lib`, which is not needed in previous esp-idf release branches such as release/v5.0. Please check whether your esp-idf directory contains this git submodule. If not, you can use command `git submodule update --init` to fetch the new git submodule.

Pespiri, 2023-09-04:
Ah, I see.

I believe I've managed to track down my issue to be an outdated `sdkconfig.defaults` in which some of the configuration options were outdated/deprecated. Updating my sdkconfig, specifically the ones for WiFi fixed the issue.

closed 2023-09-04 by Pespiri