espressif/esp-idf #12185: Undefined reference to `phy_bbpll_en_usb` when using USB CDC and WiFi at the same time on ESP32-S2

Strat00s, 2023-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1.1, master

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should compile.

### What is the actual behavior?

Looks like S2 is missing `phy_bbpll_en_usb` implementation as all I get is `undefined reference to 'phy_bbpll_en_usb'`.

### Steps to reproduce.

0. Have ESP32-S2
1. Create wifi station or scan example (probably any wifi example will do, I tried just these 2).
2. Enable USB when phy init, set USB CDC for console output.
3. Try and build it.

### Build or installation Logs.

```plain
[1/111] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32s2/bootloader_soc.c.obj
[2/111] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_console_loader.c.obj
[3/111] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32s2/bootloader_sha.c.obj
[4/111] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_panic.c.obj
[5/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_fields.c.obj
[6/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_table.c.obj
[7/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_rtc_calib.c.obj
[8/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_rtc_table.c.obj
[9/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32s2/rtc_sleep.c.obj
[10/111] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log.c.obj
[11/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32s2/rtc_clk_init.c.obj
[12/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32s2/rtc_init.c.obj
[13/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_utility.c.obj
[14/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj
[15/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_fields.c.obj
[16/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_api.c.obj
[17/111] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/efuse_controller/keys/with_key_purposes/esp_efuse_api_key.c.obj
[18/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/cpu.c.obj
[19/111] Building C object esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/esp_err.c.obj
[20/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/esp_memory_utils.c.obj
[21/111] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32s2/cpu_region_protect.c.obj
[22/111] Generating project_elf_src_esp32s2.c
[23/111] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xt_trax.c.obj
[24/111] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj
[25/111] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[26/111] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/dport_access_common.c.obj
[27/111] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s2/interrupts.c.obj
[28/111] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s2/gpio_periph.c.obj
[29/111] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32s2/spi_periph.c.obj
[30/111] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32s2/bootloader_esp32s2.c.obj
[... 75 lines cut ...]
[106/111] Linking C static library esp-idf\soc\libsoc.a
[107/111] Linking C static library esp-idf\xtensa\libxtensa.a
[108/111] Linking C static library esp-idf\main\libmain.a
[109/111] Linking C executable bootloader.elf
[110/111] Generating binary image from built executable
esptool.py v4.7.dev1
Creating esp32s2 image...
Merged 2 ELF sections
Successfully created esp32s2 image.
Generated C:/Users/Stratos/Desktop/wifi_tests/station/build/bootloader/bootloader.bin
[111/111] cmd.exe /C "cd /D C:\Users\Stratos\Desktop\wifi_tests\station\build\bootloader\esp-idf\esptool_py && C:\Users\Stratos\Espressif\master\tools\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/Stratos/Espressif/master/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/Stratos/Desktop/wifi_tests/station/build/bootloader/bootloader.bin"       
Bootloader binary size 0x5650 bytes. 0x19b0 bytes (23%) free.
[916/918] Linking CXX executable wifi_station.elf
FAILED: wifi_station.elf
cmd.exe /C "cd . && C:\Users\Stratos\Espressif\master\tools\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S2=0 -Wl,--Map=C:/Users/Stratos/Desktop/wifi_tests/station/build/wifi_station.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s2.peripherals.ld -T esp32s2.rom.ld -T esp32s2.rom.api.ld -T esp32s2.rom.libgcc.ld -T esp32s2.rom.newlib-funcs.ld -T esp32s2.rom.newlib-data.ld -T esp32s2.rom.spiflash.ld -T memory.ld -T sections.ld @CMakeFiles\wifi_station.elf.rsp -o wifi_station.elf  && cd ."
c:/users/stratos/espressif/master/tools/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_phy/libesp_phy.a(phy_init.c.obj):(.literal.esp_phy_load_cal_and_init+0x28): undefined reference to `phy_bbpll_en_usb'
c:/users/stratos/espressif/master/tools/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_phy/libesp_phy.a(phy_init.c.obj): in function `esp_phy_load_cal_and_init':
C:/Users/Stratos/Espressif/master/esp-idf/components/esp_phy/src/phy_init.c:788: undefined reference to `phy_bbpll_en_usb'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### More Information.

Tried it on 5.1.1 and master.

tore-espressif, 2023-11-23:
Hello @Strat00s ,
You're right and I'm sorry about the problem.

The `phy_bbpll_en_usb()` was introduced to tackle WiFi and USB interference on ESP32-S3 and ESP32-C3. We haven't encountered similar problem with the S2, so the `phy_bbpll_en_usb()` function is not implemented for S2.

Thus this option will be hidden for ESP32-S2, thanks!

imoverclocked, 2023-11-23:
Does it make sense to alias this to a noop for non-esp32-s3/c3?

tore-espressif, 2023-11-27:
> Does it make sense to alias this to a noop for non-esp32-s3/c3?

No, it doesn't. `phy_bbpll_en_usb()` is considered to be a workaround for a bug that is specific to S3 and C3, so we don't want to define (empty) implementation for other chips. It might confuse everyone. A better solution is to offer the workaround only for chips that need it

Akellych, 2024-03-03:
I have the same error when I enable ESP_PHY_ENABLE_USB
in the phy_init.c file there are the following lines
#if CONFIG_ESP_PHY_ENABLE_USB
      phy_bbpll_en_usb(true);
#endif
in function esp_phy_load_cal_and_init(void)
maybe this is the problem on chip S2

closed 2024-03-07 by espressif-bot