espressif/esp-idf #12190: ESP32-S3 enabling GPIO41 after burning EFuses

Stencht, 2023-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have custom PCB's with ESP32-S3-WROOM-1 which use **GPIO41 as an output**, but for some reason I am not able to set this pin high. This is topic is similar to one on the [espressif forum](https://esp32.com/viewtopic.php?t=29047), but it grow stale so I am hoping this forum is more alive :) 

The code works on a dev-board using the same chip with the same code, the only difference is that the custom boards use DFU/CDC (so no UART). For which I had to burn the efuse **USB_PHY_SEL**. 

I noticed that GPIO_NUM_41 is used for JTAG when enabled. For what I could tell it should already be disabled, but just to be sure I followed the instructions as described in the datasheet under '**JTAG Signal Source Control**':

1. Set GPIO3 high (should matter)
2. Burned fuses **DIS_PAD_JTAG** and **DIS_USB_JTAG**, which should permanently disable JTAG

But still no luck.

Other test:

- I tried to set another pin high on my custom board, not a problem
- I verified the solder connection with a microscope, looked fine
- I tested on two different custom boards

Any ideas?
The only idea I have left is opting for a flying wire to a GPIO I can set :(

Stencht, 2023-09-04:
Update: 

- I tried to add a flying wire to another IO (which I was sure I could set high), but that didn't work because something (GPIO41 which is connected by internal PCB tracing?) is pulling the IO down apparently. 
- I did the same test but with 3.3V instead of an IO and the board worked, but I think, that a lot of current is being sank somewhere (in GPIO41?).
- Resistance between GPIO41 and GND is around 1MOhm (board unpowered ofc)

 PS: The only thing the IO should do is enable a voltage switch, thats all. I checked the circuit and routing again and everything looks clean, pretty straight forward connection... I reused this circuit from a previous project (but with different processor). 

[image]

Stencht, 2023-09-05:
I can confirm that it is **not a hardware issue**.

After recreating the circuit on breadboard and confirming the correct value at pin 41 on a new **ESP32-S3-DevKitC-1**, I burned the efuse USB_PHY_SEL to 1.
After doing this, the same behaviour occurs as with my custom boards. Then I proceeded with flashing the other two efuses (DIS_USB_JTAG & DIS_PAD_JTAG), still no change.

I did notice some interesting things:

- When connect to a 1M resistor to ground (without the MT9700 as shown in the circuit above), the measured voltage over the resistor is zero.
- When I remove this resistor and leave GPIO41 unconnected, I'm measuring a voltage of 1.8V over GPIO41 and ground.

Current method of setting GPIO41 as output (which worked before burning efuse):

```
// Configure IO
ESP_ERROR_CHECK(gpio_pullup_dis(PWR_SWITCH_PERIPHERALS_GPIO));
ESP_ERROR_CHECK(gpio_pulldown_dis(PWR_SWITCH_PERIPHERALS_GPIO));
ESP_ERROR_CHECK(gpio_intr_disable(PWR_SWITCH_PERIPHERALS_GPIO));
ESP_ERROR_CHECK(gpio_set_direction(PWR_SWITCH_PERIPHERALS_GPIO, GPIO_MODE_OUTPUT));
// Set output high
ESP_ERROR_CHECK(gpio_set_level(PWR_SWITCH_PERIPHERALS_GPIO, 1));
```

Maybe the default function of the pin has changed?
Although I would think that gpio_set_direction would set it as a GPIO (and not as MTDI or CLK_OUT1).

[image]

Any help would be highly appreciated!

Stencht, 2023-09-07:
Nobody? 😥

faizannazir, 2024-01-21:
> Nobody? 😥

[image]

[image]

BY Default they are gpio  d+ and  d- are used for jtag?

adixmasz, 2024-06-04:
In my case (custom board with ESP32-S3-WROOM-1U-N8) pin GPIO41 works when invoke gpio_reset_pin(GPIO_NUM_41) before set its direction and level.

`		gpio_reset_pin(GPIO_NUM_41); // Gpio needs reset to exit JTAG mode
		gpio_set_direction(GPIO_NUM_41, GPIO_MODE_OUTPUT);
		gpio_set_level(GPIO_NUM_41, 1);
`

Take it from that thread on forum you have mention before: [https://www.esp32.com/viewtopic.php?t=38364](url)